openbmc/openbmc #3382: Flash an AMI BIOS by BMC

lockless, 2018-09-19:
I need to flash the AMI BIOS by BMC.
But now, it seems that the “pflash” process can only flash OpenPower NOR

throw error like this：
```
Sep  6 12:21:13 zaius daemon.info flash_bios.exe[955]: Failed to init flash: -1
Sep  6 12:21:13 zaius daemon.info flash_bios.exe[955]: Starting flasher: bios,/tmp/bios.bin,/org/openbmc/control/flash/bios,
Sep  6 12:21:13 zaius daemon.info flash_bios.exe[955]: flasher: /tmp/bios.bin, BMC = 1, address = 0x0
Sep  6 12:21:13 zaius daemon.info flash_bios.exe[955]: Setting up flash
```

did openbmc now has other processes support this function ?
thanks !

mine260309, 2018-09-19:
You could use `pflash --help` to find out the `-b, --bmc` option to target for BMC chip.
And this is better to be asked in mailing list, instead of a github issue.

gtmills, 2018-10-12:
Hi @lockless, this is a good question for IRC or the mailing list, as it's not a bug or feature request. You can sign yourself up for the mailing list at https://lists.ozlabs.org/listinfo/openbmc

closed 2018-10-12 by gtmills