openbmc/openbmc #3384: SDK failing to build for witherspoon

geissonator, 2018-09-20:
The new phosphor-tiny option has broken the witherspoon SDK builds out at https://openpower.xyz/job/openbmc-build-sdk/

```
20:17:44 | cp: cannot stat '/tmp/openbmc/work/x86_64-nativesdk-oesdk-linux/nativesdk-python/2.7.14-r1/recipe-sysroot-native//usr/local/oecore-x86_64/sysroots/x86_64-oesdk-linux/usr/lib/python2.7/_sysconfigdata.py': No such file or directory
```

This was from the change I needed to do to get the .py files removed from the BMC rofs. I'll get a fix up.

amboar, 2018-09-25:
Please fix this! It was introduced by

```
2 18:03:24 andrew@dave:~/src/openbmc/openbmc ((25aa00c080db...)|BISECTING) $ git bisect good
764b88f4056cc98082e233216704e94613499e64 is the first bad commit
commit 764b88f4056cc98082e233216704e94613499e64
Author: Andrew Geissler <geissonator@yahoo.com>
Date:   Fri Sep 7 10:26:42 2018 -0500

    phosphor-tiny: Enable tiny distro on witherspoon
    
    Witherspoon is flash space constrained so enable new
    tiny distro feature to help alleviate this
    
    (From meta-ibm rev: 49f2b5b31a21eb3dea2d83d2d13551475ef4bb27)
    
    Change-Id: Ia3e90281d4c5da1c95cddb1b802f20c614b25885
    Signed-off-by: Andrew Geissler <geissonator@yahoo.com>
    Signed-off-by: Brad Bishop <bradleyb@fuzziesquirrel.com>

:040000 040000 95a03f9888f1e6d6a7b58a3d516b0b824c4915de 881b27f91d2cc3c68ed9c8a87822ce26fc2f5f0f M      meta-ibm
```
Lets revert it until you can sort it out properly

rfrandse, 2018-09-25:
<https://gerrit.openbmc-project.xyz/13201> Keep _sysconfigdata.py in image
Resolves  #3384 SDK failing to build for witherspoon

closed 2018-09-27 by geissonator