openbmc/openbmc #3386: WebUI lists hardware components in wrong order

CPM80, 2018-09-24:
If there are multiple components that share the name and only differ in their positional number, they are displayed in the list out of normal sorting order. E.g., DIMMs are listed as:
    DIMM 0
    DIMM 1
    DIMM 10
    DIMM 100
    DIMM 101
    ...
    DIMM 109
    DIMM 11
    DIMM 110
    ...
Instead of:
    DIMM 0
    DIMM 1
    DIMM 2
    DIMM 3
    ...
    DIMM 10
    DIMM 11
    ...
    DIMM 99
    DIMM 100
    DIMM 101
  The WebUI receives an inventory list in Json format from the rest-dbus interface. The rest-dbus implemented in rest-dbus.py uses the JSONPlugin from the Bottle framework to generate a Json responce from python's dictionary. JSONPlugin uses a standart json library, which sorts the dictionary keys with the sort_keys option. The library doesn't support custom sorting functions for the dictionary keys. To change the keys sorting order, the method _It should be reimplemented for the type DictKey based on the standart std type. This will changed the sorting order to the proper one (see attached patch). However after these changes the time to obtain the inventory list through WebUI will increase significantly (from 5s to 10s for the Vesnin server). Could you suggest any other solution?
[0001-inventory-sort.patch.txt](https://github.com/openbmc/openbmc/files/2409656/0001-inventory-sort.patch.txt)

gtmills, 2018-09-24:
[image]

rfrandse, 2018-10-01:
<https://gerrit.openbmc-project.xyz/13430> Order hardware components
Resolves  #3386 WebUI lists hardware components in wrong order

gtmills, 2018-10-04:
Using the algorithm in https://gerrit.openbmc-project.xyz/13430/1/ we get 
``` 
Fan 0   - 4
Fan 1   - 4
Fan 3   - 4 
Powersupply 0   - 4
Powersupply 1   - 4
BMC    - 5
Fan 0_0    - 5
Fan 0_1    - 5
Fan 1_0    - 5
Fan 1_1    - 5
... 
```
I think 

``` 
Fan 0 - 4
Fan 0_0 - 5
Fan 0_1  - 5
Fan 1 - 4
Fan 1_0  - 5
Fan 1_1  - 5
Fan 3 - 4 
Fan 3_0  - 5
Fan 3_1  - 5
Powersupply 0 - 4
Powersupply 1 - 4
... 
```
would be better..

gtmills, 2018-10-04:
What if we sort alphanumerically the children but they are still under there parent. So more of a Depth-first search.

gtmills, 2018-10-05:
Opened https://github.com/openbmc/openbmc/issues/3401 for the backend, but will make this change in the GUI until the backend does this.

closed 2018-10-11 by geissonator

nest1ing, 2019-06-10:
There is same issue on the page with sensors list and I've pushed a fix for this:
https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-webui/+/22433