openbmc/openbmc #3388: MAC#2 RGMII mode can't ping other IP address on AST2500

xiuzhi123, 2018-09-26:
My hardware set AST2500 MAC#2  on RGMII mode. the device is Broadcom BCM54612, The PHY and control is good . But it can not ping other IP. 
My openbmc version is release v2.3, 
My kernel version KBRANCH = "dev-4.17" LINUX_VERSION ?= "4.17.4" and "4.17.14"    
The attachments are : 
1,kernel config file: [ast2500kernelconfig.txt](https://github.com/openbmc/openbmc/files/2418477/ast2500kernelconfig.txt)
2,dmesg file:  
[dmesg-ast2500-mac2-pingerror.log](https://github.com/openbmc/openbmc/files/2418485/dmesg-ast2500-mac2-pingerror.log)
3, .dts file:  
[aspeed-bmc-opp-romulus.dts.txt](https://github.com/openbmc/openbmc/files/2418490/aspeed-bmc-opp-romulus.dts.txt)
4, serial mesg file:
[serialmesg-ast2500-mac2-pingerror.log](https://github.com/openbmc/openbmc/files/2418488/serialmesg-ast2500-mac2-pingerror.log)

The &mac1 in file .dts  is:
&mac1 {

        status = "okay";

        pinctrl-names = "default";

        pinctrl-0 = <&pinctrl_rgmii2_default>;

};

The kernel boot message error snippet: 
... 
[    4.181961] aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl: end allocate all pins
[    4.189035] aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl: end acquired the pins
[    4.196116] aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl: end ret = ops->set_mux
[    4.203342] aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl: ret >0  ERROR goto err_setmux ret=-1
[    4.211776] aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl: enter err_set_mux ret=-1
[    4.219167] aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl: err_pin_request
[    4.225780] ftgmac100 1e680000.ethernet: PIN_MAP_TYPE_MUX_GROUP
[    4.231699] ftgmac100 1e680000.ethernet: ret=-1
[    4.236349] ftgmac100 1e680000.ethernet: Error applying setting, reverse things back
[    4.244338] ftgmac100 1e680000.ethernet: Generated random MAC address 86:7b:c3:69:dc:77
[    4.260689] libphy: ftgmac100_mdio: probed
[    4.264981] Generic PHY 1e680000.ethernet--1:00: attached PHY driver [Generic PHY] (mii_bus:phy_addr=1e680000.ethernet--1:00, irq=POLL)
[    4.288343] ftgmac100 1e680000.ethernet eth1: irq 20, mapped at 66942b4f
...

The values of register SCU0C and SCU70 are correct, the values are:

#devmem 0x1e6e200C

0xFFDD7A8B,  the bit 21 is 0(MAC#2 enable clock running) ,the bit 20 is 1

#dev mem 0x1e6e2070

0xF109F296 , the bit 7 is 1 (RGMII),the bit6 is 0

Execting the command "ifconfig eth0 up" log is:

root@romulus:~# ifconfig eth0 up 

[  274.978358] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready

root@romulus:~# [  274.986878] ftgmac100 1e680000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx

[  274.995944] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

My bmc ip is 192.168.120.141, it can't ping other IPs.

I also set the gateway. route add default gw 192.168.120.254.

I trace the error  "ftgmac100 1e680000.ethernet: Error applying setting, reverse things back" in file drivers/pinctrl/core.c 
L1206
list_for_each_entry(setting, &stte->settings,node)
switch(setting->type）{
case PIN_MAP_TYPE_MUX_GROUP
      ret = pinmux_enable_setting(setting);
}
the ret value is -1
My kernel version is 4.17.14. The other kernel versions reproduce the same error.
What is wrong with it?

legoater, 2018-09-26:
The mac1 definition is not in sync with the OpenBMC kernel : https://github.com/openbmc/linux/blob/dev-4.17/arch/arm/boot/dts/aspeed-ast2500-evb.dts#L74

Did you change it ?

legoater, 2018-09-26:
I am taking it that back. You are on a romulus and not a evb. 

What about the "use-ncsi" property ? Shouldn't you be using it ?

xiuzhi123, 2018-09-26:
I modified the &mac1 to aspeed-ast2500-evb, the same error 
The eth0 on my hardware can use ncsi. the eth1  has no ncsi available,    
updating the openbmc version to the latest  version , kernel branch dev-4.18, got the same error

legoater, 2018-09-26:
what about the D2-PLL, is it running ?

xiuzhi123, 2018-09-26:
@legoater , Do you think this issue is related to the host platform?   Please see my hardware sch file :
[image]

xiuzhi123, 2018-09-26:
[image]

xiuzhi123, 2018-09-26:
devmem 0x1e6e213c : 0x00000580
devmem 0x1e6e201c : 0x8E00A17C  , D2-PLL enabled

legoater, 2018-09-26:
yes the D2-PLL looks correct  : 250MHz.

What about SCU90 ? you wrote that you added : 
          `
           pinctrl-0 = <&pinctrl_rgmii2_default &pinctrl_mdio2_default>;
          `
to `mac1` and  the princtrl driver emitted the same errors ? 

Is that a new OpenPOWER platform you are trying to support ?

xiuzhi123, 2018-09-27:
devmem 0x1e6e2090 ：0x00000004 ,  the SCU  bit15 is 0 ,not the default 1
I found this error reason was  that MAC#2 was not initialized in uboot drivers/net/aspeednic.c
I got the patch from meta-zaius 
[0002-board-aspeed-aspeednic-Use-MAC2-for-networking.patch.txt](https://github.com/openbmc/openbmc/files/2422214/0002-board-aspeed-aspeednic-Use-MAC2-for-networking.patch.txt)
now it works.

mine260309, 2018-09-27:
I see multiple users are having this kind of issue, is it possible to improve this in linux kernel (or via dts), so it does not rely on u-boot to init specific register to make a network device to work?

legoater, 2018-09-27:
@xiuzhi123 good to hear ! Although I would have thought that the Linux pinctrl driver would have set correctly these pins. May be we have a bug ? 

It is true that we have two network drivers in the OpenBMC U-Boot, the modified ftgmac100 driver for rgmii  (which initializes correctly both MACs) and the aspeednic driver for NCSI. 

I am trying to push in mainline U-Boot a new network driver for the Aspeed SoC : 

http://patchwork.ozlabs.org/cover/968067/

On which we could then work to add a NC-SI backend. I lack the HW locally. 

@mine260309 yes, Linux needs to be improved in some areas to stop relying on U-Boot initialization. WDT reset_mask is one example.

amboar, 2018-11-08:
Sorry, I'm getting to this really late, but regardless:

If `SCU90[15]` is the problem then this is a bug in the devicetree.

https://github.com/openbmc/linux/blob/dev-4.18/drivers/pinctrl/aspeed/pinctrl-aspeed-g5.c#L2427

In theory if you add the `bias-disable` property to the `pinctrl_rgmii2_default` node (see https://github.com/openbmc/linux/blob/dev-4.18/Documentation/devicetree/bindings/pinctrl/pinctrl-aspeed.txt#L47) you should get the right outcome. If that doesn't work then there are bugs that need fixing. There is a slightly awkward catch in that we can't specify the pulldown state separately for Tx and Rx as the hardware allows, as the RGMII2 function is described as a whole by 1 mux bit (`SCU70[7]`) and the devicetree bindings map to the pinmux bits, not the pinconf bits. Not sure what to do if that's problematic.

closed 2018-12-07 by xiuzhi123