openbmc/openbmc #3390: Not any system instances in system inventory after the host os booted.

jinneal, 2018-09-26:
I can not find any implemented (Item.Board) objects in system inventory implemented in our real hardware:

root@openbmc:~# busctl tree xyz.openbmc_project.Inventory.Manager |cat
└─/xyz
  └─/xyz/openbmc_project
    └─/xyz/openbmc_project/inventory
      └─/xyz/openbmc_project/inventory/system

 root@evb-ast2500:~# busctl introspect xyz.openbmc_project.Inventory.Manager /xyz/openbmc_project/inventory/system |cat
NAME                                             TYPE      SIGNATURE RESULT/VALUE FLAGS
org.freedesktop.DBus.Introspectable              interface -         -            -
.Introspect                                      method    -         s            -
org.freedesktop.DBus.Peer                        interface -         -            -
.GetMachineId                                    method    -         s            -
.Ping                                            method    -         -            -
org.freedesktop.DBus.Properties                  interface -         -            -
.Get                                             method    ss        v            -
.GetAll                                          method    s         a{sv}        -
.Set                                             method    ssv       -            -
.PropertiesChanged                               signal    sa{sv}as  -            -
xyz.openbmc_project.Inventory.Decorator.AssetTag interface -         -            -
.AssetTag                                        property  s         ""           emits-change writable

and the result of enumerate list from Dbus:

root@host-os:~# curl -c cjar -b cjar -k -H "Content-Type: application/json" https://${bmc}/xyz/openbmc_project/enumerate
{
  "data": {
    "/xyz/openbmc_project/control": {}, 
    "/xyz/openbmc_project/control/host0": {}, 
    "/xyz/openbmc_project/control/host0/TPMEnable": {
      "TPMEnable": false
    }, 
    "/xyz/openbmc_project/control/host0/auto_reboot": {
      "AutoReboot": true
    }, 
    "/xyz/openbmc_project/control/host0/boot": {
      "BootMode": "xyz.openbmc_project.Control.Boot.Mode.Modes.Regular", 
      "BootSource": "xyz.openbmc_project.Control.Boot.Source.Sources.Default"
    }, 
    "/xyz/openbmc_project/control/host0/boot/one_time": {
      "BootMode": "xyz.openbmc_project.Control.Boot.Mode.Modes.Regular", 
      "BootSource": "xyz.openbmc_project.Control.Boot.Source.Sources.Default", 
      "Enabled": true
    }, 
    "/xyz/openbmc_project/control/host0/power_cap": {
      "PowerCap": 0, 
      "PowerCapEnable": false
    }, 
    "/xyz/openbmc_project/control/host0/power_restore_policy": {
      "PowerRestorePolicy": "xyz.openbmc_project.Control.Power.RestorePolicy.Policy.AlwaysOff"
    }, 
    "/xyz/openbmc_project/control/host0/restriction_mode": {
      "RestrictionMode": "xyz.openbmc_project.Control.Security.RestrictionMode.Modes.None"
    }, 
    "/xyz/openbmc_project/control/host0/turbo_allowed": {
      "TurboAllowed": true
    }, 
    "/xyz/openbmc_project/control/minimum_ship_level_required": {
      "MinimumShipLevelRequired": true
    }, 
    "/xyz/openbmc_project/control/power_supply_redundancy": {
      "PowerSupplyRedundancyEnabled": true
    }, 
    "/xyz/openbmc_project/dump": {}, 
    "/xyz/openbmc_project/dump/entry/1": {
      "Elapsed": 1537262533, 
      "Size": 45544
    }, 
    "/xyz/openbmc_project/dump/entry/2": {
[... 268 lines cut ...]
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm20", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm27", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm26", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm25", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm24", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm29", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm28", 
    "/xyz/openbmc_project/object_mapper", 
    "/xyz/openbmc_project/logging/internal/manager", 
    "/xyz/openbmc_project/led/groups", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1/core11", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1/core10", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core6", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core7", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core4", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core5", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core2", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core3", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core0", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core1"
  ], 
  "message": "200 OK", 
  "status": "ok"

Who know the reason? any configuration or code missing?

jinneal, 2018-09-26:
In general a way, which service implements the Item.Board?  which service provide the objects of system inventory?

mine260309, 2018-09-26:
If it's a fresh new system (or you did factory-reset), you need to power on the host once to get the inventories.

jinneal, 2018-09-26:
Do you mean all the inventories are generated by inventory_items.py a which will read eeprom to create all the objects to inventory manager after host power on?

closed 2018-09-27 by jinneal