openbmc/openbmc #3391: Ability to specify boot partition UUID

AlexanderAmelkin, 2018-09-26:
IPMI Specification defines a number of "boot parameters" that can be set via Set System Boot Options command. Among those parameters are:

    5: Boot flags
    7: Boot initiator mailbox

The former is fully supported by Petitboot, while the latter isn't. The data contained in the "mailbox" is not specified except for the size and the first four data bytes.

An [issue has been filed with Petitboot](/open-power/petitboot/issues/45) to develop a specification for this mailbox. Once that specification is developed, the appropriate support on openbmc side will have to be added.

In fact, support for setting this mailbox data via IPMI can be added right away, without waiting for the specification. Support for setting this via REST will have to wait for the specification.

AlexanderAmelkin, 2019-01-18:
Now as Petitboot supports that (see https://github.com/open-power/petitboot/commit/78c3a044d2302bacf27ac2d9ef179bc35824af4c), I guess we need to support boot initiator mailbox in openbmc.

AlexanderAmelkin, 2019-05-31:
This is just a note that there is a work in progress on our side for this issue.
It's almost done and will soon be pushed to Gerrit.
Please don't pay attention to those commits above, they were early and temporary.

rfrandse, 2019-06-03:
<https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-host-ipmid/+/22195> Add 3-byte conversion functions to endian.hpp
Partially resolves  #3391 Ability to specify boot partition UUID

AlexanderAmelkin, 2019-06-04:
@rfrandse, there is a better link to the whole set of changes:
https://gerrit.openbmc-project.xyz/q/topic:Add-Boot-Mailbox-GH-3391

AlexanderAmelkin, 2019-12-04:
No it's not stale. The work will be continued soon.

AlexanderAmelkin, 2020-07-30:
We've lost hope to persuade the maintainers to include this feature into the common code base.
Thus the feature has been included into meta-yadro layer as a set of patches (see https://github.com/openbmc/meta-yadro/commit/9ee3c28294c6a87912edbcf09b999b926634f836)
Anyone willing to use it is free to take it from there.

closed 2021-07-11 by stale[bot]