openbmc/openbmc #3393: obmc-chassis-poweroff@0.target: Job obmc-chassis-poweroff@0.target/start failed

gkeishin, 2018-09-27:
```
Sep 27 10:58:58 witherspoon systemd[1]: obmc-chassis-poweroff@0.target: Job obmc-chassis-poweroff@0.target/start failed with result 'dependency'.
Sep 27 10:58:58 witherspoon systemd[1]: obmc-host-stop@0.target: Job obmc-host-stop@0.target/start failed with result 'dependency'.
Sep 27 10:58:58 witherspoon systemd[1]: obmc-host-stop@0.target: Triggering OnFailure= dependencies.
Sep 27 10:58:58 witherspoon systemd[1]: op-occ-disable@0.service: Start request repeated too quickly.
Sep 27 10:58:58 witherspoon systemd[1]: op-occ-disable@0.service: Failed with result 'start-limit-hit'.
Sep 27 10:58:58 witherspoon systemd[1]: Failed to start OpenPOWER OCC Active Disable.
Sep 27 10:58:58 witherspoon systemd[1]: Dependency failed for Stop Host0.
Sep 27 10:58:58 witherspoon systemd[1]: Dependency failed for Chassis0 (Power Off).
Sep 27 10:58:58 witherspoon systemd[1]: obmc-chassis-poweroff@0.target: Job obmc-chassis-poweroff@0.target/start failed with result 'dependency'.
Sep 27 10:58:58 witherspoon systemd[1]: obmc-host-stop@0.target: Job obmc-host-stop@0.target/start failed with result 'dependency'.
Sep 27 10:58:58 witherspoon systemd[1]: obmc-host-stop@0.target: Triggering OnFailure= dependencies.
Sep 27 10:58:58 witherspoon systemd-udevd[712]: seq 1532 '/devices/platform/gpio-fsi/fsi0/slave@00:00/00:00:00:06/sbefifo1-dev0/occ-hwmon.1/hwmon/hwmon12' is taking a long time
Sep 27 10:58:58 witherspoon systemd[1]: op-occ-disable@0.service: Start request repeated too quickly.
Sep 27 10:58:58 witherspoon systemd[1]: op-occ-disable@0.service: Failed with result 'start-limit-hit'.
Sep 27 10:58:58 witherspoon systemd[1]: Failed to start OpenPOWER OCC Active Disable.
```

Once it starts crashing it keeps crashing over and over again..

-bash-4.1$ cat log.txt | grep "obmc-chassis-poweroff@0.target" | wc -l
256

geissonator, 2018-09-27:
yeah, just doing a power off in the middle of hostboot booting got me this, still not obvious what's failing.  The service didn't seem to even try to execute the script within it:
```
Sep 27 18:46:53 witherspoon phosphor-host-state-manager[1351]: Host State transaction request
Sep 27 18:46:53 witherspoon systemd[1]: Starting Soft power off of the host...
Sep 27 18:46:53 witherspoon systemd[1]: Stopped target Start Host0.
Sep 27 18:46:53 witherspoon systemd[1]: Stopped target Start Host0 Minimum.
Sep 27 18:46:53 witherspoon systemd[1]: Stopped Start Watchdog0.
Sep 27 18:46:53 witherspoon systemd[1]: Stopped target Host0 (Started).
Sep 27 18:46:53 witherspoon systemd[1]: Stopping Phosphor poweron watchdog...
Sep 27 18:46:53 witherspoon systemd[1]: Stopped target Chassis0 (Power On).
Sep 27 18:46:53 witherspoon systemd[1]: Stopped Witherspoon Fan Watchdog Reset.
Sep 27 18:46:54 witherspoon systemd[1]: op-occ-disable@0.service: Start request repeated too quickly.
Sep 27 18:46:54 witherspoon systemd[1]: op-occ-disable@0.service: Failed with result 'start-limit-hit'.
Sep 27 18:46:54 witherspoon systemd[1]: Failed to start OpenPOWER OCC Active Disable.
Sep 27 18:46:54 witherspoon systemd[1]: Dependency failed for Stop Host0.
Sep 27 18:46:54 witherspoon systemd[1]: Dependency failed for Chassis0 (Power Off).
Sep 27 18:46:54 witherspoon systemd[1]: Dependency failed for Power0 Host Off.

```

rfrandse, 2018-09-27:
<https://gerrit.openbmc-project.xyz/13315> Revert "Increase StartLimitIntervalSec to 240s"
Resolves  #3393 obmc-chassis-poweroff@0.target: Job obmc-chassis-poweroff@0.target/start failed

geissonator, 2018-09-27:
@spinler made a great observation, I recently put in https://github.com/openbmc/meta-phosphor/commit/ee52526c80eaca65a581c01bcf703861ec1a80b6.  The symptom here is that the service doesn't even get a chance to start. The issue is that the DefaultStartLimitBurst and DefaultStartLimitIntervalSec settings apply to service restarts, even in non-error situations!

This is why the system suddenly fixes itself, exactly 240s after the first start, it starts working again. I'll need to revisit what I was originally trying to fix with that issue. This new problem is definitely worse so I've reverted for now in https://gerrit.openbmc-project.xyz/#/c/openbmc/meta-phosphor/+/13315/