openbmc/openbmc #3394: ipmitool write fru failed --2400 platform

gx0128, 2018-09-28:
EEPROM M24256
 .dts
[image]
[image]

The FRU.bin size is 104 bytes. I don't know why the fru size is always 8 bytes when I  write the bin file to eeprom.
root@zhaoxinc:/tmp# ./ipmitool -I dbus fru write 0 /tmp/FRU.bin
Fru Size: 8bytes
Size to Write :8 bytes

I check the code, the ipmitool fru write command will call the function ipmi_fru_write from_bin(struct ipmi_intf *intf, char* fFileName,uint8_t fruId); 
then it will call the function readAllProerties in phosphor-host-ipmid. How can get the actual eeprom  properties? 
And I did not found SET_FRU_DATA 0X12 command register in Storagehandler.cpp , so it has not carried out yet, has it?

jinneal, 2018-09-28:
I think you can find the 0x12 command register callback function in /openbmc/ipmi-fru-parser/strgfnhandler.cpp, such as strgfnhandler.so found in lib

closed 2018-12-02 by gx0128

sheldonl3, 2019-04-09:
Could you tell me where can I find FRU.bin file? And is it the solution of issue #3395 if I write it to eeprom successfully?