openbmc/openbmc #3395: Can not show server information in system overview--AST2400

gx0128, 2018-09-29:
Currently, I send the following curl command to BMC.
curl -b cjar -k https://${bmc}/xyz/openbmc_project/inventory/system{ 
  "data": {
    "AssetTag": ""
  }, 
  "message": "200 OK", 
  "status": "ok"
}

It's not like the example in https://github.com/openbmc/docs/blob/master/rest-api.md#uploading-images 
[image]

Manufacture,SerialNumber,Model did not show in the response. How can I add these properties in the right way?Could you give me an example? Thanks!

CPM80, 2018-10-03:
1. Do you use static config.yaml for the phosphor-ipmi-fru (like: https://github.com/openbmc/openbmc/blob/master/meta-ibm/meta-palmetto/recipes-phosphor/ipmi/palmetto-ipmi-inventory-map/config.yaml) or dynamic yaml generation from server config xml (like the Witherspoon)?
  2. Does the host send fru data using IPMI to the BMC? Could you check sended FRU data?

gx0128, 2018-10-09:
thanks for your reply. @CPM80 
1. Yes. I use static config.yaml for the phosphor-ipmi-fru.  like:https://github.com/openbmc/openbmc/blob/master/meta-ingrasys/meta-zaius/recipes-phosphor/ipmi/zaius-ipmi-inventory-map/config.yaml
1:
  /system/chassis/motherboard/cpu0:
    entityID: 3
    entityInstance: 1
    interfaces:
      xyz.openbmc_project.Inventory.Decorator.Revision:
        Version:
          IPMIFruProperty: Custom Field 2
          IPMIFruSection: Board
          IPMIFruValueDelimiter: 58
      xyz.openbmc_project.Inventory.Decorator.Asset:
        BuildDate:
          IPMIFruProperty: Mfg Date
          IPMIFruSection: Board
        PartNumber:
          IPMIFruProperty: Part Number
          IPMIFruSection: Board
        Manufacturer:
          IPMIFruProperty: Manufacturer
          IPMIFruSection: Board
        SerialNumber:
          IPMIFruProperty: Serial Number
          IPMIFruSection: Board
      xyz.openbmc_project.Inventory.Item:
        PrettyName:
          IPMIFruProperty: Name
          IPMIFruSection: Board
2:
  /system/chassis/motherboard/cpu1:
    entityID: 3
    entityInstance: 2
    interfaces:
      xyz.openbmc_project.Inventory.Decorator.Revision:
        Version:
          IPMIFruProperty: Custom Field 2
          IPMIFruSection: Board
          IPMIFruValueDelimiter: 58
      xyz.openbmc_project.Inventory.Decorator.Asset:
        BuildDate:
          IPMIFruProperty: Mfg Date
          IPMIFruSection: Board
        PartNumber:
          IPMIFruProperty: Part Number
          IPMIFruSection: Board
        Manufacturer:
          IPMIFruProperty: Manufacturer
          IPMIFruSection: Board
        SerialNumber:
          IPMIFruProperty: Serial Number
          IPMIFruSection: Board
      xyz.openbmc_project.Inventory.Item:
        PrettyName:
          IPMIFruProperty: Name
          IPMIFruSection: Board
2.  If i did not write FRU data to eeprom, it will not response the Manufacturer , SerialNumber and etc after I send curl -b cjar -k https://${bmc}/xyz/openbmc_project/inventory/system?

gx0128, 2018-10-09:
@CPM80 I think Manufacturer ,SerialNumber, Model are properties. Even I did not write FRU data to eeprom, it still response like the example in  https://github.com/openbmc/docs/blob/master/rest-api.md#uploading-images.

CPM80, 2018-10-16:
Path /xyz/openbmc_project/inventory/system in the inventory is filled by phosphor-ipmi-fru. Host firmware reads FRU (or VPD) data from EEPROM and sends it to BMC using IPMI. phosphor-ipmi-fru receives this data, stores it in the /tmp/ipmifruXX directory (where XX is FRU number) and publish it in the inventory.

closed 2018-12-02 by gx0128

229024527, 2020-09-25:
> Path /xyz/openbmc_project/inventory/system in the inventory is filled by phosphor-ipmi-fru. Host firmware reads FRU (or VPD) data from EEPROM and sends it to BMC using IPMI. phosphor-ipmi-fru [...]

Hi can you tell how to generate fru bin file in openbmc