openbmc/openbmc #3399: Failed to reboot BMC

henbin, 2018-10-03:
Hi,
I encountered some issue about 'BMC reboot.'

1. Clone the latest openbmc repo and build romulus machine.
2. Start up the QEMU
sudo ./qemu-system-arm -m 256 -M romulus-bmc -nographic -drive file=flash-romulus,format=raw,if=mtd -net nic,macaddr=C0:FF:EE:00:00:02,model=ftgmac100 -net bridge,id=net0,helper=../qemu-bridge-helper,br=br0

3. Login the shell and execute 'reboot' command.
[image]

4. Reboot process would be stopped.
[image]

Did anyone encounter the issue?

Thanks.

nest1ing, 2018-10-03:
This problem is reproduced at our hardware.
It seems that all firmware with a static flash layout are subject to this problem.

I push our solution to the [gerrit](https://gerrit.openbmc-project.xyz/c/openbmc/meta-phosphor/+/13512).

henbin, 2018-10-03:
> This problem is reproduced at our hardware.
> It seems that all firmware with a static flash layout are subject to this problem.
> 
> I push our solution to the [gerrit](https://gerrit.openbmc-project.xyz/c/openbmc/meta-phosphor/+/13512).

It does work. 
Thanks for your solution.

rfrandse, 2018-10-03:
<https://gerrit.openbmc-project.xyz/13512> meta-phosphor: fix reboot of BMC
Resolves  #3399 Failed to reboot BMC

closed 2018-10-05 by geissonator