openbmc/openbmc #3411: Error building openbmc - phosphor-mapper Function failed: do_qa_configure

anoo1, 2018-10-17:
With the latest openbmc master, `bitbake obmc-phosphor-image` fails with:

```ERROR: phosphor-mapper-1.0+gitAUTOINC+db27894204-r1 do_configure: This autoconf log indicates errors, it looked at host include and/or library paths while determining system capabilities.
Rerun configure task after fixing this.
ERROR: phosphor-mapper-1.0+gitAUTOINC+db27894204-r1 do_configure: Function failed: do_qa_configure
ERROR: Logfile of failure stored in: /tmp/witherspoon/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-mapper/1.0+gitAUTOINC+db27894204-r1/temp/log.do_configure.20653
ERROR: Task (/openbmc/openbmc/meta-phosphor/recipes-phosphor/dbus/phosphor-mapper_git.bb:do_configure) failed with exit code '1'```

The referenced configure file shows:

```
configure: creating ./config.status
config.status: creating Makefile
config.status: creating setup.py
config.status: creating fail-monitor/Makefile
config.status: creating libmapper/config.h
config.status: executing depfiles commands
config.status: executing libtool commands
DEBUG: Shell function do_configure finished
DEBUG: Executing python function do_qa_configure
NOTE: Checking autotools environment for common misconfiguration
ERROR: This autoconf log indicates errors, it looked at host include and/or library paths while determining system capabilities.
Rerun configure task after fixing this.
DEBUG: Python function do_qa_configure finished
ERROR: Function failed: do_qa_configure
```

geissonator, 2018-10-22:
This seems to only affect systems that have a local installation of the boost package. For some reason it's using this local boost installation instead of the one in the yocto recipe.

geissonator, 2018-10-22:
I think we're going to just revert for now.

spinler, 2018-10-22:
The AX_BOOST_BASE macro, at http://git.savannah.gnu.org/gitweb/?p=autoconf-archive.git;a=blob_plain;f=m4/ax_boost_base.m4  is hardcoded to check in /usr for the boost/version.hpp file.  

It vaguely looks like it can look in another path, but I haven't figured that out yet.

philenotfound, 2018-10-23:
I was able to force the boost path and it builds now:
```diff
diff --git a/meta-phosphor/recipes-phosphor/dbus/phosphor-mapper_git.bb b/meta-phosphor/recipes-phosphor/dbus/phosphor-mapper_git.bb
index a67a3cae9..a34ba0346 100644
--- a/meta-phosphor/recipes-phosphor/dbus/phosphor-mapper_git.bb
+++ b/meta-phosphor/recipes-phosphor/dbus/phosphor-mapper_git.bb
@@ -42,6 +42,8 @@ SRCREV = "db27894204a630a675b74aab8021ca5d5d8c45b2"
 
 S = "${WORKDIR}/git"
 
+EXTRA_OECONF = "--with-boost=${RECIPE_SYSROOT}/usr"
+
 python populate_packages_prepend () {
     mapperlibdir = d.getVar("libdir", True)
     do_split_packages(d, mapperlibdir, '^lib(.*)\.so\.*', 'lib%s', 'Phosphor mapper %s library', extra_depends='', allow_links=True)
```

msbarth, 2018-10-23:
I've found a couple options. Either update configure.ac to directly check for the headers used:
```
AC_CHECK_HEADER(boost/algorithm/string/predicate.hpp, ,[AC_MSG_ERROR([Could not find boost/algorithm/string/predicate.hpp])])
AC_CHECK_HEADER(boost/container/flat_map.hpp, ,[AC_MSG_ERROR([Could not find boost/container/flat_map.hpp])])
AC_CHECK_HEADER(boost/container/flat_set.hpp, ,[AC_MSG_ERROR([Could not find boost/container/flat_set.hpp])])
```
or use the recipe update @philenotfound noted along with updating the configure.ac to handle building locally within an SDK or not:
```
AS_IF([test "x$OECORE_TARGET_SYSROOT" != "x"],
    # Use OE SDK boost location
    AC_SUBST([BOOST_ROOT], ["${OECORE_TARGET_SYSROOT}/usr"])
)
AX_BOOST_BASE([1.66],, [AC_MSG_ERROR(["Requires boost."])])
```

gtmills, 2018-10-23:
Hit this..

msbarth, 2018-10-24:
For those interested, the fix is up for review:
https://gerrit.openbmc-project.xyz/#/c/openbmc/phosphor-objmgr/+/14667/

rfrandse, 2018-10-25:
<https://gerrit.openbmc-project.xyz/14667> Check for boost headers used
Resolves:  #3411 Error building openbmc - phosphor-mapper Function failed: do_qa_configure

closed 2018-10-26 by geissonator

aaronw2, 2018-10-29:
The fix only applies to phosphor-objmgr. I'm not sure how to use phosphor-objmgr at this time, but building the main openbmc still fails unless I apply the patch @philenotfound mentioned earlier.

I'm looking into switching from the Facebook OpenBMC to this one for future development.