openbmc/openbmc #3412: Witherspoon: Failed to start OpenPOWER PNOR Minimum Ship Level Verification

gkeishin, 2018-10-18:
Pervassive failure on boot to standby
```
root@witherspoon:~# systemctl status op-pnor-msl.service | cat
● op-pnor-msl.service - OpenPOWER PNOR Minimum Ship Level Verification
   Loaded: loaded (/lib/systemd/system/op-pnor-msl.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2018-10-18 10:56:17 UTC; 1min 21s ago
  Process: 1322 ExecStart=/usr/bin/env openpower-pnor-msl (code=exited, status=127)
 Main PID: 1322 (code=exited, status=127)

Oct 18 10:56:17 witherspoon systemd[1]: Starting OpenPOWER PNOR Minimum Ship Level Verification...
Oct 18 10:56:17 witherspoon openpower-pnor-msl[1322]: env: can't execute 'openpower-pnor-msl': No such file or directory
Oct 18 10:56:17 witherspoon systemd[1]: op-pnor-msl.service: Main process exited, code=exited, status=127/n/a
Oct 18 10:56:17 witherspoon systemd[1]: op-pnor-msl.service: Failed with result 'exit-code'.
Oct 18 10:56:17 witherspoon systemd[1]: Failed to start OpenPOWER PNOR Minimum Ship Level Verification.
root@witherspoon:~#
```

gkeishin, 2018-10-18:
@anoo1 @geissonator  ^^^

anoo1, 2018-11-07:
The service file was merged before the app, fixed with https://gerrit.openbmc-project.xyz/#/c/openbmc/meta-openpower/+/14326/ .

closed 2018-11-07 by anoo1