openbmc/openbmc #3414: Read registers in AST-2500

akashgj, 2018-10-22:
I used the following code to get the value of the control register  'SCU13C' in BMC Aspeed AST-2500.

#include<stdio.h>

#define SCU13C_ADDR 0x1E6E213C
int main()
{
	long reg;
	reg=123;
	reg = *(volatile long *)(SCU13C_ADDR);
	printf("%li",reg);
	return 1;
}

Running the code I am getting Segmentation Fault (Core Dumped). Please someone help to resolve the issue.

gtmills, 2018-10-23:
https://github.com/openbmc/openbmc/issues/3413 looks to be the same issue. Can this be closed?

amboar, 2018-10-24:
@akashgj this will not work a as you're application is running in a virtual address space not the physical address space. You need to either use /dev/mem to access the physical address space from userspace, or write a kernel module and use ioremap() to get an appropriate virtual to physical mapping

closed 2018-10-24 by amboar