openbmc/openbmc #3416: Petitboot can not access after BIOS program

Nick24117, 2018-10-24:
Use version v2.2 on bmc chip to do bios program 
Sometime Petitboot can not be accessed
The petitboot's log is as follow

[  147.057258731,3] MBOX-FLASH: BMC NOT RESPONDING 4 second wait
[  148.057336204,3] MBOX-FLASH: BMC NOT RESPONDING 5 second wait
[  149.057435959,3] MBOX-FLASH: BMC NOT RESPONDING 6 second wait
[  150.057505992,3] MBOX-FLASH: BMC NOT RESPONDING 7 second wait
[  150.159974778,3] MBOX-FLASH: Timeout waiting for BMC
[  150.160036071,3] MBOX-FLASH: Error waiting for BMC
[  150.160084774,3] MBOX-FLASH: Failed to enqueue/send BMC MBOX message
[  150.160086901,7] LPC: Routing irq 4, policy: 1 (r=2)
[  150.160087991,7] LPC: SerIRQ 4 using route 2 targetted at OS
[  150.160089613,7] UART: Enabled as OS pass-through
[  150.160090505,7] SLW: Init chip 0x0
[  150.160092392,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160094485,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160098013,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160100893,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160103074,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160105402,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160109000,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160111000,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160113520,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160116560,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160118684,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160121852,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160124065,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160126422,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160128615,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160131181,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160134239,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160136233,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160142699,6] SLW: Configuring self-restore for HRMOR
[  150.160151349,7] SLW: Init chip 0x8
[  150.160156917,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160159154,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160161340,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160163352,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160165547,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160167972,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160170072,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160172184,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160174428,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160176530,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160178562,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160180784,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160182980,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160184986,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160187119,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160189289,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160191390,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160193490,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[  150.160194957,6] SLW: Configuring self-restore for HRMOR
[  150.160202899,7] CPU idle state device tree init
[  150.160204124,6] SLW: Configuring self-restore for HRMOR
[  150.160211321,6] SLW: Configuring self-restore for NCU_SPEC_BAR
[  150.160220882,6] SLW: Configuring self-restore for P9X_EX_NCU_DARN_BAR
[  150.160232105,7] NVRAM: 'opal-stop-state-disable-mask' not found
[  150.160233426,7] NVRAM stop disable mask: 0
[  150.160234103,6] SLW: Enabling: stop0_lite
[  150.160234911,6] SLW: Enabling: stop0
[  150.160235604,6] SLW: Enabling: stop1
[  150.160236301,6] SLW: Enabling: stop2
[  150.160236996,6] SLW: Enabling: stop4
[  150.160237586,6] SLW: Enabling: stop5
[  150.160240103,7] OCC: Chip 00 Data (0000200ffd8e2000) = 0190015a4b3c0000
[  150.160241683,7] OCC: Chip 08 Data (0000200ffdce2000) = 0190005a4b3c0000
[  150.160243096,5] OCC: All Chip Rdy after 0 ms
[  150.160272646,7] OCC: CPU pstate state device tree init
[  150.160273374,7] OCC: Data (    200ffd8e2000) =  190015a4b3c0000                0
[  150.160275097,7] OCC: Version 90 Min 90 Nom 75 Max 0 Nr States 91
[  150.160289610,7] NVRAM: 'pcie-max-link-speed' not found
[  150.160342527,6] CAPI: Preloading ucode 202d1
[  150.160343459,7] FLASH: Queueing preload of 2/202d1
[  150.160344585,7] FLASH: Queueing preload of 0/0
[  150.160345594,7] FLASH: Queueing preload of 1/0
[  150.161189270,6] VAS: Initialized chip 0
[  150.162013631,6] VAS: Initialized chip 8
[  150.162069300,6] NX RNG[0] pace:2000
[  150.162070253,6] NX0: Crypto at 0x2010000
[  150.162071311,6] NX0: POWER9 nx-crypto not yet supported
[  150.162072394,6] NX0: 842 at 0x2010000
[  150.162073089,7] Found ibm,power9-nx
[  150.162074016,7] NX0: UMAC VAS MMIO BAR, 0x0006019100000000, xcfg 0x20110d4
[  150.162075777,7] NX0: DMA VAS MMIO BAR, 0x0006019100000000, xcfg 0x201105e
[  150.162077974,7] NX0: UMAC SEND WC BAR, 0x0000000039000000, xcfg 0x20110d2
[  150.162080342,7] NX0: Setting UMAC Status Control 0x7000000000000000
[  150.162082218,7] NX0:   DMA 0x00ffc1987f800080
[  150.162086895,7] NX0: Setting UMAC FIFO bar 0x0000000039808280
[  150.162088980,7] NX0: Setting UMAC notify match 0x0fff000003000011
[  150.162094559,7] NX0: Setting UMAC FIFO bar 0x0000000039818280
[  150.162096671,7] NX0: Setting UMAC notify match 0x0fff000003000021
[  150.162099361,7] NX0:   Engine Enable 0x8000000000000067
[  150.162100358,6] NX0: 842 Coprocessor Enabled
[  150.162101043,6] NX0: gzip at 0x2010000
[  150.162103572,7] NX0:   DMA 0x00ffc1987f800080
[  150.162105984,7] NX0:   Engine Enable 0x8000000000000067
[  150.162110774,7] NX0: Setting UMAC FIFO bar 0x0000000039828280
[  150.162113407,7] NX0: Setting UMAC notify match 0x0fff000002000011
[  150.162119194,7] NX0: Setting UMAC FIFO bar 0x0000000039838280
[  150.162121322,7] NX0: Setting UMAC notify match 0x0fff000002000021
[  150.162123250,6] NX0: gzip Coprocessor Enabled
[  150.162130663,6] NX RNG[8] pace:2000
[  150.162131473,6] NX8: Crypto at 0x2010000
[  150.162132443,6] NX8: POWER9 nx-crypto not yet supported
[  150.162133433,6] NX8: 842 at 0x2010000
[  150.162134141,7] Found ibm,power9-nx
[  150.162135160,7] NX8: UMAC VAS MMIO BAR, 0x0006219100000000, xcfg 0x20110d4
[  150.162136757,7] NX8: DMA VAS MMIO BAR, 0x0006219100000000, xcfg 0x201105e
[  150.162138451,7] NX8: UMAC SEND WC BAR, 0x0000200003000000, xcfg 0x20110d2
[  150.162140529,7] NX8: Setting UMAC Status Control 0x7000000000000000
[  150.162142346,7] NX8:   DMA 0x00ffc1987f800080
[  150.162146976,7] NX8: Setting UMAC FIFO bar 0x0000200003808280
[  150.162148894,7] NX8: Setting UMAC notify match 0x0fff000003000031
[  150.162154496,7] NX8: Setting UMAC FIFO bar 0x0000200003818280
[  150.162156269,7] NX8: Setting UMAC notify match 0x0fff000003000041
[  150.162158896,7] NX8:   Engine Enable 0x8000000000000067
[  150.162159811,6] NX8: 842 Coprocessor Enabled
[  150.162160586,6] NX8: gzip at 0x2010000
[  150.162162083,7] NX8:   DMA 0x00ffc1987f800080
[  150.162163761,7] NX8:   Engine Enable 0x8000000000000067
[  150.162168808,7] NX8: Setting UMAC FIFO bar 0x0000200003828280
[  150.162170697,7] NX8: Setting UMAC notify match 0x0fff000002000031
[  150.162176418,7] NX8: Setting UMAC FIFO bar 0x0000200003838280
[  150.162178200,7] NX8: Setting UMAC notify match 0x0fff000002000041
[  150.162180020,6] NX8: gzip Coprocessor Enabled
[  150.164031734,3] FFS: Error -2 retrieving flash info
[  150.164063442,3] FLASH: Can't open ffs handle
[  150.164091147,3] MBOX-FLASH: Failed to enqueue/send BMC MBOX message
[  150.164128263,3] FFS: Error -2 retrieving flash info
[  150.164159912,3] FLASH: Can't open ffs handle
[  150.164185174,3] MBOX-FLASH: Failed to enqueue/send BMC MBOX message
[  150.164222575,3] FFS: Error -2 retrieving flash info
[  150.164249958,3] FLASH: Can't open ffs handle
[  150.164274044,3] MBOX-FLASH: Failed to enqueue/send BMC MBOX message
[  150.164312664,3] FFS: Error -2 retrieving flash info
[  150.164341683,3] FLASH: Can't open ffs handle

geissonator, 2018-10-24:
@amboar ?

amboar, 2018-10-24:
@Nick24117 can you please provide exact versions of the BMC and host firmwares? I need to see the output of `cat /etc/os-release` on the BMC, and `pflash -r /dev/stdout -P VERSION` on the host

mine260309, 2018-10-25:
`v2.2` is old, maybe try the latest master (or latest tag v2.3) and see if the issue is gone.

I was doing mbox stress test a few weeks ago, and the latest OpenBMC runs it well.

Nick24117, 2018-10-25:
@amboar 
BMC version
[image]

HOST
[image]

amboar, 2018-10-25:
@Nick24117 in the future can you please just copy and paste the text into the issue tracker rather than taking screen shots and attaching images. Images make the data unnecessarily hard to work with

Nick24117, 2018-10-26:
@amboar OK, update information again
BMC:
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.2-0"
VERSION_ID="v2.2-0-g6c9010845"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.2-0"
BUILD_ID="v2.2"

Host:
OpenPower-BarreleyeG2-v0.2.2
op-build-v2.1-33-g6329e4f
buildroot-2018.05-5-g378a934
skiboot-v6.1-pc04e3ec
hostboot-876b79a-p98e73bb
occ-5c01b54-p05fc523
linux-4.17.4-openpower1-p922e087
petitboot-1.8.0-pbd8cefa
machine-xml-40bf092-p4363dda
hostboot-binaries-hw070718b.920
capp-ucode-p9-dd2-v4
sbe-fad6732
hcode-hw061618a.920

amboar, 2018-10-26:
@Nick24117 can you please grab the journal off the BMC when this condition arises? This looks like it might be a previously-undiscovered bug. I don't expect the journal will tell us much, but I expect mboxd has crashed.

amboar, 2018-10-26:
Also, please make sure to attach complete logs as files to the issue (from skiboot and the BMC), we're going to need as much context as possible.

Nick24117, 2018-10-29:
@amboar 
We get BMC and host's the information to you 
[BIOS_Flash_Cycles.txt](https://github.com/openbmc/openbmc/files/2524104/BIOS_Flash_Cycles.txt)

[BMC_journal_.zip](https://github.com/openbmc/openbmc/files/2524111/BMC_journal_.zip)

amboar, 2018-11-08:
```
[  129.220869141,3] NVRAM: Partition at offset 0xffff0 extends beyond end of nvram !
[  129.220946689,3] NVRAM: Re-initializing (size: 0x00090000)
[  134.099834459,3] MBOX-FLASH: BMC NOT RESPONDING 4 second wait
[  135.099910010,3] MBOX-FLASH: BMC NOT RESPONDING 5 second wait
[  136.099991512,3] MBOX-FLASH: BMC NOT RESPONDING 6 second wait
[  137.100074983,3] MBOX-FLASH: BMC NOT RESPONDING 7 second wait
[  137.202545802,3] MBOX-FLASH: Timeout waiting for BMC
[  137.202605046,3] MBOX-FLASH: Error waiting for BMC
[  137.202671240,3] MBOX-FLASH: Failed to enqueue/send BMC MBOX message
```
Interesting.

However, there's not really any useful information in the BMC journal, so I'll need to try to reproduce the failure. It looks like I should be able to build the BMC image as I can reach openbmc/openbmc@6c9010845, however it doesn't look like `6329e4f` is a public patch for op-build. Can you point me to an op-build tree with that commit?

Foxconn-bmc-tpa, 2018-11-12:
@amboar 
The pflash stress process is on BMC side
I provide our test process as below :
step 1 : obmcutil poweroff
step 2 : cd /tmp
step 3 : download "pnor file" to /tmp
step 4 : command "pflash -e -p -f pnor file "
step 5 : obmcutil poweron after pflash finish

Nick24117, 2018-11-12:
Hi @amboar 
The pflash stress process is on BMC side
I provide our test process as below :
step 1 : obmcutil poweroff
step 2 : cd /tmp
step 3 : download "pnor file" to /tmp
step 4 : command "pflash -e -p -f pnor file "
step 5 : obmcutil poweron after pflash finish

Nick24117, 2018-11-12:
@mine260309 I try to update code but can not fix the issue.

amboar, 2018-11-15:
I've tried and failed to reproduce this on a barreleye g2 machine running the following firmwares:

```
root@zaius:/tmp# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v01.30.2560-RAX"
VERSION_ID="v01.30.2560-rax-1-gac3f8dcb2"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v01.30.2560-RAX"
BUILD_ID="v01.30.2560-RAX"
```
and
```
root@zaius:/tmp# pflash -P VERSION -r /dev/stdout
Reading to "/dev/stdout" from 0x02c08000..0x02c09000 !
[                                                  ] 0%open-power-zaius-v2.1-33-g6329e4f-dirty
        buildroot-2018.05-5-g378a934e26
        skiboot-v6.1-p3ce4ea3
        hostboot-876b79a-pdc9dc3a
        occ-5c01b54-p66dda04
        linux-4.17.4-openpower1-pe3db803
        petitboot-1.8.0-p048f465
        machine-xml-40bf092-pcb8c73d
        hostboot-binaries-hw070718b.920
        capp-ucode-p9-dd2-v4
        sbe-fad6732
        hcode-hw061618a.920
[==================================================] 100%
```
which was built from
```
andrewrj@w37l:~/src/open-power/op-build$ git describe --tags
v0.2.2-barreleye-g2
```

I'll try again with v2.2-0-g6c9010845 tomorrow.

Nick24117, 2018-11-15:
> `122233`

Nick24117, 2018-11-16:
@amboar OK, you can download and test with bmc image

[image-bmc.zip](https://github.com/openbmc/openbmc/files/2587670/image-bmc.zip)

amboar, 2018-11-17:
Okay, here's what I've discovered. Corresponding to the following skiboot output:

```
[  145.498105338,3] NVRAM: Partition at offset 0xffff0 extends beyond end of nvram !
[  145.498183398,3] NVRAM: Re-initializing (size: 0x00090000)                
[  151.062836331,3] MBOX-FLASH: BMC NOT RESPONDING 4 second wait             
[  152.062923562,3] MBOX-FLASH: BMC NOT RESPONDING 5 second wait         
[  153.063003016,3] MBOX-FLASH: BMC NOT RESPONDING 6 second wait         
[  154.065565671,3] MBOX-FLASH: BMC NOT RESPONDING 7 second wait         
[  154.171298092,3] MBOX-FLASH: Timeout waiting for BMC                  
[  154.171364284,3] MBOX-FLASH: Error waiting for BMC                    
[  154.171429546,3] MBOX-FLASH: Failed to enqueue/send BMC MBOX message  
```
We get the following from mboxd:
```
[ 1542450504.371091732] MBOX cmd: 8
[ 1542450504.371211207] MBOX seq: 10
[ 1542450504.371343328] MBOX arg[0]: 0x00
[ 1542450504.371464500] MBOX arg[1]: 0x00
[ 1542450504.371584217] MBOX arg[2]: 0x00
[ 1542450504.371700662] MBOX arg[3]: 0x00
[ 1542450504.371833551] MBOX arg[4]: 0x00
[ 1542450504.371946803] MBOX arg[5]: 0x00
[ 1542450504.374643126] MBOX arg[6]: 0x00
[ 1542450504.375820945] MBOX arg[7]: 0x00
[ 1542450504.376593591] MBOX arg[8]: 0x00
[ 1542450504.376859450] MBOX arg[9]: 0x00
[ 1542450504.377600783] MBOX arg[10]: 0x00
[ 1542450504.377862884] Received MBOX command: 8
[ 1542450504.378556177] Flush window @ 0x72fb6000 for size 0x00100000 which maps flash @ 0x00000000
[ 1542450504.378890965] Erase flash @ 0x00030000 for 0x000a0000
[ 1542450504.395465066] Erase flash @ 0x00030000 for 0x000a0000
[ 1542450507.256007289] Set flash bytemap @ 0x00030000 for 0x000a0000 to ERASED
[ 1542450507.262518925] Write flash @ 0x00030000 for 0x000a0000 from 0x72fe6000
[ 1542450512.412035815] Set flash bytemap @ 0x00030000 for 0x000a0000 to DIRTY
[ 1542450512.413817835] Translating errno 0: Success
[ 1542450512.420209471] Writing MBOX response:
[ 1542450512.421632704] MBOX cmd: 8
[ 1542450512.423325188] MBOX seq: 10
[ 1542450512.423818522] MBOX arg[0]: 0x00
[ 1542450512.424118360] MBOX arg[1]: 0x00
[ 1542450512.425051936] MBOX arg[2]: 0x00
[ 1542450512.425389310] MBOX arg[3]: 0x00
[ 1542450512.426208057] MBOX arg[4]: 0x00
[ 1542450512.426779289] MBOX arg[5]: 0x00
[ 1542450512.427387532] MBOX arg[6]: 0x00
[ 1542450512.428615289] MBOX arg[7]: 0x00
[ 1542450512.428898522] MBOX arg[8]: 0x00
[ 1542450512.429351249] MBOX arg[9]: 0x00
[ 1542450512.429945714] MBOX arg[10]: 0x00
[ 1542450512.430857996] Writing MBOX response: 1
```
So we exceed the 8s timeout whilst fixing the NVRAM partition.

Nick24117, 2018-11-19:
@amboar 
Yes, BMC can not response to pnor when it is busy.

Phosphor mapper use a lot of cpu resource.
**Following from BMC loading log:**
```
[2018-11-14 06:31:04.400] CPU: 52% usr 47% sys 0% nic 0% idle 0% io 0% irq 0% sirq
[2018-11-14 06:31:04.400] Load average: 3.36 3.31 3.92 2/136 2626
[2018-11-14 06:31:04.400] PID PPID USER STAT VSZ %VSZ %CPU COMMAND

> [2018-11-14 06:31:04.400] 1100 1 root R 16964 2% 37% python /usr/sbin/phosphor-mapper --pat
phosphor mapper grip much resource

[2018-11-14 06:31:04.400] 95 2 root RW 0 0% 15% [kworker/u2:2]
[2018-11-14 06:31:04.400] 2615 2 root SW 0 0% 9% [kworker/u2:1]
[2018-11-14 06:31:04.400] 1122 1 root S 69324 7% 8% mboxd --flash 64M --window-size 1M
[2018-11-14 06:31:04.400] 1094 1 messageb S 5464 1% 8% /usr/bin/dbus-daemon --system --addres
[2018-11-14 06:31:04.400] 1193 1 root D 6312 1% 3% phosphor-hwmon-readd -o /ahb/apb/i2c@1
[2018-11-14 06:31:04.402] 2620 1 root S 6316 1% 3% phosphor-hwmon-readd -o /devices/platf
[2018-11-14 06:31:04.402] 1159 1 root S 6312 1% 3% phosphor-hwmon-readd -o /ahb/apb/i2c@1
[2018-11-14 06:31:04.409] 1187 1 root S 16228 2% 2% python /usr/sbin/system_manager.py
[2018-11-14 06:31:04.409] 2626 1 root S 6316 1% 2% phosphor-hwmon-readd -o /devices/platf
[2018-11-14 06:31:04.409] 1152 1 root S 6312 1% 1% phosphor-hwmon-readd -o /ahb/apb/i2c@1
[2018-11-14 06:31:04.409] 1214 1 root S 6312 1% 1% phosphor-hwmon-readd -o /ahb/apb/i2c@1
[2018-11-14 06:31:04.409] 1202 1 root S 6312 1% 1% phosphor-hwmon-readd -o /iio-hwmon
[2018-11-14 06:31:04.409] 1269 1 root S 3828 0% 1% /usr/sbin/fan_algorithm.exe
[2018-11-14 06:31:04.409] 1743 1731 root R 3236 0% 1% top
[2018-11-14 06:31:04.409] 631 1 root S 13940 1% 1% /lib/systemd/systemd-journald
[2018-11-14 06:31:04.411] 1204 1 root S 6312 1% 1% phosphor-hwmon-readd -o /ahb/apb/i2c@1
[2018-11-14 06:31:04.452] 7 2 root SW 0 0% 1% [ksoftirqd/0]
[2018-11-14 06:31:04.452] 1106 1 root S 15288 2% 0% python /usr/sbin/chassis_control.py
[2018-11-14 06:31:04.452] 1206 1 root D 6312 1% 0% phosphor-hwmon-readd -o /ahb/apb/i2c@1Mem: 156228K used, 794196K free, 15644K shrd, 11328K buff, 53684K cached
[2018-11-14 06:31:09.714] CPU: 66% usr 33% sys 0% nic 0% idle 0% io 0% irq 0% sirq
[2018-11-14 06:31:09.714] Load average: 3.49 3.34 3.92 8/136 2626
[2018-11-14 06:31:09.714] PID PPID USER STAT VSZ %VSZ %CPU COMMAND

> [2018-11-14 06:31:09.714] 1100 1 root R 17220 2% 43% python /usr/sbin/phosphor-mapper --pat
phosphor mapper grip much resource

[2018-11-14 06:31:09.714] 2615 2 root RW 0 0% 14% [kworker/u2:1]
[2018-11-14 06:31:09.714] 95 2 root RW 0 0% 12% [kworker/u2:2]
[2018-11-14 06:31:09.714] 1094 1 messageb S 5464 1% 9% /usr/bin/dbus-daemon --system --addres
[2018-11-14 06:31:09.714] 1159 1 root D 6312 1% 3% phosphor-hwmon-readd -o /ahb/apb/i2c@1
[2018-11-14 06:31:09.714] 1193 1 root S 6312 1% 3% phosphor-hwmon-readd -o /ahb/apb/i2c@1
[2018-11-14 06:31:09.714] 2620 1 root S 6428 1% 3% phosphor-hwmon-readd -o /devices/platf
[2018-11-14 06:31:09.714] 2626 1 root S 6428 1% 3% phosphor-hwmon-readd -o /devices/platf
[2018-11-14 06:31:09.714] 1152 1 root S 6312 1% 1% phosphor-hwmon-readd -o /ahb/apb/i2c@1
[2018-11-14 06:31:09.714] 1214 1 root R 6312 1% 1% phosphor-hwmon-readd -o /ahb/apb/i2c@1
[2018-11-14 06:31:09.714] 1743 1731 root R 3236 0% 1% top
[2018-11-14 06:31:09.718] 1202 1 root S 6312 1% 1% phosphor-hwmon-readd -o /iio-hwmon
[2018-11-14 06:31:09.718] 1269 1 root S 3828 0% 1% /usr/sbin/fan_algorithm.exe
[2018-11-14 06:31:09.718] 7 2 root RW 0 0% 1% [ksoftirqd/0]
[2018-11-14 06:31:09.718] 1253 1 root S 32624 3% 0% power_control.exe 3000 10
[2018-11-14 06:31:09.718] 1106 1 root S 15288 2% 0% python /usr/sbin/chassis_control.py
[2018-11-14 06:31:09.718] 1289 1 root S 7840 1% 0% ipmid
[2018-11-14 06:31:09.718] 1205 1 root S 6312 1% 0% phosphor-hwmon-readd -o /ahb/apb/i2c@1
[2018-11-14 06:31:09.718] 1204 1 root D 6312 1% 0% phosphor-hwmon-readd -o /ahb/apb/i2c@1
[2018-11-14 06:31:09.718] 1206 1 root S 6312 1% 0% phosphor-hwmon-readd -o /ahb/apb/i2c@1Mem: 157252K used, 793172K free, 15644K shrd, 11328K buff, 53684K cached
```

amboar, 2018-11-19:
> Yes, BMC can not response to pnor when it is busy.

Correct. I have been working to get an environment together where I can run `perf timechart record` over the period where the problem occurs to get an exact snapshot of why mboxd can't be scheduled.

> Phosphor mapper use a lot of cpu resource.

Yes, I have noticed that the mapper is consuming a lot of CPU time. Mid May this year I worked on optimising the mapper under some specific circumstances, this may help you, it may not. See this post:

https://github.com/openbmc/openbmc/issues/2860#issuecomment-389038592

The corresponding patches are here:

https://github.com/openbmc/phosphor-objmgr/compare/f9b24bf1b064600bb3f450dbbce475937742c121...005e47adf4498040ca457840e01d74d58b918db0

Nick24117, 2018-11-21:
@amboar 
Our phospor-objmgr commit link is https://github.com/openbmc/phosphor-objmgr/commit/94c92a9a36b6fdd5a5cbf6ef0d583cc488f4318a

We verified the patch which was provided by you
BMC can not open Host service
`root@zaius:~# obmcutil state`
`CurrentBMCState     : xyz.openbmc_project.State.BMC.BMCState.NotReady`
`CurrentPowerState   : xyz.openbmc_project.State.Chassis.PowerState.Off`
`DBus error occurred: The name xyz.openbmc_project.State.Host was not provided by any .service files`
[journalctl.zip](https://github.com/openbmc/openbmc/files/2603757/journalctl.zip)

So we updated the commit to [openbmc/phosphor-objmgr@8e1f4ab...d43e588](https://github.com/openbmc/phosphor-objmgr/tree/8e1f4ab73faf3afd7dffb06a481c74004d43e588) then tested 150 times without fail and keep stressed

a-barrera, 2018-12-06:
@Nick24117 please close the issue now that it has been resolved by increasing the mbox timeout and applying the phospor-objmgr updates.

closed 2018-12-07 by Nick24117