openbmc/openbmc #3417: IPMI sensors for large systems don't fit

AlexanderAmelkin, 2018-10-24:
The IPMI specification was written in the very old days and only reserves a single unsigned byte for a sensor ID. That allows for no more than 256 IPMI sensors. That poses a problem for large systems like YADRO VESNIN having 128 DIMM modules, 16 membufs and 4 CPUs with 12 cores each. Just the discrete sensors used to set the 'Functional' attributes for those entities take 192 sensor IDs. There remains no space for DIMM, membuf and core temperature sensors (another 192 entries). Let alone other sensors like FANs, inlet and outlet temperatures, various voltages, etc.

This issue is a request for ideas on how to overcome the 256 sensor ID limitation and not lose the 'Functional' attributes.

One of the possible solutions is to move the 'Functional' reporting from IPMI sensors to an OEM command. We could leave some combined discrete IPMI sensors like 'DIMMs_func', 'cores_func', etc. that, when false, would indicate that an OEM command must be issued to obtain the detailed info.

Any other ideas are welcome.

adx-git, 2018-10-24:
I am afraid this will end up in a violation of the IPMI spec. anyway.
Maybe we could use the Sensor Owner ID as System Software ID to represent more than 256 sensors, each different ID would give additional 256 sensors, but I am not 100% sure how other software like the IPMITool would behave.
On bladed systems with a Shelf Manager, the Shelf Manager shows blade sensors using the Sensor Owner ID, where the ID is the I2C slave address and bit[0] is '0'.
See Table 5-4 in the IPMI spec about the potential System Software IDs

peanutsandwich, 2018-10-26:
One solution to this that I have used before is to split the sensors across LUNs, meaning you can in theory have up to 1024 sensors.

I don't think there is much handling of LUNs done in the current IPMI daemons for OpenBMC, but that is something you might consider investigating?

AlexanderAmelkin, 2018-10-29:
@peanutsandwich, LUNs were my first thought too, but it appears that normal IPMI over LAN and BT requests don't have any way to pass the LUN. LUNs are only supported for bridged requests to IPMB or on KCS and SMIC interfaces.

Even the thesaurus in IPMB spec says that LUNs are only for bridging to I2C:
"Logical Unit Number. In the context of the Intelligent Platform Management Bus protocol, this
is a sub-address that allows messages to be routed to different ‘logical units’ that reside
behind the same I 2 C slave address."

Looking at the "IPMI LAN Message Format" section it appears that LUNs indeed can be passed in a LAN command request but so far this is seems to be supported by neither OpenBMC, nor ipmitool, nor even ipmiutil. I could add support for LUNs over LAN into ipmitool, but it would take a long time to get into major distributions so most admins would be left to deal with those LUNs using raw commands, and I'd hate that if I were them.

edtanous, 2018-11-01:
+1 for the splitting sensors across LUNs, although only 3 LUNs are usable for that purpose, one of the LUNs is reserved in the specification.

@AlexanderAmelkin I'm not sure what you mean when you say that LUNs aren't supported in ipmitool.  They're supported today using the -l parameter.  In theory the only changes required would be to add support to the openbmc infrastructure.

excerpt from "ipmitool -h" from v 1.8.18
"-l lun         Set destination lun for raw commands"

AlexanderAmelkin, 2018-11-02:
Besides, I think that 3 LUNs, even if supported, won't be enough for 16-socket systems.

AlexanderAmelkin, 2018-12-19:
JFYI, we're now working on support for LUNs to allow for splitting the sensor set for VESNIN across the 3 available LUNs.

edtanous, 2018-12-19:
Out of curiosity, why haven't you looked into redfish?  It's specifically designed to allow theoretically infinite sensor counts, and provides a hierarchy for reading them in (as a flat list of 1000 sensors isn't very usable).  Did you find some critical flaw in it?

AlexanderAmelkin, 2018-12-20:
@edtanous, yes, there is a critical flaw: RedFish isn't available for inband communications. You can't ask host's BMC via LAN from the host itself when the share the same NC-SI PHY and aren't externally routed to each other. That is at least so with the Broadcom NC-SI chip that we inherited from Palmetto reference design.

edtanous, 2018-12-22:
"RedFish isn't available for inband communications."
I'm not quite following.  Redfish has a whole spec devoted to supporting in-band channels.
https://www.dmtf.org/sites/default/files/standards/documents/DSP0270_1.0.0.pdf

"You can't ask host's BMC via LAN from the host itself when the share the same NC-SI PHY and aren't externally routed to each other."
I may not have understood your sentence correctly, but most modern NICs have mechanisms for exposing "virtual" in-band communication channels for exactly this use case, over NCSI.  I haven't looked into the Broadcom NIC that Palmetto uses, so I'll take your word for it that's not possible.  Even if there's no way to do it through the NIC, Redfish allows for supporting inband over a virtual USB to ethernet NIC, using the existing USB hub used generally for KVM, and bypassing the network hardware altogether.  Would that work for Palmetto?

AlexanderAmelkin, 2018-12-24:
Saying that RedFish isn't available for inband I meant that there is no, to my knowledge, a ready working solution in OpenBMC _and_ modern Linux distributions comparable to `/dev/ipmi0` and `ipmitool`. Please correct me if I'm wrong.

Using a virtual USB LAN may be a great solution for inband communications. We'll try it, thanks for the idea.

However, as long as IPMI is still the transport between PNOR and OpenBMC used to send device presence and 'functionality' data, the limit on the number of sensors is still an issue. Besides, lots of customer scripts are based on `ipmitool`.

edtanous, 2019-01-23:
We're working through host-redfish investigations now.  It's looking like it's just a new network interface, with the BMC side libraries being essentially unchanged.  It certainly won't be as prevalent as ipmitool, but it should be close.
Not debating the prevalence of IPMI, and I certainly recognize the problem;  We hit the same problem on 2 socket platforms.  The solution we're going with for the moment is redfish, with IPMI returning a abbreviated sensor list, redfish returning the full sensor list.

bradbishop, 2019-01-23:
fwiw I don't think in-band redfish will ever be used to transport device presence and functionality data to the bmc on OpenPOWER systems because of our host firmware design.  It is simply not feasible to have ip and http stacks in there.

Instead, we are headed to mctp+pldm to transport between bmc/hostboot (or skiboot).  There has been much activity on the mailing list about this.

bradbishop, 2019-01-23:
@jk-ozlabs @dkodihal for your awareness...

AlexanderAmelkin, 2019-07-26:
We still need a solution for this. So far we're using an internal solution employing LUNs. We're planning to upstream it later.

yuennancy, 2019-08-30:
Just wondering where you are with your LUN solution?  Do you have anything you can share? We are running into the similar issues.

And a question about LUNs, does it actually give a separate sensor ID space?  It's been a while since I read the IPMI spec but I was under the impression that LUNs were for routing to satellite controllers but the sensor ID was across all the controllers.  I'm happy to be wrong, hoping someone more well versed in IPMI can clear up my confusion.

AlexanderAmelkin, 2019-09-02:
@yuennancy, we have implemented that internally but aren't yet ready to share (we will as soon as the patch is more or less ready for that). Implementing LUNs does indeed give you 2 extra sensor spaces. We've moved all discrete 'presence' sensors into LUN 1 (patched hostboot for that) because they are easy to read with ipmitool raw commands. ipmitool does not so far support LUNs for sensors, and reading analog sensors with raw commands is painful, so we left them in LUN 0.

AlexanderAmelkin, 2019-10-03:
@yuennancy, we've also found a bug in skiboot that didn't let our implementation work properly from host because it turned out that skiboot forcefully sets LUN to 0 in all ipmi messages sent via OPAL.
FYR, https://github.com/YADRO-KNS/op-build/pull/29

AlexanderAmelkin, 2019-10-03:
By the way, LUNs are a part of SDR info, so if some sensors are moved to a LUN and that is properly reflected in SDR, then `ipmitool` shows them correctly without any additional modification of the tool.

yuennancy, 2020-01-14:
@AlexanderAmelkin just curious what the status of your LUN support?

Krellan, 2020-08-07:
I noticed this patch recently was submitted: https://github.com/openbmc/intel-ipmi-oem/commit/308c3a8bff3d667b43f997856bd884dc15a78ded

I haven't tried it yet, but it looks promising. It implements the idea of LUN's, bumping up the limit from 256 to 768, as there's 3 usable LUN numbers.

closed 2021-07-08 by stale[bot]