openbmc/openbmc #3424: NSLCD taking 48MB of system ram

edtanous, 2018-11-01:
A recent top run on my system shows NSLCD taking a lot of system ram, approximately 11% of total on a 512 MB ram system, VSZ of 48,000kb.  Given that OpenBMC needs to target systems with far less ram in a lot of cases, this usage is unacceptable, especially for a component that is included by default.  The quanta platform only has 128MB of ram total, so this would account for 48% used simply for ldap.

This needs to be improved.  My initial gut feeling is to disable ldap in master for the time being until this can be resolved, but I'm open if there are easier less invasive methods that would reduce the system load.

As a greater issue, we need a better way of vetting these types of components when they are enabled in OpenBMC as a whole.  Ideally this would've been found before the initial code was merged, but to do that we need to define guidelines for what kinds of system resource usage is considered acceptable.

geissonator, 2018-11-01:
FYI @dkodihal @ratagupt

bradbishop, 2018-11-01:
If I'm reading into your words too much let me know but it seems like you suggest we have an image recipe with some kind of lowest common denominator feature set?

Do we need more image recipes?  That would be my suggestion.  There are two somewhat conflicting needs to be served by image recipes I think:

1 - The project itself needs a way to test _all_ the functionality available.
2 - End users (system integrators) need to selectively enable components that they want (and can fit within the constraints of their systems).

LCD feature set sounds fine but we still need an image with everything so that when pull requests come in so we can test everything.  Right?

Would it be fair to call this a feature request?  For a new image recipe with a minimal set of functions?

dkodihal, 2018-11-02:
nslcd and other ldap related daemons can be removed from the image by turning off a distro feature (ldap). Does that help here? I don't recall if we concluded on opt-in vs opt-out.

As far as the reported VSZ goes (47-48 Mbytes), that seems to tally with what nslcd is known to consume. In other words, I don't think there's something the in the nslcd.conf that we can turn off to easily cut down this. There's potentially some cache related config that I'm looking at, but next we'd have to look at the nslcd source.

dkodihal, 2018-11-02:
I've posted about this to the nslcd mailing list as well, FYI.

bradbishop, 2018-11-02:
https://lists.arthurdejong.org/nss-pam-ldapd-users/2018/msg00065.html

dkodihal, 2018-11-05:
Need to try out the suggestion(s) by nslcd's author - https://lists.arthurdejong.org/nss-pam-ldapd-users/2018/msg00066.html.

edtanous, 2018-11-06:
That looks like it could help a lot.  Assuming a lot of the data-structures are per-thread, that should get us down to reasonable levels.

dkodihal, 2018-11-07:
I used the threads (*threads NUM*) runtime config to set the number of threads to two (default is five), and with that nslcd takes up about 24 MB RAM. Quick tests with this look fine, but I need to do some more detailed verification. 24 MB is probably still more than a desirable value here, but it's at least a start.

The author recommends at least two threads, because an application could be performing two requests simultaneous (e.g. go over all users and find groups per user).

dkodihal, 2018-11-14:
Single thread does give us more savings (11 - 12 MB consumption). I'm still not sure it gets slower, didn't seem to be the case with some quick tests, but I need to make sure by running more tests.

closed 2019-06-15 by stale[bot]