openbmc/openbmc #3426: mrw-api-native: LIC_FILES_CHKSUM points to an invalid file

geissonator, 2018-11-09:
Our build team builds everything from scratch and is hitting the following error:

```
ERROR: mrw-api-native-1.0-r1 do_populate_lic: QA Issue: mrw-api-native: LIC_FILES_CHKSUM points to an invalid file: /home/bmcbuild/releases/obmc910/Builds/ob910.1845.20181108h/cf6e4e0e1ceefeb03cd20d1dfd3a90b7922f26e9/build/tmp/work/x86_64-linux/mrw-api-native/1.0-r1/LICENSE [license-checksum]
Build failed
```

I assume our CI is not seeing it due to bitbake caching.

The earliest seen was with 2879d326eb675a9e49447739918589c8814f015d (although the builds have wrapped so probably earlier then this commit) and also seen in the most recent 9b5277ed7a12fe79cf805eaa09e3c2c7b53c587b

geissonator, 2018-11-09:
Any thoughts @pstrinkle ?

bradbishop, 2018-11-09:
fix here: https://gerrit.openbmc-project.xyz/#/c/openbmc/meta-phosphor/+/15652/

closed 2018-11-09 by geissonator