openbmc/openbmc #3429: romulus: The FSI master scan did not create a hub scan file

geissonator, 2018-11-15:
Intermittently we're seeing this failure on Romulus power on. The theory is it's related to the new coldfire enablement on Romulus AST2500. Disabling the coldfire engine makes this go away. The symptom is the following in the journal on power on:

```
Nov 12 03:23:16 romulus openpower-proc-control[1790]: The FSI master scan did not create a hub scan file
Nov 12 03:23:16 romulus openpower-proc-control[1790]: Failed while detecting the master processor
Nov 12 03:23:17 romulus systemd[1]: Failed to start Scan FSI devices.
```

A workaround is to do the following:
```
echo gpio-fsi > /sys/bus/platform/drivers/fsi-master-acf/unbind && echo gpio-fsi > /sys/bus/platform/drivers/fsi-master-acf/bind
```

Ben H. was thinking some retries may help in the FSI driver.

Per @mine260309 "As my test, the openbmc master on Sep.26 is OK, and a later build Oct.9 has such issue."

geissonator, 2018-11-15:
FYI @mine260309 @shenki

shenki, 2019-04-30:
This was resolved some time ago.

closed 2019-04-30 by shenki