openbmc/openbmc #3432: Unable to set new root password

rahulmah, 2018-11-21:
Unable to set new root password with latest master. SetPassword method seems to be missing. 

```
bash-4.1$ curl -c cjar -b cjar -k H "Content-Type: application/json" -X POST https://${BMC_IP}/xyz/openbmc_project/user/root/action/SetPassword -d '{"data": "abc123" }'
{
  "data": {
    "description": "The specified method cannot be found: 'SetPassword'"
  },
  "message": "404 Not Found",
  "status": "error"
}
```
```bash-4.1$ curl -c cjar -b cjar -k X GET  https://$BMC_IP/xyz/openbmc_project/user/enumerate                                                                   {
  "data": {
    "/xyz/openbmc_project/user/root": {
      "UserEnabled": true,
      "UserGroups": [
        "ipmi",
        "redfish",
        "ssh",
        "web"
      ],
      "UserLockedForFailedAttempt": false,
      "UserPrivilege": "priv-admin"
    }
  },
  "message": "200 OK",
  "status": "ok"
}bash-4.1$
```

System Configuration:
BMC : v2.4-14-g528c1fe
PNOR : OP9-v2.0.10-2.6

edtanous, 2018-12-22:
I haven't verified, but based on your password, I'm going to guess you're getting rejected by cracklib.  Cracklib attempts to disallow "insecure" passwords based on a common password dictionary.

rthomaiy, 2018-12-24:
Yes, SetPassword method is removed from D-Bus method (phosphor-user-manager doesn't implement SetPassword method for security purpose). The correct way to SetPassword is through rest server implementing password change through pam_chauthotok() api call.

gtmills, 2019-01-09:
https://gerrit.openbmc-project.xyz/#/c/openbmc/phosphor-rest-server/+/13976/

xiuzhi123, 2019-03-08:
The openbmc cannot login webpage when patching the phosphor-rest-server/+/13976 to openbmc  v2.6 source

rahulmah, 2019-07-10:
We did not face the problem in later build to it seem to have fixed.

closed 2019-07-10 by rahulmah