openbmc/openbmc #3435: xyz.openbmc_project.Hwmon@.service: Failed to resolve unit specifiers on %f

cmao0914, 2018-11-23:
Hi all,

I am using openbmc for our hardware.
I want to add a sensor and expose it to DBus but failed.
I refer to the following links to add a sensor but I don't know why hwmon service is not running(why invalid argument). 
[issue#3063](https://github.com/openbmc/openbmc/issues/3063)
[issue#3074](https://github.com/openbmc/openbmc/issues/3074)
[doc#sensor-architecture.md](https://github.com/openbmc/docs/blob/master/sensor-architecture.md#other-interfaces)
Can someone give me some clues?

Error Log:
>systemd[1]: /lib/systemd/system/xyz.openbmc_project.Hwmon@.service:8: Failed to resolve unit specifiers on %f: Invalid argument
systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-i2c@1e78a000-i2c-bus@1c0-tmp421@4e.service: Unit configuration has fatal error, unit will not be started
systemd-udevd[812]: Process '/usr/bin/start_hwmon.sh start /devices/platform/ahb/ahb:apb/ahb:apb:i2c@1e78a000/1e78a1c0.i2c-bus/i2c-6/6-004e/hwmon/hwmon1 /ahb/apb/i2c@1e78a000/i2c-bus@1c0/tmp421@4e' failed with exit code 1

My BMC version:
>#cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.4-42"
VERSION_ID="v2.4-42-gf941a11"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.4-42"
BUILD_ID="v2.4"

My device tree:
>&i2c6 {
        status = "okay";
        tmp421@4e {
                compatible = "ti,tmp421";
                reg = <0x4e>;
        };
        eeprom@54 {
                compatible = "atmel,24c64";
                reg = <0x54>;
                pagesize = <32>;
        };
};

The device is created successfully:
> #cat /sys/firmware/devicetree/base/ahb/apb/i2c\@1e78a000/i2c-bus\@1c0/tmp421\@4e/compatible
ti,tmp421
#cat /sys/firmware/devicetree/base/ahb/apb/i2c\@1e78a000/i2c-bus\@1c0/tmp421\@4e/name
tmp421

The hwmon1 point to tmp421:
>#cat /sys/class/hwmon/hwmon1/of_node/compatible
ti,tmp421
#cat /sys/class/hwmon/hwmon1/of_node/name
tmp421
#hexdump /sys/class/hwmon/hwmon1/of_node/reg
0000000 0000 4e00

tmp421@4e.conf is created based on the device tree structure:
> recipes-phosphor/sensors/phosphor-hwmon%/obmc/hwmon/ahb/apb/i2c@1e78a000/i2c-bus@1c0/tmp421@4e.conf

and the conf content is:
> LABEL_temp1=ambient
WARNLO_temp1=10000
WARNHI_temp1=40000

The phosphor-hwmon%.bbappend file:
>FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"
CHIPS = "i2c@1e78a000/i2c-bus@1c0/tmp421@4e"
ITEMSFMT = "ahb/apb/{0}.conf"
ITEMS = "${@compose_list(d, 'ITEMSFMT', 'CHIPS')}"
ENVS = "obmc/hwmon/{0}"
SYSTEMD_ENVIRONMENT_FILE_${PN} += "${@compose_list(d, 'ENVS', 'ITEMS')}"

Also, I can find the conf file in /etc/default/obmc. I think this means that the phosphor-hwmon%.bbappend setting is correct.
>#find /etc/default/ -name "\*tmp421\*"
/etc/default/obmc/hwmon/ahb/apb/i2c@1e78a000/i2c-bus@1c0/tmp421@4e.conf
#cat /etc/default/obmc/hwmon/ahb/apb/i2c@1e78a000/i2c-bus@1c0/tmp421@4e.conf
LABEL_temp1=ambient
WARNLO_temp1=10000
WARNHI_temp1=40000

Run udevadm and no errors were found
>#udevadm test /sys/class/hwmon/hwmon1
calling: test
version 237
This program is for debugging only, it does not run any program
specified by a RUN key. It may show incorrect results, because
some values may be different, or not available at a simulation run.
Load module index
Parsed configuration file /lib/systemd/network/99-default.link
Created link configuration context.
Reading rules file: /lib/udev/rules.d/50-firmware.rules
Reading rules file: /lib/udev/rules.d/50-udev-default.rules
Reading rules file: /lib/udev/rules.d/60-block.rules
Reading rules file: /lib/udev/rules.d/60-cdrom_id.rules
Reading rules file: /lib/udev/rules.d/60-drm.rules
Reading rules file: /lib/udev/rules.d/60-evdev.rules
Reading rules file: /lib/udev/rules.d/60-input-id.rules
Reading rules file: /lib/udev/rules.d/60-persistent-alsa.rules
Reading rules file: /lib/udev/rules.d/60-persistent-input.rules
Reading rules file: /lib/udev/rules.d/60-persistent-storage-tape.rules
Reading rules file: /lib/udev/rules.d/60-persistent-storage.rules
Reading rules file: /lib/udev/rules.d/60-persistent-v4l.rules
Reading rules file: /lib/udev/rules.d/60-sensor.rules
Reading rules file: /lib/udev/rules.d/60-serial.rules
Reading rules file: /lib/udev/rules.d/61-aspeed-vuart.rules
Reading rules file: /lib/udev/rules.d/64-btrfs.rules
Reading rules file: /lib/udev/rules.d/70-hwmon.rules
Reading rules file: /lib/udev/rules.d/70-iio.rules
Reading rules file: /lib/udev/rules.d/70-joystick.rules
Reading rules file: /lib/udev/rules.d/70-leds.rules
Reading rules file: /lib/udev/rules.d/70-mouse.rules
Reading rules file: /lib/udev/rules.d/70-touchpad.rules
Reading rules file: /lib/udev/rules.d/75-net-description.rules
Reading rules file: /lib/udev/rules.d/75-probe_mtd.rules
Reading rules file: /lib/udev/rules.d/76-aspeed-mtd-partitions.rules
Reading rules file: /lib/udev/rules.d/78-sound-card.rules
Reading rules file: /lib/udev/rules.d/80-drivers.rules
Reading rules file: /lib/udev/rules.d/80-net-setup-link.rules
Reading rules file: /lib/udev/rules.d/90-vconsole.rules
Reading rules file: /lib/udev/rules.d/99-systemd.rules
Reading rules file: /etc/udev/rules.d/touchscreen.rules
rules contain 24576 bytes tokens (2048 * 12 bytes), 9684 bytes strings
1289 strings (16787 bytes), 820 de-duplicated (7573 bytes), 470 trie nodes used
RUN '/usr/bin/start_hwmon.sh start %p %E{OF_FULLNAME}' /lib/udev/rules.d/70-hwmon.rules:1
ACTION=add
DEVPATH=/devices/platform/ahb/ahb:apb/ahb:apb:i2c@1e78a000/1e78a1c0.i2c-bus/i2c-6/6-004e/hwmon/hwmon1
OF_COMPATIBLE_0=ti,tmp421
OF_COMPATIBLE_N=1
OF_FULLNAME=/ahb/apb/i2c@1e78a000/i2c-bus@1c0/tmp421@4e
OF_NAME=tmp421
SUBSYSTEM=hwmon
USEC_INITIALIZED=4635153441
run: '/usr/bin/start_hwmon.sh start /devices/platform/ahb/ahb:apb/ahb:apb:i2c@1e78a000/1e78a1c0.i2c-bus/i2c-6/6-004e/hwmon/hwmon1 /ahb/apb/i2c@1e78a000/i2c-bus@1c0/tmp421@4e'
Unload module index
Unloaded link configuration context.

Although the hwmon service is not started, I can manually run phosphor-hwmon-readd successfully. [reference link](https://github.com/openbmc/openbmc/issues/3074#issuecomment-380372692)

>export LABEL_temp1=ambient
phosphor-hwmon-readd -p /sys/class/hwmon/hwmon1

>curl b cjar -k https://192.168.88.176/xyz/openbmc_project/sensors/temperature/ambient
{
  "data": {
    "MaxValue": 0,
    "MinValue": 0,
    "Scale": -3,
    "Unit": "xyz.openbmc_project.Sensor.Value.Unit.DegreesC",
    "Value": 48000
  },
  "message": "200 OK",
  "status": "ok"
}

Is there any setting that I forgot to set?
Thank you!

mine260309, 2018-11-26:
The udev test shows it is good to start the hwmon service:

> RUN '/usr/bin/start_hwmon.sh start %p %E{OF_FULLNAME}' /lib/udev/rules.d/70-hwmon.rules:1

So hwmon is expected to start with the above command.
Could you check the journal log to see what happens when the service is started?

cmao0914, 2018-11-26:
Hi @mine260309 

Thanks for your reply.

The journal log about hwmon is only:
[image]
In addition to these, I can't find other logs about hwmon. [boot_log_fyi.txt](https://github.com/openbmc/openbmc/files/2614140/log.txt)

Here are the content of **xyz.openbmc_project.Hwmon@.service**, **70-hwmon.rules**, and **tmp421.conf**
I have not modified this, so there should be no problem.
[image]
[image]
[image]

I've also printed the variable of **path**, **action**, **of_fullname**, and **devpath** in **/usr/bin/start_hwmon.sh** and no abnormal message is found
[image]
[image]

The below log is printed by **systemctl status xyz.openbmc_project.Hwmon@-ahb-apb-i2c@1e78a000-i2c-bus@1c0-tmp421@4e.service**
>● xyz.openbmc_project.Hwmon@-ahb-apb-i2c@1e78a000-i2c-bus@1c0-tmp421@4e.service - Phosphor Hwmon Poller
   Loaded: error (Reason: Exec format error)
   Active: inactive (dead)
Nov 26 05:32:22 test systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-i2c@1e78a000-i2c-bus@1c0-tmp421@4e.service: Failed to load configuration: Exec format error
Nov 26 05:32:22 test systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-i2c@1e78a000-i2c-bus@1c0-tmp421@4e.service: Collecting.
Nov 26 05:39:35 test systemd[1]: /lib/systemd/system/xyz.openbmc_project.Hwmon@.service:8: Failed to resolve unit specifiers on %f: Invalid argument
Nov 26 05:39:35 test systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-i2c@1e78a000-i2c-bus@1c0-tmp421@4e.service: Unit configuration has fatal error, unit will not be started.
Nov 26 05:39:35 test systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-i2c@1e78a000-i2c-bus@1c0-tmp421@4e.service: Failed to load configuration: Exec format error
Nov 26 05:39:35 test systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-i2c@1e78a000-i2c-bus@1c0-tmp421@4e.service: Collecting.
Nov 26 05:40:21 test systemd[1]: /lib/systemd/system/xyz.openbmc_project.Hwmon@.service:8: Failed to resolve unit specifiers on %f: Invalid argument
Nov 26 05:40:21 test systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-i2c@1e78a000-i2c-bus@1c0-tmp421@4e.service: Unit configuration has fatal error, unit will not be started.
Nov 26 05:40:21 test systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-i2c@1e78a000-i2c-bus@1c0-tmp421@4e.service: Failed to load configuration: Exec format error
Nov 26 05:40:21 test systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-i2c@1e78a000-i2c-bus@1c0-tmp421@4e.service: Collecting.

For now, I can know there are two error messages as below:
1. Failed to load configuration: Exec format error
2. Failed to resolve unit specifiers on %f: Invalid argument

Now, I will try to check what %f is replaced with.

Do you have any suggestions? 

Thank you!

By the way, does hwmon depend on any other service? I'm afraid that this is caused by dependency.

nest1ing, 2018-11-26:
We had the same problem with start of `phosphor-hwmon` at our VESNIN server and I posted our workaround to [the gerrit](https://gerrit.openbmc-project.xyz/c/openbmc/meta-phosphor/+/16119).

rfrandse, 2018-11-26:
<https://gerrit.openbmc-project.xyz/16119> Fix the starting of phosphor-hwmon
Partially resolves:  #3435 xyz.openbmc_project.Hwmon@.service: Failed to resolve unit specifiers on %f

cmao0914, 2018-11-27:
Hi @rfrandse , @nest1ing , and @mine260309 ,

Thank you for helping me clarify this problem.

After making changes based on the link you provided, it can work!
Now, i can get the sensor info via redfish.
[image]

closed 2018-11-27 by cmao0914

AloneWarrior11011, 2025-04-22:
> The udev test shows it is good to start the hwmon service:
> 
> > RUN '/usr/bin/start_hwmon.sh start %p %E{OF_FULLNAME}' /lib/udev/rules.d/70-hwmon.rules:1
> 
> So hwmon is expected to start with the above command. Could you check the journal log to see what happens when the service is started?

```
root@raspberrypi3:~# systemctl status   xyz.openbmc_project.Hwmon@-soc-i2c\x407e804000-tmp75\x404d.service   
* xyz.openbmc_project.Hwmon@-soc-i2cx407e804000-tmp75x404d.service - Phosphor Hwmon Poller
     Loaded: loaded (/usr/lib/systemd/system/xyz.openbmc_project.Hwmon@.service; static)
     Active: inactive (dead)

Apr 22 07:25:19 raspberrypi3 systemd[1]: Phosphor Hwmon Poller was skipped because of an unmet co...conf).
Hint: Some lines were ellipsized, use -l to show in full.

```
but conf file is located under /etc/default/obmc/hwmon/soc/i2c@7e804000/tmp75@4d.conf 
but phosphor-hwmon-readd expects it from 
/etc/default/obmc/hwmon/soc/i2cx407e804000/tmp75x404d.conf 

I tried alot but cannot fix the problem do help me even if the sensor got exposed on dbus : 

even able to introspect dbus object : 

```busctl introspect xyz.openbmc_project.Hwmon-f6ddcb1b36b4d684fe6d19c36af10f59797735dcf90a85cde8badbd4b618c356.Hwmon1 /xyz/openbmc_project/sensors/temperature/tmp75_sensor
NAME                                                  TYPE      SIGNATURE RESULT/VALUE                             FLAGS
org.freedesktop.DBus.Introspectable                   interface -         -                                        -
.Introspect                                           method    -         s                                        -
org.freedesktop.DBus.Peer                             interface -         -                                        -
.GetMachineId                                         method    -         s                                        -
.Ping                                                 method    -         -                                        -
org.freedesktop.DBus.Properties                       interface -         -                                        -
.Get                                                  method    ss        v                                        -
.GetAll                                               method    s         a{sv}                                    -
.Set                                                  method    ssv       -                                        -
.PropertiesChanged                                    signal    sa{sv}as  -                                        -
xyz.openbmc_project.Sensor.Threshold.Critical         interface -         -                                        -
.CriticalAlarmHigh                                    property  b         false                                    emits-change writable
.CriticalAlarmLow                                     property  b         false                                    emits-change writable
.CriticalHigh                                         property  d         40                                       emits-change writable
.CriticalLow                                          property  d         0                                        emits-change writable
.CriticalHighAlarmAsserted                            signal    d         -                                        -
.CriticalHighAlarmDeasserted                          signal    d         -                                        -
.CriticalLowAlarmAsserted                             signal    d         -                                        -
.CriticalLowAlarmDeasserted                           signal    d         -                                        -
xyz.openbmc_project.Sensor.Threshold.Warning          interface -         -                                        -
.WarningAlarmHigh                                     property  b         true                                     emits-change writable
.WarningAlarmLow                                      property  b         false                                    emits-change writable
.WarningHigh                                          property  d         10                                       emits-change writable
.WarningLow                                           property  d         0                                        emits-change writable
.WarningHighAlarmAsserted                             signal    d         -                                        -
.WarningHighAlarmDeasserted                           signal    d         -                                        -
.WarningLowAlarmAsserted                              signal    d         -                                        -
.WarningLowAlarmDeasserted                            signal    d         -                                        -
xyz.openbmc_project.Sensor.Value                      interface -         -                                        -
.MaxValue                                             property  d         inf                                      emits-change writable
.MinValue                                             property  d         -inf                                     emits-change writable
.Unit                                                 property  s         "xyz.openbmc_project.Sensor.Value.Uni... emits-change writable
.Value                                                property  d         26.125                                   emits-change writable
xyz.openbmc_project.State.Decorator.OperationalStatus interface -         -                                        -
.Functional                                           property  b         true                                     emits-change writable

```