openbmc/openbmc #3441: how to power on system from BMC

zhuifeng091, 2018-11-29:
the dts file can configure gpio, but power_up_soc gpio need first pull down, sleep 1s then pull up, how to configure the pull down and pull up GPIO?

gtmills, 2018-12-03:
Hi @zhuifeng091, this is a good question for the mailing list, as it's not a bug or feature request. You can sign yourself up at https://lists.ozlabs.org/listinfo/openbmc

Thanks,
Gunnar

From https://github.com/openbmc/docs/blob/master/REST-cheatsheet.md
Host power on:
```
curl -c cjar -b cjar -k -H "Content-Type: application/json" -d '{"data": "xyz.openbmc_project.State.Host.Transition.On"}
```

gtmills, 2018-12-03:
You could also use `obmcutil poweron` from the BMC command line. 
```
root@witherspoon:~# obmcutil --help
usage: obmcutil [-h] [--verbose] [--wait]
                {bmcstate,bootprogress,chassiskill,chassisoff,chassison,chassisstate,hoststate,power,poweroff,poweron,state,status}

positional arguments:
  {bmcstate,bootprogress,chassiskill,chassisoff,chassison,chassisstate,hoststate,power,poweroff,poweron,state,status}

optional arguments:
  -h, --help            show this help message and exit
  --verbose, -v         Verbose output
  --wait, -w            Block until the state transition succeeds or fails
```

closed 2018-12-03 by gtmills