openbmc/openbmc #3443: witherspoon phosphor-fan-control: Uncaught DBus service lookup failure exception

gkeishin, 2018-11-29:
In the latest Master, during power on, and is a persistent failure
```
Nov 29 12:00:35 witherspoon phosphor-fan-control[1637]: Uncaught DBus service lookup failure exception
Nov 29 12:00:35 witherspoon systemd[1]: Started Scan FSI devices.
Nov 29 12:00:35 witherspoon systemd[1]: phosphor-fan-control-init@0.service: Main process exited, code=exited, status=1/FAILURE
Nov 29 12:00:35 witherspoon systemd[1]: phosphor-fan-control-init@0.service: Failed with result 'exit-code'.
Nov 29 12:00:35 witherspoon systemd[1]: phosphor-fan-control-init@0.service: Triggering OnFailure= dependencies.
Nov 29 12:00:35 witherspoon systemd[1]: Created slice system-witherspoon\x2dfan\x2dwatchdog\x2dmonitor.slic
```

msbarth, 2018-11-29:
The hwmon instance for the max31785 controller chip isnt running
```
root@witherspoon:~# ps | grep -i hwmon
 1226 root      6760 S    phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@100/power-supply@69
 1256 root      6868 S    phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@100/dps310@76
 1257 root      6760 S    phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@100/power-supply@68
 1296 root      6760 S    phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@380/tmp275@4a
 1817 root      6868 S    phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@180/ir35221@71
 1818 root      6868 S    phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@180/ir35221@70
 1819 root      6760 S    phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@140/ir35221@71
 1820 root      6868 D    phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@140/ir35221@70
```
This causes fan control to fail in setting the initial fan speed target. Was there any recent changes to hwmon?

msbarth, 2018-11-29:
@spinler found the following:

root@witherspoon:/sys/bus/i2c/drivers/max31785# Nov 29 14:59:38 witherspoon systemd[1]: Starting MAX31785 hwmon helper...
Nov 29 14:59:38 witherspoon max31785-hwmon-helper[2057]: cp: can't stat '/etc/default/obmc/hwmon//ahb/apb/i2c@1e78a000/**i2c/bus**@100/max31785@52_water.conf': No such file or directory

**i2c/bus** should be **i2c-bus**

msbarth, 2018-12-04:
This issue was a result of merging https://gerrit.openbmc-project.xyz/#/c/openbmc/meta-phosphor/+/15607/.

This change has been temporarily reverted until the necessary updates can be made to start hwmon device instances using udev rules to start the systemd service for each device. The intention is to remove the need for intermediate shell scripts to start the services along with have a single `70-hwmon.rules` using the device path.

msbarth, 2019-01-09:
A fix has been submitted to correctly start the max31785 hwmon service.
https://gerrit.openbmc-project.xyz/#/c/openbmc/meta-ibm/+/16799/

A separate issue has been created to track the work on converting udev rules into starting the hwmon services directly as noted previously.
https://github.com/openbmc/openbmc/issues/3455

closed 2019-01-09 by bradbishop