openbmc/openbmc #3444: Newer SBEFIFO DD doesn't allow chipOps when in istep mode

sannerd, 2018-12-04:
During some prototyping on Witherspoon ran into the fact that newer BMC levels prevent chipOps to the SBEFIFO DD when in "istep mode".   Given that we use chipOps to istep the SBE... means that istep is broken.  @eddiejames found this in the SBEFIFO Driver:

sbefifo 00:00:00:06: SBE state=-16
    ```case SBE_STATE_IPLING:
    case SBE_STATE_ISTEP:
    case SBE_STATE_MPIPL:
    case SBE_STATE_DMT:
        return -EBUSY;```

Please fix ASAP?  I do know that a version from May works... not sure when this got changed.

eddiejames, 2019-02-13:
I sent a patch to the linux kernel list several weeks ago. It hasn't been acknowledged.

apopple, 2019-06-13:
@eddiejames which patch was it? Guessing this one https://lkml.org/lkml/2019/1/21/1212?

eddiejames, 2019-06-14:
Thats correct @apopple

shenki, 2019-06-17:
This patch has landed in the openbmc kernel tree, and is in openbmc master.

closed 2019-12-18 by shenki