openbmc/openbmc #3445: open BMC

zhuifeng091, 2018-12-05:
i am configuring interface from openBMC and SoC in the arch/arm/boot/dts/*.dts file. openBMC act as slave device and SoC act as master on one i2c bus, openBMC act as master and SoC act as slave on another i2c bus, how should do in the dts file and what's the driver of SoC when it act as master and slave? it is not similar with general sensor like tmp421.

gtmills, 2019-01-23:
Hi @zhuifeng091, this is a good question for the mailing list, as it's not a bug or feature request. You can sign yourself up at https://lists.ozlabs.org/listinfo/openbmc

Thanks,
Gunnar

@amboar Might be able to help

edtanous, 2019-01-23:
Based on the flow you describe it sounds very similar to ipmb.  You might try starting by looking at the ipmi daemon.

closed 2019-07-26 by amboar