openbmc/openbmc #3450: IPv4 rg.freedesktop.DBus.Error.FileNotFound: path or object not found"

gkeishin, 2018-12-19:
On Master, IPV4 property cant be access directly via URL, but ONLY through enumerate
```
Executing: GET  /xyz/openbmc_project/network/eth0/ipv4

/xyz/openbmc_project/network/eth0/ipv4
{
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found"
}

Executing: GET  /xyz/openbmc_project/network/eth0/ipv4/

/xyz/openbmc_project/network/eth0/ipv4/
{
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found"
}

Executing: GET  /xyz/openbmc_project/network/eth0/ipv4/enumerate

/xyz/openbmc_project/network/eth0/ipv4/enumerate
{
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found"
}

```

```
Executing: GET  /xyz/openbmc_project/network/eth0/enumerate

/xyz/openbmc_project/network/eth0/enumerate
{
    "/xyz/openbmc_project/network/eth0/ipv4/6170c3bf": {
        "Address": "xx.xx.xx.83",
        "Gateway": "xx.xx.xx.1",
        "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.Static",
        "PrefixLength": 23,
        "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv4"
    },
    "/xyz/openbmc_project/network/eth0/ipv4/7ff23945": {
        "Address": "169.254.241.23",
        "Gateway": "0.0.0.0",
        "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.LinkLocal",
        "PrefixLength": 16,
        "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv4"
    },
    "/xyz/openbmc_project/network/eth0/ipv6/99385aaf": {
        "Address": "fe80::72e2:84ff:fe14:3365",
        "Gateway": "",
        "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.LinkLocal",
        "PrefixLength": 64,
        "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv6"
    }
}
```

bradbishop, 2018-12-19:
@spinler I'm guessing this is a c++ mapper problem...

spinler, 2019-01-02:
Now what I think is happening is when objects are created with a path like /root/A/B, where /root/A didn't previously exist, then the mapper would get the interfacesAdded signal for it, and only add the interfaces in that signal message, which would only be for /root/A/B.  That leaves /root/A out in the cold.

If these same paths already exist when the mapper starts, then it will introspect everything and find all the paths, and we won't hit this.

spinler, 2019-01-03:
I put up a fix for this at https://gerrit.openbmc-project.xyz/#/c/openbmc/phosphor-objmgr/+/17053/

rfrandse, 2019-01-04:
<https://gerrit.openbmc-project.xyz/17053> mapper: Add missing parent paths in ifaces-added
Resolves  #3450 REST Regression : Org.freedesktop.DBus.Error.FileNotFound: path or object not found"

closed 2019-01-11 by geissonator