openbmc/openbmc #3451: Internal error when changing the MAC address

awilliams-mrvl, 2018-12-29:
I am getting an internal error when I attempt to change the MAC address, either via REST or busctl:

# busctl set-property xyz.openbmc_project.Network /xyz/openbmc_project/network/eth0 xyz.openbmc_project.Network.M
ACAddress MACAddress s "00:0f:b7:11:a9:57"
The operation failed internally.

curl -c cjar -b cjar -k -H "Content-Type: application/json" -X PUT -d '{"data":"00:0f:b7:11:a9:57"}' https://10.85.93.62/xyz/openbmc_project/network/eth0/attr/MACAddress
{
  "data": {
    "description": "Internal Server Error", 
    "exception": "DBusException('The operation failed internally.',)", 
    "traceback": [
      "Traceback (most recent call last):", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 862, in _handle", 
      "    return route.call(**args)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 1281, in __call__", 
      "    return self.callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 1346, in wrap", 
      "    return callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 1570, in __call__", 
      "    return self.callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 1544, in wrap", 
      "    return self.to_jsonp(callback(*a, **kw))", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 1742, in wrapper", 
      "    rv = callback(*a, **ka)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 1508, in wrap", 
      "    data = callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 1397, in wrap", 
      "    return callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 1600, in __call__", 
      "    resp = self.callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 243, in __call__", 
      "    return getattr(self, 'do_' + request.method.lower())(**kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 498, in do_put", 
      "    properties_iface.Set(iface, prop, value)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__", 
      "    **keywords)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking", 
      "    message, timeout)", 
      "DBusException: xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally."
    ]
  }, 
  "message": "500 Internal Server Error", 
  "status": "error"
}

Note that in our case we have made a slight modification to U-Boot in order to swap the two Ethernet ports.

bradbishop, 2019-04-18:
@ratagupt

ratagupt, 2019-07-23:
We were doing some restructuring around this area, it is done, Can you please run it again and let us know.

ratagupt, 2019-07-23:
@awilliams-mrvl ^^^

z1908144712, 2019-09-06:
@ratagupt I also encountered the same problem. When I modify the first two hexadecimal digits of the MAC address, the above errors will occur. Modifying any remaining hexadecimal digits will not cause this problem. How do I solve this problem?

mdmillerii, 2019-09-06:
@z198144712 The braodcast and administratively defined bits are the low two bits of the leading byte of the mac address.  Did you keep the constraints when trying to change the MAC?

z1908144712, 2019-09-08:
@mdmillerii I used the default settings, maybe there are some constraints, because I am new to join, I don't know much about these constraints. When I modify 82:XX:XX:XX:XX:XX (BMC starts randomly generated) is 88::XX:XX:XX:XX:XX or 77::XX:XX:XX:XX:XX or something else I will get this internal error, I am not sure if it is caused by some constraints. If it is caused by a constraint, I think a normal return is needed here instead of an internal error 500. So do you have some suggestions?

mdmillerii, 2019-09-09:
@z1908144712  yes, if the original was 82 then the second digit needs to be one of `2`,`6`,`a`, or `e`.

@ratagupt This part of the code requires a non-broadcast administratively assigned address, and the return is not being handled.

z1908144712, 2019-09-10:
@mdmillerii @ratagupt At present, a serious problem has been found. Once this internal error is triggered, some network services of openbmc system will be ineffective, such as rest-dbus, SSH and other services will be ineffective for a period of time.

mdmillerii, 2019-09-10:
> Modifying any remaining hexadecimal digits will not cause this problem. 

@z1908144712 can you confirm setting bytes 1-7 or nibble 0 to any hex digit, or nibble 1 to one in the list of 4 values above, work equally well?

z1908144712, 2019-09-11:
@mdmillerii Yes, for example FE: XX: XX: XX: XX: XX: XX, I can change XX to any number, even FF or 00. According to my latest test, F can be changed to any number, E can only be changed to 2, 6, a, E. In my previous tests, F could not be modified, E could not be modified to 2, 6, a, e or other numbers. I don't know what the problem is, but I can fix it now as you say.

mdmillerii, 2019-09-11:
Thanks.  The settings code is requiring that the string be a valid mac address.  That includes checking that the address is not a broadcast address.

mdmillerii, 2020-03-12:
@ratagupt can we create a clean error return here?
@gkeishin the test here is attempt to set a multicast MAC ; it fails but just internal error.

closed 2020-10-12 by stale[bot]