openbmc/openbmc #3452: Redfish links on click takes to openbmc GUI page

sivassrr, 2019-01-02:
System is loaded with latest master version
 cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.4-207"
VERSION_ID="v2.4-207-g34f5eff"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.4-207"
BUILD_ID="v2.4"

[image]

In the above screen, on click any of the redfish link, it takes to openbmc GUI page.

edtanous, 2019-01-03:
Have you logged in?  I what you’re describing is expected behavior if you haven’t yet logged in.  The redfish service root is available without logging in, that’s why it works correctly.

Please let me know if logging in to the webui resolves your issue.

sivassrr, 2019-01-07:
No I have not logged in. but we expect data like the main redfish/v1 or expansion of that, right.

sivassrr, 2019-01-08:
@edtanous expecting similar to mock https://redfish.dmtf.org/redfish/mockups/v1/863

If you click it should go to either redfish login and then come back expand as shown in the mockup  url.

edtanous, 2019-01-08:
I'm not sure what you mean by "redfish login"  The page you linked, has no login, and is a mockup.  As such, when realized on a real BMC, the implementation can't always exactly match, in this case because of security.

Redfish itself technically requires a login prior to accessing any of those pages below service root, so per the specification, we should be returning 401, and nothing else.  Given that redfish is itself a rest API, it has no GUI elements, nor is it really intended to be used from a browser.  In this case, you're using a browser, which can be detected, and handled a little more gracefully, by returning you to a login screen.

After logging in, with the latest versions of both Bmcweb and phosphor-webui, it does return you to the page that you clicked on.  I tested this earlier this week on my pure bmcweb system, and it works correctly.  I do no believe this feature works with the nginx/phosphor-rest/bmcweb flow that IBM systems currently implement.  You might consider talking to the nginx/phosphor-rest maintainers about adding that support, although it sounds like IBM is moving over to bmcweb anyway, so it might be moot.

I think what would help here is an exact flow that you expect.  The flow that works on my system is:
1. open /redfish/v1/ in a browser, observe service root.
2. Click on /redfish/v1/SessionService
3. Observe the phosphor-rest login page.  Provide correct credentials and log in
4. Browser is redirected to /redfish/v1/SessionService/

This works because the webui shares shares credentials with Redfish in a pure bmcweb system.  This is not true of IBM systems.  If you're interested in implementing a redfish specific browser login mechanism, I'm happy to help support and review, but I think the efforts are better spent on getting your systems into a pure bmcweb configuration.

gtmills, 2019-01-23:
@sivassrr Can this be closed?

edtanous, 2019-01-23:
@sivassrr What would you prefer to happen if a user requests a page they don't have access to from a browser?

The secondary forwarding is broken in the nginx + bmcweb setup, so we could leave this open until nginx is deprecated, but it would be good to know if you're expecting something different than the way bmcweb solo behaves today.

sivassrr, 2019-01-23:
Expectation:
By default if we are getting the above data (in the picture) with all the links enabled and if we click that link, if we already login then it should provide data, if not then we should be asked to login into Redfish. 

Fine with waiting till nginx gets deprecated.

edtanous, 2019-01-23:
I believe this is the behavior when we move to the pure bmcweb solution.  Once that's merged (or you have a chance to try a test build) it would be good if you could comment back here and let me know if this is the behavior you were looking for.

spinler, 2019-01-23:
@sivassrr If you want a test build I have one.

sivassrr, 2019-01-24:
> @sivassrr If you want a test build I have one.

Sure @spinler  that helps. I can test that. Thank you

sivassrr, 2019-01-24:
With @spinler sandbox able to login in to bmcweb where it has Redfish API. 
From source code output as follows:
[image]

Couldnt power on the system though. As this sandbox , will check once we get real master release.

closed 2019-01-24 by sivassrr