openbmc/openbmc #3453: SNMP configuration not working via REST

prkatti1, 2019-01-09:
SNMP configuration via REST in not working, where as it works via busctl.

curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d '{"data": ["55.55.55.55",222] }' https://bmcxxx/xyz/openbmc_project/network/snmp/manager/action/Client
{
  "data": {
    "description": "Invalid number of args"
  },
  "message": "400 Bad Request",
  "status": "error"

prkatti1, 2019-01-09:
busctl works fine tfor snmp configuration:
Comand used is:
busctl call  xyz.openbmc_project.Network.SNMP /xyz/openbmc_project/network/snmp/manager xyz.openbmc_project.Network.Client.Create Client sq "55.55.55.55" 162
s "/xyz/openbmc_project/network/snmp/manager/2"

gkeishin, 2019-01-11:
@spinler changes does have regression but haven't seen at parameter level errors for other module like NW. where we do similar. So, wondering if its something else here @ratagupt

spinler, 2019-01-11:
Is this in master, or some IBM driver?

gkeishin, 2019-01-11:
On master @spinler and latest 2.6

ratagupt, 2019-01-14:
@spinler I got a chance to take a look into this, I put some debug traces in rest-server while rest server gets the signature of the method in the do_post.
```
              signature_list = get_method_signature(self.bus, self.service,
                                                      path, self.interface,
                                                      method)
                print("**************Debug****************")
                print(self.bus, self.service, path, self.interface, method)
                print(signature_list)
                print("***********************************")
```

ratagupt, 2019-01-14:
@spinler I figured out that there is a bug in the Rest-server.. which creates the parameter signature type by looking at the "arg",but the arg can be "input/output".so here we need to put a code which creates the method signature by looking at the arg (input/output) type.
```

def get_method_signature(bus, service, object_path, interface, method):
    obj = bus.get_object(service, object_path)
    iface = dbus.Interface(obj, 'org.freedesktop.DBus.Introspectable')
    xml_string = iface.Introspect()
    arglist = []
 
    root = ElementTree.fromstring(xml_string)
    for dbus_intf in root.findall('interface'):
        if (dbus_intf.get('name') == interface):
            for dbus_method in dbus_intf.findall('method'):
                      if(dbus_method.get('name') == method):
                    for arg in dbus_method.findall('arg'):
                        arglist.append(arg.get('type'))
                    return arglist
```
example introspect xml like below
```Jan 14 12:46:15 witherspoon phosphor-gevent[1929]:  <interface name="xyz.openbmc_project.Network.Client.Create">
Jan 14 12:46:15 witherspoon phosphor-gevent[1929]:   <method name="Client">
Jan 14 12:46:15 witherspoon phosphor-gevent[1929]:    <arg type="s" direction="in"/>
Jan 14 12:46:15 witherspoon phosphor-gevent[1929]:    <arg type="q" direction="in"/>
Jan 14 12:46:15 witherspoon phosphor-gevent[1929]:    <arg type="s" direction="out"/>
```

gkeishin, 2019-01-14:
Thanks @ratagupt

rfrandse, 2019-01-16:
<https://gerrit.openbmc-project.xyz/17408> Fixes the method signature generation logic
Fixes  #3453 SNMP configuration not working via REST

closed 2019-01-17 by geissonator