openbmc/openbmc #3454: GUI: Able to create multiple users with various roles but failed to login with them

sivassrr, 2019-01-09:
[image]

root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.4-247"
VERSION_ID="v2.4-247-g1a4b7ee"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.4-247"
BUILD_ID="v2.4"

Journalctl output 
root@witherspoon:~# journalctl -f
-- Logs begin at Wed 2019-01-09 09:42:05 UTC. --
Jan 09 09:48:16 witherspoon systemd[1]: rsyslog.service: Main process exited, code=exited, status=1/FAILURE
Jan 09 09:48:16 witherspoon systemd[1]: rsyslog.service: Failed with result 'exit-code'.
Jan 09 09:48:16 witherspoon systemd[1]: Failed to start System Logging Service.
Jan 09 09:48:17 witherspoon systemd[1]: rsyslog.service: Service RestartSec=1s expired, scheduling restart.
Jan 09 09:48:17 witherspoon systemd[1]: rsyslog.service: Scheduled restart job, restart counter is at 9192.
Jan 09 09:48:17 witherspoon systemd[1]: Stopped System Logging Service.
Jan 09 09:48:17 witherspoon systemd[1]: Starting System Logging Service...
Jan 09 09:48:17 witherspoon systemd[1]: rsyslog.service: Main process exited, code=exited, status=1/FAILURE
Jan 09 09:48:17 witherspoon systemd[1]: rsyslog.service: Failed with result 'exit-code'.
Jan 09 09:48:17 witherspoon systemd[1]: Failed to start System Logging Service.
Jan 09 09:48:18 witherspoon systemd[1]: rsyslog.service: Service RestartSec=1s expired, scheduling restart.
Jan 09 09:48:18 witherspoon systemd[1]: rsyslog.service: Scheduled restart job, restart counter is at 9193.
Jan 09 09:48:18 witherspoon systemd-journald[741]: Forwarding to syslog missed 169 messages.
Jan 09 09:48:18 witherspoon systemd[1]: Stopped System Logging Service.
Jan 09 09:48:18 witherspoon systemd[1]: Starting System Logging Service...
Jan 09 09:48:18 witherspoon systemd[1]: rsyslog.service: Main process exited, code=exited, status=1/FAILURE
Jan 09 09:48:18 witherspoon systemd[1]: rsyslog.service: Failed with result 'exit-code'.
Jan 09 09:48:18 witherspoon systemd[1]: Failed to start System Logging Service.
Jan 09 09:48:19 witherspoon systemd[1]: rsyslog.service: Service RestartSec=1s expired, scheduling restart.
Jan 09 09:48:19 witherspoon systemd[1]: rsyslog.service: Scheduled restart job, restart counter is at 9194.
Jan 09 09:48:19 witherspoon systemd[1]: Stopped System Logging Service.
Jan 09 09:48:19 witherspoon systemd[1]: Starting System Logging Service...
Jan 09 09:48:19 witherspoon systemd[1]: rsyslog.service: Main process exited, code=exited, status=1/FAILURE
Jan 09 09:48:19 witherspoon systemd[1]: rsyslog.service: Failed with result 'exit-code'.
Jan 09 09:48:19 witherspoon systemd[1]: Failed to start System Logging Service.
Jan 09 09:48:21 witherspoon systemd[1]: rsyslog.service: Service RestartSec=1s expired, scheduling restart.
Jan 09 09:48:21 witherspoon systemd[1]: rsyslog.service: Scheduled restart job, restart counter is at 9195.
Jan 09 09:48:21 witherspoon systemd[1]: Stopped System Logging Service.
Jan 09 09:48:21 witherspoon systemd[1]: Starting System Logging Service...
Jan 09 09:48:21 witherspoon systemd[1]: rsyslog.service: Main process exited, code=exited, status=1/FAILURE
Jan 09 09:48:21 witherspoon systemd[1]: rsyslog.service: Failed with result 'exit-code'.
Jan 09 09:48:21 witherspoon systemd[1]: Failed to start System Logging Service.
Jan 09 09:48:22 witherspoon systemd[1]: rsyslog.service: Service RestartSec=1s expired, scheduling restart.
Jan 09 09:48:22 witherspoon systemd[1]: rsyslog.service: Scheduled restart job, restart counter is at 9196.
Jan 09 09:48:22 witherspoon systemd[1]: Stopped System Logging Service.
Jan 09 09:48:22 witherspoon systemd[1]: Starting System Logging Service...
Jan 09 09:48:22 witherspoon systemd[1]: rsyslog.service: Main process exited, code=exited, status=1/FAILURE
Jan 09 09:48:22 witherspoon systemd[1]: rsyslog.service: Failed with result 'exit-code'.
Jan 09 09:48:22 witherspoon systemd[1]: Failed to start System Logging Service.
Jan 09 09:48:23 witherspoon systemd[1]: rsyslog.service: Service RestartSec=1s expired, scheduling restart.
Jan 09 09:48:23 witherspoon systemd[1]: rsyslog.service: Scheduled restart job, restart counter is at 9197.
Jan 09 09:48:23 witherspoon systemd[1]: Stopped System Logging Service.
Jan 09 09:48:23 witherspoon systemd[1]: Starting System Logging Service...
Jan 09 09:48:23 witherspoon systemd[1]: rsyslog.service: Main process exited, code=exited, status=1/FAILURE
Jan 09 09:48:23 witherspoon systemd[1]: rsyslog.service: Failed with result 'exit-code'.
Jan 09 09:48:23 witherspoon systemd[1]: Failed to start System Logging Service.

sivassrr, 2019-01-09:
```root@witherspoon:~# users
root root```

sivassrr, 2019-01-09:
[image]

rthomaiy, 2019-01-09:
Is there any log attached for this issue? If yes, where the same can be seen. 
If not, can you post the output of these commands (Replace the user name accordingly)
1. busctl introspect xyz.openbmc_project.User.Manager /xyz/openbmc_project/user/<user which you are seeing the problem>
2. cat /etc/groups

sivassrr, 2019-01-09:
1. busctl introspect xyz.openbmc_project.User.Manager /xyz/openbmc_project/user

```
root@witherspoon:~# busctl introspect xyz.openbmc_project.User.Manager /xyz/openbmc_project/user | cat
NAME                                   TYPE      SIGNATURE  RESULT/VALUE                             FLAGS
org.freedesktop.DBus.Introspectable    interface -          -                                        -
.Introspect                            method    -          s                                        -
org.freedesktop.DBus.ObjectManager     interface -          -                                        -
.GetManagedObjects                     method    -          a{oa{sa{sv}}}                            -
.InterfacesAdded                       signal    oa{sa{sv}} -                                        -
.InterfacesRemoved                     signal    oas        -                                        -
org.freedesktop.DBus.Peer              interface -          -                                        -
.GetMachineId                          method    -          s                                        -
.Ping                                  method    -          -                                        -
org.freedesktop.DBus.Properties        interface -          -                                        -
.Get                                   method    ss         v                                        -
.GetAll                                method    s          a{sv}                                    -
.Set                                   method    ssv        -                                        -
.PropertiesChanged                     signal    sa{sv}as   -                                        -
xyz.openbmc_project.User.AccountPolicy interface -          -                                        -
.AccountUnlockTimeout                  property  u          0                                        emits-change writable
.MaxLoginAttemptBeforeLockout          property  q          0                                        emits-change writable
.MinPasswordLength                     property  y          8                                        emits-change writable
.RememberOldPasswordTimes              property  y          0                                        emits-change writable
xyz.openbmc_project.User.Manager       interface -          -                                        -
.CreateUser                            method    sassb      -                                        -
.RenameUser                            method    ss         -                                        -
.AllGroups                             property  as         4 "ipmi" "redfish" "ssh" "web"           emits-change writable
.AllPrivileges                         property  as         4 "priv-admin" "priv-operator" "priv-us… emits-change writable
.UserRenamed                           signal    ss         -                                        -
root@witherspoon:~#```

2. cat /etc/group

```root@witherspoon:~#  cat /etc/group
root:x:0:
daemon:x:1:
bin:x:2:
sys:x:3:
adm:x:4:
tty:x:5:
disk:x:6:
lp:x:7:
mail:x:8:
news:x:9:
uucp:x:10:
man:x:12:
proxy:x:13:
kmem:x:15:
input:x:19:
dialout:x:20:
fax:x:21:
voice:x:22:
cdrom:x:24:
floppy:x:25:
tape:x:26:
sudo:x:27:
audio:x:29:
dip:x:30:
www-data:x:33:www
backup:x:34:
operator:x:37:
list:x:38:
irc:x:39:
src:x:40:
gnats:x:41:
shadow:x:42:
utmp:x:43:
video:x:44:
sasl:x:45:
plugdev:x:46:
staff:x:50:
games:x:60:
shutdown:x:70:
users:x:100:
systemd-bus-proxy:x:991:
systemd-resolve:x:992:
systemd-coredump:x:993:
systemd-network:x:994:
systemd-timesync:x:995:
systemd-journal:x:996:
messagebus:x:997:
netdev:x:998:
www:x:999:
priv-admin:x:1000:root,sivas,sivas0penBmc
priv-operator:x:1001:sivasoperator
priv-user:x:1002:sivasuser
priv-callback:x:1003:sivascallback
web:x:1004:root,sivas,sivasoperator,sivasuser,sivascallback,sivas0penBmc
redfish:x:1005:root,sivas,sivasoperator,sivasuser,sivascallback,sivas0penBmc
ipmi:x:1006:root
nogroup:x:65534:
wheel:x:990:
kvm:x:989:
render:x:988:
nobody:x:987:
root@witherspoon:~#```

3. cat /etc/passwd

```root@witherspoon:~# cat /etc/passwd
root:x:0:0:root:/home/root:/bin/sh
daemon:x:1:1:daemon:/usr/sbin:/bin/sh
bin:x:2:2:bin:/bin:/bin/sh
sys:x:3:3:sys:/dev:/bin/sh
sync:x:4:65534:sync:/bin:/bin/sync
games:x:5:60:games:/usr/games:/bin/sh
man:x:6:12:man:/var/cache/man:/bin/sh
lp:x:7:7:lp:/var/spool/lpd:/bin/sh
mail:x:8:8:mail:/var/mail:/bin/sh
news:x:9:9:news:/var/spool/news:/bin/sh
uucp:x:10:10:uucp:/var/spool/uucp:/bin/sh
proxy:x:13:13:proxy:/bin:/bin/sh
www-data:x:33:33:www-data:/var/www:/bin/sh
backup:x:34:34:backup:/var/backups:/bin/sh
list:x:38:38:Mailing List Manager:/var/list:/bin/sh
irc:x:39:39:ircd:/var/run/ircd:/bin/sh
gnats:x:41:41:Gnats Bug-Reporting System (admin):/var/lib/gnats:/bin/sh
systemd-bus-proxy:x:993:991::/:/bin/nologin
systemd-resolve:x:994:992::/:/bin/nologin
systemd-coredump:x:995:993::/:/bin/nologin
systemd-network:x:996:994::/:/bin/nologin
systemd-timesync:x:997:995::/:/bin/nologin
messagebus:x:998:997::/var/lib/dbus:/bin/false
www:x:999:999::/var/www/localhost:/bin/sh
nobody:x:65534:65534:nobody:/nonexistent:/bin/sh
sivas:x:1000:100::/home/sivas:/bin/sh
sivasoperator:x:1001:100::/home/sivasoperator:/bin/sh
sivasuser:x:1002:100::/home/sivasuser:/bin/sh
sivascallback:x:1003:100::/home/sivascallback:/bin/sh
sivas0penBmc:x:1004:100::/home/sivas0penBmc:/bin/sh
root@witherspoon:~#```

sivassrr, 2019-01-10:
Thank you @gtmills

rfrandse, 2019-02-13:
<https://gerrit.openbmc-project.xyz/18276> Make default redfishSupportEnabled
Resolves  #3454 GUI: Able to create multiple users with various roles but failed to login with them

closed 2019-02-14 by geissonator