openbmc/openbmc #3458: How to match power state between openbmc and redfish interface?

xuweibj, 2019-01-11:
I run commands against the same node, get response like this:
* openbmc:
```
# curl -k -c cjar -b cjar -X GET -H "Content-Type: application/json" https://10.5.16.101/xyz/openbmc_project/state/enumerate
{
  "data": {
    "/xyz/openbmc_project/state/bmc0": {
      "CurrentBMCState": "xyz.openbmc_project.State.BMC.BMCState.Ready",
      "LastRebootTime": 1547135492000,
      "RequestedBMCTransition": "xyz.openbmc_project.State.BMC.Transition.None"
    },
    "/xyz/openbmc_project/state/chassis0": {
      "CurrentPowerState": "xyz.openbmc_project.State.Chassis.PowerState.On",
      "LastStateChangeTime": 1547170791002,
      "POHCounter": 10,
      "RequestedPowerTransition": "xyz.openbmc_project.State.Chassis.Transition.Off"
    },
    "/xyz/openbmc_project/state/host0": {
      "AttemptsLeft": 3,
      "BootProgress": "xyz.openbmc_project.State.Boot.Progress.ProgressStages.OSStart",
      "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Running",
      "OperatingSystemState": "xyz.openbmc_project.State.OperatingSystem.Status.OSStatus.BootComplete",
      "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.On"
    }
  },
  "message": "200 OK",
  "status": "ok"
}
```

* redfish
```
/redfish/v1/Chassis/motherboard
"PowerState": "On",
```
```
/redfish/v1/Chassis/system
"PowerState": "On",
```
Which above one is matched ``/xyz/openbmc_project/state/chassis0``?
```
/redfish/v1/Systems/motherboard
"PowerState": "Off",
```
Is this matched ``/xyz/openbmc_project/state/host0``?

@zet809 @gurevichmark @whowutwut

gtmills, 2019-01-11:
Hi @xuweibj, this is a good question for the mailing list or IRC, as it's not a bug or feature request. You can sign yourself up for the mailing list at https://lists.ozlabs.org/listinfo/openbmc

Let me try to answer the question here. `/xyz/openbmc_project/state/host0` maps to `/redfish/v1/Systems/<systemId>`. https://github.com/openbmc/bmcweb/blob/947326618971c0f658f4a5348df2f47d8819b165/redfish-core/lib/systems.hpp#L475

`/xyz/openbmc_project/state/chassis0` maps to `/redfish/v1/Chassis/<ChassisId>` although there is a bug to get the correct powerstate https://github.com/openbmc/bmcweb/issues/27 

Thanks,
Gunnar

xuweibj, 2019-01-14:
@gtmills Thanks for your reply.
I still have a question that from your reply ``/xyz/openbmc_project/state/chassis0`` maps to ``/redfish/v1/Chassis/<ChassisId>``, but I could see 2 <ChassisId> for one node here, only one ``/xyz/openbmc_project/state/chassis0``. So how to match this? Or will be modified to only one <ChassisId> for one node?

And another question, what other chassisid could be? I saw ``planar`` and ``backplane`` for boston machine.

xuweibj, 2019-01-21:
@gtmills Could you help to answer my above question please?

gtmills, 2019-01-23:
@xuweibj  On Witherspoon (and similar systems) there will only be 1 chassis. There is a commit here to remove `Inventory.Item.System` from the list of interfaces bmcweb uses to determine what a chassis is: 
https://gerrit.openbmc-project.xyz/#/c/openbmc/bmcweb/+/16586/  Once in, `/redfish/v1/Chassis/system` should no longer appear. 

There is also an option being considered here, https://gerrit.openbmc-project.xyz/#/c/openbmc/bmcweb/+/17563/ which would tell bmcweb there is only chassis. This is needed for a different problem (mapping sensors at `/xyz/openbmc_project/sensors/` to a chassis). This commit would present the lone chassis object at `/redfish/v1/Chassis/chassis/`. Note: You should never assume the chassis object path, but should query  `/redfish/v1/Chassis/` for the available chassis.