openbmc/openbmc #3459: REST request to set BMC time fails with 403 Forbidden error

rahulmah, 2019-01-11:
REST fails with 403 Forbidden error while setting BMC time. See below step to hit the issue. It is seen that if we give 20-30 seconds delay between each step below, then we will not hit the issue. Seem like some process is not getting complete if we don't give the delay. 

Step to recreate:
1. Set BMC as time owner
```
bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"xyz.openbmc_project.Time.Owner.Owners.BMC\"}" -X PUT  https://$BMC_IP/xy/openbmc_project/time/owner/attr/TimeOwner
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
```

2. Set NTP as Time synchronization method.
```
bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"xyz.openbmc_project.Time.Synchronization.Method.NTP\"}" -X PUT  https://$BMC_IP/xyz/openbmc_project/time/sync_method/attr/TimeSyncMethod
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
```

3. Now Set Manual as time synchronization method.
```
curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"xyz.openbmc_project.Time.Synchronization.Method.Manual\"}" -X PUT  https://$BMC_IP/xyz/openbmc_project/time/sync_method/attr/TimeSyncMethod
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
```

4. Check BMC and HOST time.
```
bash-4.1$ curl -c cjar -b cjar -k -X GET  https://$BMC_IP/xyz/openbmc_project/time/enumerate
{
  "data": {
    "/xyz/openbmc_project/time/bmc": {
      "Elapsed": 1547191668895457
    },
    "/xyz/openbmc_project/time/host": {
      "Elapsed": 1547191668931703
    },
    "/xyz/openbmc_project/time/owner": {
      "TimeOwner": "xyz.openbmc_project.Time.Owner.Owners.BMC"
    },
    "/xyz/openbmc_project/time/sync_method": {
      "TimeSyncMethod": "xyz.openbmc_project.Time.Synchronization.Method.Manual"
    }
  },
  "message": "200 OK",
  "status": "ok"
}
```

5. Set BMC time to 1487304700000000(GMT - Wednesday, September 18, 1974 10:01:10 AM).

```
bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": 1487304700000000}" -X PUT  https://$BMC_IP/xyz/openbmc_project/time/bmc/attr/Elapsed
{
  "data": {
    "description": "xyz.openbmc_project.Time.Error.Failed: The operation failed"
  },
  "message": "403 Forbidden",
  "status": "error"
}
```

Journal log of the failure - 
[journal log time.txt](https://github.com/openbmc/openbmc/files/2748535/journal.log.time.txt)

System Details
BMC : 2.6.0.rc1-23-g5a2b44f
PNOR : IBM-witherspoon-OP9-v2.0.10-2.22

mine260309, 2019-01-11:
@rahulmah 
1. The `journal log time.txt` is not complete, could you post a full journal log?
2. Could you login to BMC and run below command and see the output, at the same time when you are setting the time?
    ```
        timdatectl
    ```

rahulmah, 2019-01-11:
Okay I recreated the issue and captured all the journal logs. Here it is - 
[Journal_log_time_issue_new.txt](https://github.com/openbmc/openbmc/files/2748856/Journal_log_time_issue_new.txt)

and here is the output of "timedatectl" command.
```root@witherspoon1:~# timedatectl
               Local time: Fri 2019-01-11 08:57:30 UTC
           Universal time: Fri 2019-01-11 08:57:30 UTC
                 RTC time: Fri 2019-01-11 08:57:30
                Time zone: n/a (UTC, +0000)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no```

gkeishin, 2019-01-11:
@mine260309  This one shows up recently, is this due to yacto2.6 upgrades or something else?

mine260309, 2019-01-14:
@gkeishin Yeah, I believe so, and I just verified that on 2.4 the issue does not occur.
Need some investigation in 2.6.

mine260309, 2019-01-14:
I confirm that this behavior is caused due to systemd's change.

In `method_set_time()`, previously the code checks NTP by the property:
```
        if (c->use_ntp)
                return sd_bus_error_setf(error, BUS_ERROR_AUTOMATIC_TIME_SYNC_ENABLED, "Automatic time synchronization is enabled");
```

And now in yocto 2.6's systemd 239:
```
        r = context_update_ntp_status(c, bus, m);
        if (r < 0)
                return sd_bus_error_set_errnof(error, r, "Failed to update context: %m");

        if (context_ntp_service_is_active(c) > 0)
                return sd_bus_error_set(error, BUS_ERROR_AUTOMATIC_TIME_SYNC_ENABLED, "Automatic time synchronization is enabled");
```

The code calls `context_ntp_service_is_active()` to check if ntp service is active or not, which depends on the ntp service's status.

That is why:
* In 2.4, as long as NTP is disabled`c->use_ntp` becomes false, and user could set time;
* In 2.6, it needs to NTP service to quit, and then user could set time.

rahulmah, 2019-01-21:
Me and George had a discussion regarding this . And we are fine with option 1. See if we can we add some info regarding existing behavior in some doc for customer.

gkeishin, 2019-01-21:
https://gerrit.openbmc-project.xyz/#/c/openbmc/phosphor-time-manager/+/17546/

README for the sdbus changes behavious

rfrandse, 2019-01-21:
<https://gerrit.openbmc-project.xyz/17546> README: Add special note on NTP setting change
Resolves  #3459 REST request to set BMC time fails with 403 Forbidden error

gtmills, 2019-10-02:
We still see this behavior. @mine260309 Anything we can do? We added a workaround in the GUI to wait 20 seconds after setting NTP before setting TimeSyncMethod to Manual before setting the DateTime. This isn't a create user experience.

mine260309, 2019-10-08:
There is nothing we could do unless systemd allows setting time right after NTP is disabled.

bradbishop, 2019-10-08:
at 3:41 AM, Lei YU <notifications@github.com> wrote:

> There is nothing we could do unless systemd allows setting time right  
> after NTP is disabled.

IMO the web server should not return a reply until networkd has reloaded.   
That would have a number of benefits:

1 - it allows this problem to be solved
2 - it is the behavior users expect
3 - it allows us to report an error if reloading networkd fails

When we get around to moving this API to something Redfish-like, I’d guess  
@edtanous would ask for the same behavior as I’ve described.

edtanous, 2019-10-08:
"The issue occurs when you set time while NTP service has not quit yet"
That sounds like a bug where the phosphor-time-manager isn't buffering requests properly while the NTP service is restarting.

> IMO the web server should not return a reply until networkd has reloaded.
Replace web server with phosphor-time-manager, and I'd tend to agree, that would solve the problem if the dbus call didn't return until the operation was done.  With that said, if it were up to me, phosphor-time-manager would buffer/queue change requests while waiting for the NTP service to start/stop/restart, then apply them after the operation is complete.

> We still see this behavior. @mine260309 Anything we can do? We added a workaround in the GUI to wait 20 seconds after setting NTP before setting TimeSyncMethod to Manual before setting the DateTime. This isn't a great user experience.

An arguably better workaround would be to use the /subscribe API to watch for the service restart to be completed, then trigger the next time setting.  With that said, that's really lame, and I'd much prefer it just be fixed in the daemon itself.

mine260309, 2019-10-09:
> > The issue occurs when you set time while NTP service has not quit yet
> 
> That sounds like a bug where the phosphor-time-manager isn't buffering requests properly while the NTP service is restarting.

I kind of agree, but more like a bug in systemd instead of phosphor-time-manager.

> Replace web server with phosphor-time-manager, and I'd tend to agree, that would solve the problem if the dbus call didn't return until the operation was done. With that said, if it were up to me, [...]

I totally agree with the solution of `buffer/queue change requests while waiting for the NTP service`, but I would prefer to fix this in systemd.
If you check https://github.com/systemd/systemd/issues/11420, you could see that the below simple script will fail on setting time. It's not really a phosphor-time-manager bug, but a systemd bug.
```
timedatectl set-ntp 1
sleep 10  # Wait for a while for NTP service to start
timedatectl set-ntp 0
timedatectl set-time "2019-01-12" # This is expected to work, but it fails
echo $?
```

Anyway, we could fix this in `phosphor-time-manager` as a workaround, but I prefer to let systemd to fix the issue.
What do you think?
If you agree it's a systemd bug, let's re-open systemd/systemd#11420 or create a new one, and provide enough justification for systemd to fix it.

mdmillerii, 2020-04-10:
I see upstream now issues properties-changed after the operation is complete.

Do we need to adjust?

PS: There appears two be fixes 3 deep with a unit test added to Debian.

gtmills, 2020-07-01:
I hit this today on a witherspoon with   "FirmwareVersion": "2.9.0-dev-367-g4054df21e",

```
1$ !1081
curl -k -H "Content-Type: application/json" -d "{\"data\": \"xyz.openbmc_project.Time.Synchronization.Method.Manual\"}" -X PUT  https://$BMC_IP/xyz/openbmc_project/time/sync_method/attr/TimeSyncMethod
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ !1083
curl -k -H "Content-Type: application/json" -d "{\"data\": 1487304700000000}" -X PUT  https://$BMC_IP/xyz/openbmc_project/time/bmc/attr/Elapsed
{
  "data": {
    "description": "xyz.openbmc_project.Time.Error.Failed"
  },
  "message": "The operation failed",
  "status": "error"
}

``` 
When I retried a 1 second later: 
```
curl -k -H "Content-Type: application/json" -d "{\"data\": 1487304700000000}" -X PUT  htts://$BMC_IP/xyz/openbmc_project/time/bmc/attr/Elapsed
{
  "data": null,
  "message": "200 OK",
  "status": "ok"

```

vishwabmc, 2020-07-21:
can we close this and open a new one ?. This has tests from pre and post time removal features. @gtmills @rahulmah

gtmills, 2020-07-21:
@vishwabmc I vote we just continue to track here, since it appears to be the same problem, that said, I don't care that much.

gtmills, 2020-10-05:
@vishwabmc The GUI has a 20 sec timeout to work around this problem. This is far from ideal from a user perspective. 
https://github.com/openbmc/webui-vue/blob/ecd45a8392af2ada59dc846defa6392f5fc208a1/src/store/modules/Configuration/DateTimeSettingsStore.js#L57

Can we have phosphor-time-manager not return until the systemd service has been restarted? 
Do you know how our network service code worked around this similar problem? They use systemd networkd.

sumithcl2020, 2021-02-08:
@mine260309 , In phosphor-timemanager service, to set-time just after ntpsync disable (Mode change from NTP to HostSync) added a 20 seconds sleep after disabling NtpSync. Is adding a sleep fine or should we wait in a loop to disable ntpsync? During wait phosphor time manager will pull the status of NtpSync and once NtpSync disabled service will come out from the waiting loop.

mine260309, 2021-02-18:
> @mine260309 , In phosphor-timemanager service, to set-time just after ntpsync disable (Mode change from NTP to HostSync) added a 20 seconds sleep after disabling NtpSync. Is adding a sleep fine or [...]

phosphor-time-manager is expected not to block on DBus calls, otherwise it will block other services in case one service calls time-manager.
So I do not think it's a good idea for the time-manager to wait for the ntp status and return.

vishwabmc, 2021-02-18:
@ratagupt fyi

bradbishop, 2021-02-20:
> phosphor-time-manager is expected not to block on DBus calls, otherwise it will block other services in case one service calls time-manager.
> So I do not think it's a good idea for the time-manager to wait for the ntp status and return
> 

Agreed - we need an async dbus interface here.  So two interfaces - one that allows the operation to be started and one that allows a client to track the state of the operation.

closed 2022-04-17 by stale[bot]

gtmills, 2022-09-20:
Still an issue.

gtmills, 2024-03-05:
Still referenced in webui-vue. 
Still an issue. :/