openbmc/openbmc #3462: Error in bitbake kernel for Raspberry PI 3

Rolyli, 2019-01-16:
Hi MyFriends
When I trying to build OpenBMC image for Raspberry PI 3 on deepin-15.8 / 64 bits machine.

I copy meta-raspberrypi/conf/machine/raspberrypi3-64.conf to meta-evb/meta-evb-raspberrypi/conf
export TEMPLATECONF=meta-evb/meta-evb-raspberrypi/conf
 . openbmc-env
bitbake -k  obmc-phosphor-image
[image]
[image]

[image]
The following error appears:
[image]
below is my log
[log.tar.gz](https://github.com/openbmc/openbmc/files/2762409/log.tar.gz)
Thanks 
谢谢

geissonator, 2019-01-16:
There are some changes out there to get Raspberry Pi going again, may want to try pulling those down. https://gerrit.openbmc-project.xyz/#/q/owner:%22M.+D.+Truong%22+status:open

Rolyli, 2019-01-16:
> There are some changes out there to get Raspberry Pi going again, may want to try pulling those down. https://gerrit.openbmc-project.xyz/#/q/owner:%22M.+D.+Truong%22+status:open

Thanks you very much , i will try it later.

Rolyli, 2019-01-16:
unfortunately it doesn`t work
[image]
it still has the same problem.

Rolyli, 2019-01-16:
Update this iessue
I have solved this problem.
Because i didn't set the correct MACHINE in local.conf and when i set it to raspberrypi3-64.
This problem just gone.
[image]
But it still has a problem , this one looks like a lot of people have met, and I might be able to search for solutions in github.
[image]

In the end, my English is not very good. If my grammar have some problem ,please point it out.
Thanks
谢谢

closed 2019-01-21 by Rolyli