openbmc/openbmc #3463: Excess 16 bytes in `/etc/ipmi_pass`

nest1ing, 2019-01-16:
The implementation of writing `/etc/ipmi_pass` file has one excess padding. 

When we prepare the data to encrypt, we pad the data buffer to the size which is a multiple of the cipher block size. After that, it happens again in the encryption process, but the data block has been already padded,  so `openssl` adds a new full block.

As the result, we have excess 16 bytes in the `/etc/ipmi_pass`.

Two projects affected by this problem:
  * [openbmc/pam-ipmi](https://github.com/openbmc/pam-ipmi)
    first padding happens in `src/pam_ipmisave/pam_ipmisave.c` at lines [487](https://github.com/openbmc/pam-ipmi/blob/a32999cde49845ff904f1c7d48a6154b44002db3/src/pam_ipmisave/pam_ipmisave.c#L487) and [497](https://github.com/openbmc/pam-ipmi/blob/a32999cde49845ff904f1c7d48a6154b44002db3/src/pam_ipmisave/pam_ipmisave.c#L497)
    ```c
		pwptextlen =
			block_round(writtensize, EVP_CIPHER_block_size(cipher));
    ```
    second padding in the call of `EVP_CipherFinal` at line [199](https://github.com/openbmc/pam-ipmi/blob/a32999cde49845ff904f1c7d48a6154b44002db3/src/pam_ipmisave/pam_ipmisave.c#L199)
  * [openbmc/phospohor-host-ipmid](https://github.com/openbmc/phosphor-host-ipmid)
    first padding happens in `user_channel/passwd_mgr.cpp` at line [394](https://github.com/openbmc/phosphor-host-ipmid/blob/583171207c4f9a091abe3f867cc415e7f9aa514b/user_channel/passwd_mgr.cpp#L394)
    ```c++
        // Round of to block size and padding remaing bytes with zero.
        inBytesLen = blockRound(bytesWritten, EVP_CIPHER_block_size(cipher));
        std::memset(&inBytes[0] + bytesWritten, 0, inBytesLen - bytesWritten);
    ```
    second padding in the call of `EVP_CipherFinal` at line [176](https://github.com/openbmc/phosphor-host-ipmid/blob/583171207c4f9a091abe3f867cc415e7f9aa514b/user_channel/passwd_mgr.cpp#L176)

This does not lead to any failures, but it may be worth fixing it before it will reach customers?

edtanous, 2019-01-23:
@rthomaiy Low priority, but adding you for awareness.

rthomaiy, 2019-01-28:
fixed and under review 
1. phosphor-host-ipmid -> https://gerrit.openbmc-project.xyz/#/c/openbmc/phosphor-host-ipmid/+/17720/
2. pam-ipmi -> https://gerrit.openbmc-project.xyz/#/c/openbmc/pam-ipmi/+/17721/

rfrandse, 2019-01-28:
<https://gerrit.openbmc-project.xyz/17721> Resolves openbmc/openbmc#3463
Resolves  #3463 Excess 16 bytes in `/etc/ipmi_pass`

closed 2019-01-31 by geissonator