openbmc/openbmc #3464: intel s2600wf error

chunghuihuang, 2019-01-17:
I try to do this
export TEMPLATECONF=meta-intel/meta-s2600wf/conf
But get error,error message:
fatal error:phoshor-ipmi-host/utils.hpp:No such file or directory

rthomaiy, 2019-01-23:
this is fixed as of yesterday's OpenBMC master. 
Note: Still there is one more issue with utils.cpp (of moving std::experimental::nulopt_t to std). This is also fixed and merged in intel-ipmi-oem repo, but the version is not bumped up in .bb file. 
Hence update this file https://github.com/openbmc/openbmc/blob/master/meta-intel/meta-common/recipes-intel/ipmi/intel-ipmi-oem_git.bb#L8   version to e1cbc7db86ea15999141d0a58b4423a8428e78ec

chunghuihuang, 2019-01-23:
thank you for your help , i fix this error .
and i have a new question,when i test image by qemu, how can i use RSET

ex: step.3
https://gist.github.com/geissonator/71dad3ccca8bd42c02cfb9726df6c171

best regards,
Julie

Richard Marian Thomaiyar <notifications@github.com> 於 2019年1月23日 週三
下午4:08寫道：

chunghuihuang, 2019-01-23:
> @geissonator <https://github.com/geissonator> should be able to help
> there. I have not used QEMU for REST

Thank you very much!!!

Richard Marian Thomaiyar <notifications@github.com> 於 2019年1月23日 週三
下午5:03寫道：

closed 2019-01-23 by chunghuihuang