openbmc/openbmc #3465: IPMI services taking too much time to exit on BMC reboot

mine260309, 2019-01-17:
While rebooting BMC, all services are expected to exit cleanly, and systemd will wait for services to exit.

Both host ipmid and net-ipmid take too much time to exit, we see logs like below from serial console:
```
[  OK  ] Stopped Phosphor Inband IPMI.
         Stopping Phosphor IPMI BT DBus Bridge...
[  OK  ] Stopped Clear one time boot overrides.
[  OK  ] Stopped Phosphor IPMI BT DBus Bridge.
[*     ] A stop job is running for Network IPMI daemon (1min 18s / 1min 30s)
```

Host ipmid takes about 30 seconds to stop, and net ipmid eventually times out.
So it affects BMC reboot time by about 90 seconds.

This issue happens on both Palmetto and Romulus with latest OpenBMC (e.g. cea235afb), and have been there for a while.

Note: this issue is created here because I do not know if it should be created at host ipmid or net ipmid repo.
If we have figured out the root cause, it may be moved to related repo.

nest1ing, 2019-01-17:
This issue happens on Vesnin too.

gkeishin, 2019-01-22:
Yeah, I have noticed it to in recent Master code when resetting BMC.

edtanous, 2019-01-23:
Our validation team has noticed this as well.  Unclear as to the root cause at this point.

geissonator, 2019-01-25:
Related to issue that zerotierone hit with new systemd? https://github.com/zerotier/ZeroTierOne/issues/738

geissonator, 2019-02-05:
This continues to add an extra minute to all of our HW CI runs :(

vmauery, 2019-02-06:
Autobump https://gerrit.openbmc-project.xyz/c/openbmc/meta-phosphor/+/17891 should have fixed this.

nest1ing, 2019-02-07:
On VESNIN I see now things like this:
```
[  OK  ] Removed slice system-org.openbmc.control.Power.slice.
         Stopping Network Service...
[  OK  ] Stopped Network Service.
[   ***] A stop job is running for Phosphor Inband IPMI (17s / 1min 30s)
```
It lasts about 30 seconds.

mine260309, 2019-02-11:
> On VESNIN I see now things like this:
> 
> ```
> [  OK  ] Removed slice system-org.openbmc.control.Power.slice.
>          Stopping Network Service...
> [  OK  ] Stopped Network Service.
> [   ***] A stop job is running for Phosphor Inband IPMI (17s / 1min 30s)
> ```
> It lasts about 30 seconds.

The same on Romulus as well.
Inband IPMI takes about 30s, and network ipmid takes about 1min 30s.

mine260309, 2019-03-14:
With OpenBMC e7428cb, the net-ipmid seems OK, but inband ipmid still takes about 30s to exit.

@vmauery Could you help to check inband ipmid as well?
And could you share which commit in net ipmid fixes the issue? Should the same fix apply to inband ipmid?

amboar, 2019-03-20:
phosphor-host-ipmid taking 30s to terminate is likely because openpower-host-ipmi-flash is blocking SIGTERM until the host timeout expires. At that point it should unblock SIGTERM, which will cause ipmid to then cleanly shut down.

geissonator, 2019-03-26:
Is this something you'll be able to optimize @amboar ?  Still seeing it in the 2.7.0-dev-170-g8f73848 level of code (all the latest ipmi and hiomap changes).  I see it hitting the max 1min 36s too, not 30s.

amboar, 2019-03-27:
So I don't know where 1min 36s comes from; it's possible that ipmid is not unblocking sigterm. @anoo1 has been working with this to get the phosphor-host-ipmi refactor in, so she'd be the best person to ask at this point. As for optimising, if ipmid is aware of the host state then we could avoid sending the SEL and thus avoid the timeout.

gkeishin, 2019-04-16:
Did a trial here to see whats going out there.. 

```
/etc/systemd/journald.conf
ForwardToConsole=yes 
```
rebooted BMC and I see this where the reboot was taking time .. it taking time spending shutting down the host services
```
[  OK  ] Stopped Network Service.
[    **] A stop job is running for Phosphor Inband IPMI (36s / 1min 44s)
[  248.991147] ipmid[1429]: Host control timeout hit!
[**    ] A stop job is running for Phosphor Inband IPMI (1min 44s / 1min 44s)
[***   ] A stop job is running for Phosphor Inband IPMI (1min 44s / 3min 14s)
[  317.004216] systemd[1]: phosphor-ipmi-host.service: Killing process 1429 (ipmid) with signal SIGKILL.
[  317.081965] systemd[1]: phosphor-ipmi-host.service: Main process exited, code=killed, status=9/KILL
[  OK  ] Stopped Phosphor Inband IPMI.
```

anoo1, 2019-04-16:
Pushed out a fix: https://gerrit.openbmc-project.xyz/c/openbmc/openpower-host-ipmi-flash/+/20671

Tested killing the ipmi service but need to verify the reboot path. Will do that tomorrow.

closed 2019-04-17 by bradbishop

amboar, 2019-04-17:
Oh, I saw the patch and didn't see the comment about testing the reboot path. I've merged it as is, lets do a follow up patch if necessary.

anoo1, 2019-04-17:
Thanks Andrew, George and I have checked the reboot path is not delayed anymore with the merged patch.