openbmc/openbmc #3466: Witherspoon: Host services crashing out on BMC reset reload to runtime when Host is booted

gkeishin, 2019-01-17:
Boot the host and reset BMC, 

Post reset Host remains booted  though but things becomes bad in subsequent power on's

and we see these services 
```
root@witherspoon:~# systemctl status phosphor-max31785-msl@wspoon.service | cat
● phosphor-max31785-msl@wspoon.service - Phosphor MAX31785 MSL
   Loaded: loaded (/lib/systemd/system/obmc-standby.target.wants/../phosphor-max31785-msl@.service; static; vendor preset: enabled)
   Active: failed (Result: start-limit-hit) since Thu 2019-01-17 12:26:13 UTC; 37min ago
  Process: 1472 ExecStart=/usr/bin/env max31785-msl -p ${PATHS} -b ${BUS} -r ${REVISION} (code=exited, status=0/SUCCESS)
 Main PID: 1472 (code=exited, status=0/SUCCESS)

Jan 17 12:26:05 witherspoon systemd[1]: Starting Phosphor MAX31785 MSL...
Jan 17 12:26:13 witherspoon systemd[1]: Started Phosphor MAX31785 MSL.
Jan 17 12:26:18 witherspoon systemd[1]: phosphor-max31785-msl@wspoon.service: Start request repeated too quickly.
Jan 17 12:26:18 witherspoon systemd[1]: phosphor-max31785-msl@wspoon.service: Failed with result 'start-limit-hit'.
Jan 17 12:26:18 witherspoon systemd[1]: Failed to start Phosphor MAX31785 MSL.
root@witherspoon:~# systemctl status op-pnor-msl.service | cat
● op-pnor-msl.service - OpenPOWER PNOR Minimum Ship Level Verification
   Loaded: loaded (/lib/systemd/system/op-pnor-msl.service; enabled; vendor preset: enabled)
   Active: failed (Result: start-limit-hit) since Thu 2019-01-17 12:26:13 UTC; 37min ago
  Process: 1485 ExecStart=/usr/bin/env openpower-pnor-msl (code=exited, status=0/SUCCESS)
 Main PID: 1485 (code=exited, status=0/SUCCESS)

Jan 17 12:26:06 witherspoon systemd[1]: Starting OpenPOWER PNOR Minimum Ship Level Verification...
Jan 17 12:26:13 witherspoon systemd[1]: Started OpenPOWER PNOR Minimum Ship Level Verification.
Jan 17 12:26:18 witherspoon systemd[1]: op-pnor-msl.service: Start request repeated too quickly.
Jan 17 12:26:18 witherspoon systemd[1]: op-pnor-msl.service: Failed with result 'start-limit-hit'.
Jan 17 12:26:18 witherspoon systemd[1]: Failed to start OpenPOWER PNOR Minimum Ship Level Verification.
root@witherspoon:~# systemctl status op-vpd-parser.service | cat
● op-vpd-parser.service - Read OpenPOWER-format VPD from EEPROM
   Loaded: loaded (/lib/systemd/system/op-vpd-parser.service; static; vendor preset: enabled)
   Active: failed (Result: start-limit-hit) since Thu 2019-01-17 12:26:14 UTC; 37min ago
  Process: 1500 ExecStart=/usr/bin/env openpower-read-vpd --vpd $EEPROM --fru $FRUS --object $PATHS (code=exited, status=0/SUCCESS)
 Main PID: 1500 (code=exited, status=0/SUCCESS)

Jan 17 12:26:08 witherspoon systemd[1]: Starting Read OpenPOWER-format VPD from EEPROM...
Jan 17 12:26:14 witherspoon systemd[1]: Started Read OpenPOWER-format VPD from EEPROM.
Jan 17 12:26:18 witherspoon systemd[1]: op-vpd-parser.service: Start request repeated too quickly.
Jan 17 12:26:18 witherspoon systemd[1]: op-vpd-parser.service: Failed with result 'start-limit-hit'.
Jan 17 12:26:19 witherspoon systemd[1]: Failed to start Read OpenPOWER-format VPD from EEPROM.
root@witherspoon:~#
```
- [x] Journald 
[Service_failure_rr.txt](https://github.com/openbmc/openbmc/files/2768999/Service_failure_rr.txt)

gkeishin, 2019-01-17:
```
*Master Tagged 2.6.0.rc1*
>Build: https://openpower.xyz/job/openbmc-build/1941/
>BMC Version: 2.6.0-rc1-63-gcea235a
>BMC Kernel: Linux witherspoon 4.19.15-4f641ea0f1ec3ec5fa667ee9825e8bb5967da8a0 #1 Wed Jan 16 03:11:51 UTC 2019 armv6l GNU/Linux
```

geissonator, 2019-01-21:
George confirmed it was something in this build that introduced this:
```
Ensure host-check runs after ipmid (commit: fa46cac) (detail / githubweb)
systemd: Add support for static neighbors (commit: 6b8d479) (detail / githubweb)
phosphor: Move ipmitool fork to a set of patches (commit: d9aea6f) (detail / githubweb)
linux-aspeed: Move to 4.19.15 (commit: 0ea945b) (detail / githubweb)
openssl: Don't add c_rehash (commit: 2db4f50) (detail / githubweb)
```

geissonator, 2019-01-22:
I've got a couple of theories on this one.

Something is causing the oneshot services to restart on the way to standby, specifically when we're doing a bmc reset with the host on.

One theory is that the phosphor-reset-host-running@.service calls ```systemctl start obmc-host-start@%i.target``` which has a ```Wants=obmc-standby.target``` in it. It's almost like we double start obmc-standb.target and all of the oneshot services in it.

Another theory is something like https://github.com/systemd/systemd/issues/2775 where a service is failing and restarting the obmc-standby.target.

Whatever it is, I've found a fix to be to add the ```RemainAfterExit=yes``` to each of the oneshot services in the obmc-standby.target. None of them should be executed more then once so the change makes sense. I'm not sure if we pulled something new in from systemd or if our timing changed somewhere or what suddenly caused this to be an issue though. Wondering if this merits some more prototyping to narrow down the issue and potentially an upstream bug or if they'd just tell us the ```RemainAfterExit=yes``` is required for our situation if we don't want services restarting.

geissonator, 2019-01-23:
The oneshot fixes over in https://gerrit.openbmc-project.xyz/#/q/topic:oneshot+(status:open+OR+status:merged)

geissonator, 2019-01-28:
Did not get too root cause on what's starting the services twice in the standby target but moving them to remain after exit fixed this issue.

closed 2019-01-28 by geissonator