openbmc/openbmc #3467: Witherspoon: org.open_power.Host.Boot.Error.WatchdogTimedOut  during boot testing

gkeishin, 2019-01-17:
During boot testing, we see these failures resulting in Power on/off failure when.

```
        "/xyz/openbmc_project/logging/entry/6": {
            "AdditionalData": [
                "_PID=2362"
            ],
            "Description": "Hostboot has become unresponsive",
            "EventID": "FQPSPCR0023M",
            "Id": 6,
            "Message": "org.open_power.Host.Boot.Error.WatchdogTimedOut",
            "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
            "Resolved": false,
            "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
            "Timestamp": 1547719467962,
            "Version": "2.6.0-rc1-63-gcea235a",
            "associations": []
        },
```

```
 |------------------------------------------------------------------------------|
 |                       Platform Event Log - 0x8900003B                        |
 |------------------------------------------------------------------------------|
 |                                Private Header                                |
 |------------------------------------------------------------------------------|
 | Section Version          : 1                                                 |
 | Sub-section type         : 0                                                 |
 | Created by               : ipmi                                              |
 | Created at               : DE/AD/BEEF DE:AD:FF                               |
 | Committed at             : DE/AD/BEEF DE:AD:FF                               |
 | Creator Subsystem        : HostBoot                                          |
 | CSSVER                   :                                                   |
 | Platform Log Id          : 0x8900003B                                        |
 | Entry Id                 : 0x8900003B                                        |
 | Total Log Size           : 1272                                              |
 |------------------------------------------------------------------------------|
 |                                 User Header                                  |
 |------------------------------------------------------------------------------|
 | Section Version          : 1                                                 |
 | Sub-section type         : 0                                                 |
 | Log Committed by         : tmgt                                              |
 | Subsystem                : HostBoot                                          |
 | Event Scope              : Single Platform                                   |
 | Event Severity           : Unrecoverable Error                               |
 | Event Type               : Not Applicable                                    |
 | Return Code              : 0x00000000                                        |
 | Action Flags             : None declared                                     |
 |------------------------------------------------------------------------------|
 |                        Primary System Reference Code                         |
 |------------------------------------------------------------------------------|
 | Section Version          : 1                                                 |
 | Sub-section type         : 1                                                 |
 | Created by               : ipmi                                              |
 | SRC Format               : 0xE0                                              |
 | SRC Version              : 0x02                                              |
 | Virtual Progress SRC     : False                                             |
 | I5/OS Service Event Bit  : False                                             |
 | Hypervisor Dump Initiated: False                                             |
 | Power Control Net Fault  : False                                             |
 | Deconfigured             : False                                             |
 | Garded                   : False                                             |
 | Acknowledged             : False                                             |
 |                                                                              |
 | Valid Word Count         : 0x09                                              |
 | Reference Code           : BC8A2502                                          |
 | Hex Words 2 - 5          : 000000E0 00000500 00000000 00200000               |
 | Hex Words 6 - 9          : FFFFFFFF 00040030 00000000 0000000A               |
 |                                                                              |
 | devdesc                  : ipmi_msg() failed                                 |
[... 33 lines cut ...]
 00000216.200070414|    0|IPMI            | 490|sel: <<send_esel o_err=89000037, o_cc=xff, sel recID=x00, esel recID=x00
 00000216.200077269|    0|IPMI            | 231|sel: E>DELETING l_err 0x89000037
 00000216.200095056|    0|IPMI            | 249|sel: <<process_esel
 00000216.200100760|    0|IPMI            | 129|sel: <<sendESEL
 00000216.200137683|    0|IPMI            |1582|sendGpuSensorStatus(0x17D8, 0xCB, Target 0x50000, status: 0)
 00000216.200145389|    0|IPMI            |  93|rt: calling sync 4:30  len=10
 00000221.260025863|    0|IPMI            | 109|rt: E>Failed sending ipmi msg (10:30) to bmc rc: -1.
 00000221.260064330|    0|IPMI            | 225|error returned from sendSetSensorReading() for sensor number 0xcb
 00000221.260156537|    0|IPMI            |  94|sel: >>sendESEL() 1
 00000221.260165525|    0|IPMI            | 149|sel: >>process_esel
 00000221.260171438|    0|IPMI            | 155|sel: >>sel list size 1
 00000221.260177659|    0|IPMI            | 260|sel: >>send_esel
 00000221.260184805|    0|IPMI            |  93|rt: calling sync a:42  len=0
 00000226.310026245|    0|IPMI            | 109|rt: E>Failed sending ipmi msg (28:42) to bmc rc: -1.
 00000226.310065919|    0|IPMI            | 295|sel: E>error from reserve_sel
 00000226.310072173|    0|IPMI            | 490|sel: <<send_esel o_err=89000039, o_cc=xff, sel recID=x00, esel recID=x00
 00000226.310078559|    0|IPMI            | 231|sel: E>DELETING l_err 0x89000039
 00000226.310095605|    0|IPMI            | 249|sel: <<process_esel
 00000226.310101511|    0|IPMI            | 129|sel: <<sendESEL
 00000226.310209616|    0|IPMI            |  93|rt: calling sync 4:30  len=10
 00000231.350025704|    0|IPMI            | 109|rt: E>Failed sending ipmi msg (10:30) to bmc rc: -1.
```

- [x] Journald 
[Boot_failure_log.txt](https://github.com/openbmc/openbmc/files/2769030/Boot_failure_log.txt)

gkeishin, 2019-01-17:
```
*Master Tagged 2.6.0.rc1*
>Build: https://openpower.xyz/job/openbmc-build/1941/
>BMC Version: 2.6.0-rc1-63-gcea235a
>BMC Kernel: Linux witherspoon 4.19.15-4f641ea0f1ec3ec5fa667ee9825e8bb5967da8a0 #1 Wed Jan 16 03:11:51 UTC 2019 armv6l GNU/Linux
```

geissonator, 2019-01-28:
Still seeing this one @gkeishin ?

closed 2019-07-09 by gkeishin