openbmc/openbmc #3468: phosphor-user-manager doesn't work properly after upgrade OpenBMC

nest1ing, 2019-01-17:
Upgrade from an old version of the OpenBMC (where there were not groups like `priv-*`) to the latest one makes it impossible to authorize the `root` user in `ipmi`.

This issue happens on platforms which use the static filesystem layout (e.g. romulus, palmetto, vesnin).

It happens because the old `/etc/group` file is saved during the update and as a result, it doesn't contain new required groups.

The `phosphor-user-manager ` doesn't work properly without these groups and the `phosphor-host-ipmid` is unable to authorize users.

geissonator, 2019-01-17:
Yeah, it's not forward compatible with a persistent file system. Either need to factory reset your system or run a script like this to get things updated:
```
#!/bin/bash -e

# Create groups if not available
for i in web ipmi redfish priv-admin priv-operator priv-user priv-callback; do
    if grep -q $i /etc/group; then
        echo "$i already exists"
    else
        echo "$i does not exist, add it"
        groupadd -f $i
    fi
done

# Root needs to be a member of these groups
for i in ipmi web redfish priv-admin; do
    if id -nG root | grep -q $i; then
        echo "root already in $i"
    else
        echo "root not in group $i, add it"
        usermod -a -G $i root
    fi
done
```

nest1ing, 2019-01-17:
It doesn't work for customers, because:
  * We can't modify the update procedure in the already installed  version of the OpenBMC.
  * Customers may not have access to root-shell to run this script.

However, we can put this script to the new firmware and run it at first boot, but this way doesn't look good and it will be hard to support.

Does anyone have another ideas?

closed 2019-08-18 by stale[bot]