openbmc/openbmc #3470: GUI: IE able to login but circling icon getting displayed not able to use

sivassrr, 2019-01-22:


sivassrr, 2019-01-22:
```root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.6.0-rc1"
VERSION_ID="2.6.0-rc1-77-g84748a7"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.6.0-rc1"
BUILD_ID="2.6.0-rc1"```

```root@witherspoon:/var/log/nginx# cat error.log
2019/01/09 06:26:54 [warn] 1143#1143: the "ssl" directive is deprecated, use the "listen ... ssl" directive instead in /etc/nginx/nginx.conf:61
2019/01/09 06:27:03 [warn] 1209#1209: the "ssl" directive is deprecated, use the "listen ... ssl" directive instead in /etc/nginx/nginx.conf:61
2019/01/09 06:36:09 [warn] 1140#1140: the "ssl" directive is deprecated, use the "listen ... ssl" directive instead in /etc/nginx/nginx.conf:61
2019/01/09 06:36:09 [warn] 1141#1141: the "ssl" directive is deprecated, use the "listen ... ssl" directive instead in /etc/nginx/nginx.conf:61
2019/01/16 04:53:12 [warn] 1147#1147: the "ssl" directive is deprecated, use the "listen ... ssl" directive instead in /etc/nginx/nginx.conf:61
2019/01/16 04:53:12 [warn] 1154#1154: the "ssl" directive is deprecated, use the "listen ... ssl" directive instead in /etc/nginx/nginx.conf:61
2019/01/22 04:10:02 [warn] 1149#1149: the "ssl" directive is deprecated, use the "listen ... ssl" directive instead in /etc/nginx/nginx.conf:61
2019/01/22 04:10:02 [warn] 1150#1150: the "ssl" directive is deprecated, use the "listen ... ssl" directive instead in /etc/nginx/nginx.conf:61
root@witherspoon:/var/log/nginx# date
Tue Jan 22 05:48:12 UTC 2019```

sivassrr, 2019-01-22:
Below is the screen which we get even after 25 mins of login using IE browser. 

[image]

@gtmills aware of this issue and working on it.

sunleaf2002, 2019-03-06:
[image]
me too , I use ie v11 , event-log page is also, i donot know why.
[image]

sivassrr, 2019-03-07:
@gtmills Is there a plan to support this?  Was told that IE wouldnt be supported. Could you pls confirm @gtmills. We can close it accordingly then. Thank you in advance.

gtmills, 2019-03-07:
IBM has no plans to support IE. We plan to support Chrome, Firefox, Safari and Edge browsers and will work to maintain that support upstream. Those 4 all currently work with the webUI. 

I think @edtanous still uses Internet Explorer as his primary browser.  😁😂 Ed, do you have any requirement to support IE, if not, I will close.

gtmills, 2019-03-18:
Closing issue. Thanks for submitting Sivas.

closed 2019-03-18 by gtmills