openbmc/openbmc #3477: Fails to power on due to bmc file system error

sivassrr, 2019-02-04:


sivassrr, 2019-02-04:
BMC Details:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.6.0-rc1"
VERSION_ID="2.6.0-rc1-135-g16e11aa"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.6.0-rc1"
BUILD_ID="2.6.0-rc1"
OPENBMC_TARGET_MACHINE="witherspoon"
root@witherspoon:~# cat /etc/timestamp
20190201202932```

sivassrr, 2019-02-04:
BMC File system error details 
[image]

sivassrr, 2019-02-04:
Journalctl output 
[failed2iplbmcfilesystemerror.txt](https://github.com/openbmc/openbmc/files/2828425/failed2iplbmcfilesystemerror.txt)

sivassrr, 2019-02-04:
@matt pointed this error is due to FSI error. So renamed the title accordingly.

spinler, 2019-02-04:
"Something bad" happened to that system.  Could probably stand to get its AC cycled.

spinler, 2019-02-05:
The journal is mostly filled with REST logging, so there isn't a lot of history, but I can see a power off is basically happening right on top of the power on, which causes this.
There is nothing further that can be debugged here with the attached journal.

closed 2019-02-05 by spinler