openbmc/openbmc #3479: Witherspoon : pcie-poweroff@0.service: Failed with result 'exit-code

gkeishin, 2019-02-05:
```
Feb 05 13:40:25 witherspoon phosphor-fru-fault-monitor[1301]: /xyz/openbmc_project/logging/entry/3 created
Feb 05 13:40:25 witherspoon systemd[1]: pcie-poweroff@0.service: Main process exited, code=exited, status=255/n/a
Feb 05 13:40:25 witherspoon systemd[1]: pcie-poweroff@0.service: Failed with result 'exit-code'.
Feb 05 13:40:25 witherspoon systemd[1]: Failed to start POWER9 PCIe Power-off Workaround.
```
```
------------------------------------------------------------------------ 
DRIVER INFO : FW Level	Executed : cat /etc/os-release
------------------------------------------------------------------------ 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.6.0-rc1"
VERSION_ID="2.6.0-rc1-151-gf8d0724"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.6.0-rc1"
BUILD_ID="2.6.0-rc1"
OPENBMC_TARGET_MACHINE="witherspoon"
```

spinler, 2019-02-05:
This is normal on a power off.  The service has to be stopped with a KILL so that during runtime when it stops it will trip the watchdog to get high speed fans.

closed 2019-02-05 by spinler

gkeishin, 2019-02-05:
For the error log for the PCIE we saw here @spinler got a fix https://gerrit.openbmc-project.xyz/#/c/openbmc/openpower-proc-control/+/17988/