openbmc/openbmc #3480: gitsm based recipes failing to compile with latest yocto on certain systems

geissonator, 2019-02-08:
We had a pretty lengthy discussion in IRC on this but summary is that with the latest yocto update (https://gerrit.openbmc-project.xyz/#/c/openbmc/openbmc/+/18033/) we started hitting compile failures on certain build systems.  First it was cli11.bb. We just retried enough that it seemed to finally fix itself.  A bit later we then hit it with dbus-broker.bb. One system fixed itself (builder1) but builder4 would not, so we had to disable it from the build queue. Both recipes use a "gitsm" SRC_URI so the theory is it's related to that function.

We're also seeing this issue on our internal build servers so there's def something going on.

http://git.yoctoproject.org/cgit.cgi/poky/commit/?id=1979d9162a335b7d4718a6697ca57bfa16237f0e  looked related (either to fix it or to cause this issue) but @bradbishop was going to dig a bit more.

Here's the dbus-broker fail data on builder4.
```
$ cat /tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/temp/log.do_unpack.75604
DEBUG: Executing python function do_unpack
DEBUG: Executing python function base_do_unpack
DEBUG: Running export PSEUDO_DISABLED=1; export PATH="/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin/python3-native:/home/jenkins-op/241174337/openbmc/scripts:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot/usr/bin/crossscripts:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/sbin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/sbin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/bin:/home/jenkins-op/241174337/openbmc/poky/bitbake/bin:/tmp/openbmc/hosttools"; export HOME="/home/jenkins-op"; git -c core.fsyncobjectfiles=0 branch --contains fc874afa0992d0c75ec25acb43d344679f0ee7d2 --list master 2> /dev/null | wc -l
DEBUG: Running export PSEUDO_DISABLED=1; export PATH="/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin/python3-native:/home/jenkins-op/241174337/openbmc/scripts:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot/usr/bin/crossscripts:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/sbin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/sbin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/bin:/home/jenkins-op/241174337/openbmc/poky/bitbake/bin:/tmp/openbmc/hosttools"; export HOME="/home/jenkins-op"; git -c core.fsyncobjectfiles=0 clone -s -n /home/jenkins-op/bitbake_downloads/git2/github.com.bus1.dbus-broker/ /tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/git/
DEBUG: Running export PSEUDO_DISABLED=1; export PATH="/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin/python3-native:/home/jenkins-op/241174337/openbmc/scripts:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot/usr/bin/crossscripts:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/sbin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/sbin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/bin:/home/jenkins-op/241174337/openbmc/poky/bitbake/bin:/tmp/openbmc/hosttools"; export HOME="/home/jenkins-op"; git -c core.fsyncobjectfiles=0 remote set-url origin git://github.com/bus1/dbus-broker
DEBUG: Running export PSEUDO_DISABLED=1; export PATH="/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin/python3-native:/home/jenkins-op/241174337/openbmc/scripts:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot/usr/bin/crossscripts:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/sbin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/sbin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/bin:/home/jenkins-op/241174337/openbmc/poky/bitbake/bin:/tmp/openbmc/hosttools"; export HOME="/home/jenkins-op"; git -c core.fsyncobjectfiles=0 checkout -B master fc874afa0992d0c75ec25acb43d344679f0ee7d2
DEBUG: Running export PSEUDO_DISABLED=1; export PATH="/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin/python3-native:/home/jenkins-op/241174337/openbmc/scripts:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot/usr/bin/crossscripts:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/sbin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/sbin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/bin:/home/jenkins-op/241174337/openbmc/poky/bitbake/bin:/tmp/openbmc/hosttools"; export HOME="/home/jenkins-op"; git -c core.fsyncobjectfiles=0 branch master --set-upstream-to origin/master
DEBUG: Running export PSEUDO_DISABLED=1; export PATH="/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin/python3-native:/home/jenkins-op/241174337/openbmc/scripts:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot/usr/bin/crossscripts:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/sbin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/sbin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/bin:/home/jenkins-op/241174337/openbmc/poky/bitbake/bin:/tmp/openbmc/hosttools"; export HOME="/home/jenkins-op"; git -c core.fsyncobjectfiles=0 show fc874afa0992d0c75ec25acb43d344679f0ee7d2:.gitmodules
DEBUG: Running export PSEUDO_DISABLED=1; export PATH="/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin/python3-native:/home/jenkins-op/241174337/openbmc/scripts:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/dbus-broker/16+gitAUTOINC+fc874afa09-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi:/tm [... cut ...]
fatal: unable to read tree 4366d9ee295ab1163a6ed62a5ee18ee7b059fbb5
Unable to checkout 'b245df543717ee19ad398f8139447e3807c11c96' in submodule path 'subprojects/c-dvar/subprojects/c-utf8'
fatal: unable to read tree 4366d9ee295ab1163a6ed62a5ee18ee7b059fbb5
Failed to recurse into submodule path 'subprojects/c-dvar'
Unable to checkout 'b245df543717ee19ad398f8139447e3807c11c96' in submodule path 'subprojects/c-utf8'

DEBUG: Python function base_do_unpack finished
DEBUG: Python function do_unpack finished
ERROR: Function failed: base_do_unpack
```

geissonator, 2019-02-27:
There's been talk of this in the infrastructure workgroup and the IRC channel.  It's not specific to processor architecture, it has been seen on both x86 and POWER.  I'm getting more and more reports of it from people. @wak-google mentioned that downstream this hit them pretty hard and he's been working on some fixes/workarounds.

wak-google, 2019-02-28:
On vacation this week, hopefully I can spend a day and triage this next week.

wak-google, 2019-03-06:
Is there a known difference between the version of git on these failing machines vs the working ones? Differences in the git config? We are using an impure git from the host system so maybe that is why.

geissonator, 2019-03-07:
Brad and I did a bit experimentation with this and got our build system to the same git level that was working for him. It didn't help. In theory the working system and non-working systems are all using the same docker container so it should be pretty similar.

geissonator, 2021-07-28:
This is back with our latest yocto bump: https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/45207

Build machines just intermittently hitting errors like this:

```
22:38:05 ERROR: function2-4.2.0+gitAUTOINC+02ca99831d-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export PATH="/home/jenkins-slave/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/witherspoon/openbmc/scripts:/home/jenkins-slave/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/witherspoon/build/work/arm1176jzs-openbmc-linux-gnueabi/function2/4.2.0+gitAUTOINC+02ca99831d-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi:/home/jenkins-slave/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/witherspoon/build/work/arm1176jzs-openbmc-linux-gnueabi/function2/4.2.0+gitAUTOINC+02ca99831d-r0/recipe-sysroot/usr/bin/crossscripts:/home/jenkins-slave/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/witherspoon/build/work/arm1176jzs-openbmc-linux-gnueabi/function2/4.2.0+gitAUTOINC+02ca99831d-r0/recipe-sysroot-native/usr/sbin:/home/jenkins-slave/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/witherspoon/build/work/arm1176jzs-openbmc-linux-gnueabi/function2/4.2.0+gitAUTOINC+02ca99831d-r0/recipe-sysroot-native/usr/bin:/home/jenkins-slave/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/witherspoon/build/work/arm1176jzs-openbmc-linux-gnueabi/function2/4.2.0+gitAUTOINC+02ca99831d-r0/recipe-sysroot-native/sbin:/home/jenkins-slave/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/witherspoon/build/work/arm1176jzs-openbmc-linux-gnueabi/function2/4.2.0+gitAUTOINC+02ca99831d-r0/recipe-sysroot-native/bin:/home/jenkins-slave/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/witherspoon/openbmc/poky/bitbake/bin:/home/jenkins-slave/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/witherspoon/build/hosttools"; export HOME="/home/jenkins-slave"; LANG=C git -c core.fsyncobjectfiles=0 -c gc.autoDetach=false fetch -f --progress git://github.com/Naios/function2 refs/*:refs/* failed with exit code 128, output:
22:38:05 error: Could not read 5b8e6defe76d4757f31b6eca09aca30ebb9c5cd0
22:38:05 remote: Enumerating objects: 76, done.        
22:38:05 remote: Counting objects:   1% (1/76)        
remote: Counting objects:   2% (2/76)        
remote: Counting objects:   3% (3/76)        
remote: Counting objects:   5% (4/76)        
remote: Counting objects:   6% (5/76)        
remote: Counting objects:   7% (6/76)        
remote: Counting objects:   9% (7/76)        
remote: Counting objects:  10% (8/76)        
remote: Counting objects:  11% (9/76)        
remote: Counting objects:  13% (10/76)        
remote: Counting objects:  14% (11/76)        
remote: Counting objects:  15% (12/76)        
remote: Counting objects:  17% (13/76)        
remote: Counting objects:  18% (14/76)        
remote: Counting objects:  19% (15/76)        
remote: Counting objects:  21% (16/76)        
remote: Counting objects:  22% (17/76)        
remote: Counting objects:  23% (18/76)        
remote: Counting objects:  25% (19/76)        
remote: Counting objects:  26% (20/76)        
remote: Counting objects:  27% (21/76)        
remote: Counting objects:  28% (22/76)        
remote: Counting objects:  30% (23/76)        
remote: Counting objects:  31% (24/76)        
remote: Counting objects:  32% (25/76)        
remote: Counting objects:  34% (26/76)        
remote: Counting objects:  35% (27/76)        
remote: Counting objects:  36% (28/76)        
remote: Counting objects:  38% (29/76)        
remote: Counting objects:  39% (30/76)        
remote: Counting objects:  40% (31/76)        
remote: Counting objects:  42% (32/76)        
remote: Counting objects:  43% (33/76)        
remote: Counting objects:  44% (34/76)        
remote: Counting objects:  46% (35/76)        
remote: Counting objects:  47% (36/76)        
remote: Counting objects:  48% (37/76)        
remote: Counting objects:  50% (38/76)        
remote: Counting objects:  51% (39/76)        
remote: Counting objects:  52% (40/76)        
remote: Counting objects:  53% (41/76)        
remote: Counting objects:  55% (42/76)        
remote: Counting objects:  56% (43/76)        
remote: Counting objects:  57% (44/76)        
remote: Counting objects:  59% (45/76)        
remote: Counting objects:  60% (46/76)        
remote: Counting objects:  61% (47/76)        
remote: Counting objects:  63% (48/76)        
remote: Counting objects:  64% (49/76)        
remote: Counting objects:  65% (50/76)        
remote: Counting objects:  67% (51/76)        
remote: Counting objects:  68% (52/76)        
remote: Counting objects:  69% (53/76)        
remote: Counting objects:  71% (54/76)        
remote: Counting objects:  72% (55/76)        
remote: Counting objects:  73% (56/76)        
remote: Counting objects:  75% (57/76)        
remote: Counting objects:  76% (58/76)        
remote: Counting objects:  77% (59/76)        
remote: Counting objects:  78% (60/76)        
remote: Counting objects:  80% (61/76)        
remote: Counting objects:  81% (62/76)        
[... 39 lines cut ...]
remote: Compressing objects:  62% (25/40)        
remote: Compressing objects:  65% (26/40)        
remote: Compressing objects:  67% (27/40)        
remote: Compressing objects:  70% (28/40)        
remote: Compressing objects:  72% (29/40)        
remote: Compressing objects:  75% (30/40)        
[... cut ...]
remote: Compressing objects: 100% (40/40), done.        
22:38:05 remote: Total 76 (delta 34), reused 35 (delta 14), pack-reused 0        
22:38:05 fatal: unresolved deltas left after unpacking
22:38:05 fatal: unpack-objects failed
22:38:05 
22:38:05 ERROR: function2-4.2.0+gitAUTOINC+02ca99831d-r0 do_fetch: Fetcher failure for URL: 'gitsm://github.com/Naios/function2'. Unable to fetch URL from any source.
```

I'm going to just keep resubmitting because that seemed to eventually fix it in the past.

geissonator, 2021-07-29:
Running the following against bitbake_downloads and bitbake_sharedstatecache on the 2 build machines having issues appears to have fixed the issue:
``` 
rm -rf `find ./ -name "*function2*" -print`
```

anischali, 2021-11-23:
In my case, cleaning the recipe resolves the probleme for me which was ovmf. it's the same than what @geissonator did with rm .....
`bitbake function -c clean && bitbake function -c cleanall`

williamspatrick, 2023-09-25:
I think upstream has made many improvements to gitsm at this point.  Please reopen with newer failure examples if this is still a problem.

closed 2023-09-25 by williamspatrick