openbmc/openbmc #3481: meta-aspeed?  local.conf?

paigeadelethompson, 2019-02-09:
```
erratic@molly:~/openbmc> find  -name "*local.conf*"
./meta-phosphor/conf/local.conf.sample
./meta-inspur/meta-on5263m5/conf/local.conf.sample
./meta-ibm/meta-palmetto/conf/local.conf.sample
./meta-ibm/meta-witherspoon/conf/local.conf.sample
./meta-ibm/meta-z/conf/local.conf.sample
./meta-ibm/meta-romulus/conf/local.conf.sample
./meta-quanta/meta-q71l/conf/local.conf.sample
./meta-quanta/meta-gsj/conf/local.conf.sample
./meta-quanta/meta-f0b/conf/local.conf.sample
./meta-portwell/meta-neptune/conf/local.conf.sample
./meta-qualcomm/meta-centriq2400-rep/conf/local.conf.sample
./meta-inventec/meta-lanyang/conf/local.conf.sample
./meta-facebook/meta-tiogapass/conf/local.conf.sample
./meta-mellanox/meta-msn/conf/local.conf.sample
./meta-ingrasys/meta-zaius/conf/local.conf.sample
./poky/meta-poky/conf/local.conf.sample.extended
./poky/meta-poky/conf/local.conf.sample
./meta-evb/meta-evb-aspeed/meta-evb-ast2500/conf/local.conf.sample
./meta-evb/meta-evb-enclustra/meta-evb-zx3-pm3/conf/local.conf.sample
./meta-evb/meta-evb-nuvoton/meta-evb-npcm750/conf/local.conf.sample
./meta-evb/meta-evb-raspberrypi/conf/local.conf.sample
./meta-intel/meta-s2600wf/conf/local.conf.sample
./meta-hxt/meta-stardragon4800-rep2/conf/local.conf.sample
erratic@molly:~/openbmc> 
erratic@molly:~/openbmc> find meta-aspeed/ -name "*local.conf*"
erratic@molly:~/openbmc>
```
I just wanna compile it for an ast2400 chip

nest1ing, 2019-02-09:
You can look at `meta-ibm/meta-palmetto`. At the moment there is only one hardware based on the `Aspeed-2400`. Let the other guys fix me if I wrong.

mikeBashStuff, 2019-03-19:
I have the very same question - is there any place you can fetch the missing conf files for aspeed socs or are there any docs on how you can create some? By following instructions from README.md you just end up with `oe-setup-builddir` complaining about missing `local.conf.sample` and `bblayers.conf.sample` and then when you try to execute `bitbake` against the `obmc-phosphor-image ` target it just fails:

> ERROR: Unable to start bitbake server (None)
ERROR: Server log for this session (/openbmc/build/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 4416 at 2019-03-19 16:57:57.886242 ---
ERROR: Traceback (most recent call last):
  File "/openbmc/poky/bitbake/lib/bb/cookerdata.py", line 179, in wrapped
    return func(fn, *args)
  File "/openbmc/poky/bitbake/lib/bb/cookerdata.py", line 204, in parse_config_file
    return bb.parse.handle(fn, data, include)
  File "/openbmc/poky/bitbake/lib/bb/parse/__init__.py", line 117, in handle
    return h['handle'](fn, data, include)
  File "/root/git_repos/openbmc/poky/bitbake/lib/bb/parse/parse_py/ConfHandler.py", line 134, in handle
    abs_fn = resolve_file(fn, data)
  File "/openbmc/poky/bitbake/lib/bb/parse/__init__.py", line 135, in resolve_file
    raise IOError(errno.ENOENT, "file %s not found in %s" % (fn, bbpath))
FileNotFoundError: [Errno 2] file conf/bitbake.conf not found in /openbmc/build

@nest1ing by looking into `meta-ibm/meta-palmetto` did you mean to just copy needed files or is there something more to it?

nest1ing, 2019-03-20:
> @nest1ing by looking into meta-ibm/meta-palmetto did you mean to just copy needed files or is there something more to it?

I meant that the `meta-palmetto` is a working set of config files and recipes additions for the real hardware based on `Aspeed-2400` and it might be used as an example.
But you should understand that it contains many features specific for the managed `openpower-p8` based host.
Probably not each of them are required to you.

In your copies of the `local.conf.sample` and `bblayers.conf.samle` you should change the `palmetto` specific features with your own ones.
For example:
 * in `local.conf.sample` you should change `MACHINE` and maybe `DISTRO`
 * in `'bblayers.conf.sample` you should specify your required paths to your firmware.

Also see https://github.com/openbmc/docs/blob/master/development/add-new-system.md

mikeBashStuff, 2019-03-20:
@nest1ing Great, thanks for the feedback, I will look into it!

closed 2019-10-19 by stale[bot]