openbmc/openbmc #3484: IPMI：the phosphor-ipmi-host crashed due to too long response data

zhuysh1988, 2019-02-25:
I added an OEM cmd into phosphor-ipmi-host. There is no proble when I execute the command once。 But when I execute it multiple times at a small interval（less than 1s）, the service will crash。
When the response data is less than 64bits, there is no problem, but when the return value is longer than 64（e.g. 94）, I will encounter this problem. 
The phosphor-ipmi-host status as follows:
```
[[0;1;31m●[[0m phosphor-ipmi-host.service - Phosphor Inband IPMI
   Loaded: loaded (/lib/systemd/system/phosphor-ipmi-host.service; enabled; vendor preset: enabled)
   Active: [[0;1;31mfailed[[0m (Result: signal) since Sun 2019-02-17 03:21:12 UTC; 18s ago
  Process: 1586 ExecStart=/usr/bin/env ipmid [[0;1;31m(code=killed, signal=SEGV)[[0m
 Main PID: 1586 (code=killed, signal=SEGV)

Feb 17 03:21:12 superior systemd[1]: phosphor-ipmi-host.service: Service RestartSec=1s expired, scheduling restart.
Feb 17 03:21:12 superior systemd[1]: phosphor-ipmi-host.service: Scheduled restart job, restart counter is at 4.
Feb 17 03:21:12 superior systemd[1]: Stopped Phosphor Inband IPMI.
Feb 17 03:21:12 superior systemd[1]: [[0;1;39m[[0;1;31m[[0;1;39mphosphor-ipmi-host.service: Start request repeated too quickly.[[0m
Feb 17 03:21:12 superior systemd[1]: [[0;1;39m[[0;1;31m[[0;1;39mphosphor-ipmi-host.service: Failed with result 'signal'.[[0m
Feb 17 03:21:12 superior systemd[1]: [[0;1;31m[[0;1;39m[[0;1;31mFailed to start Phosphor Inband IPMI.[[0m
```
The journalctl logs as follows:
```
Feb 17 03:21:11 superior systemd[1]: phosphor-ipmi-host.service: Main process exited, code=killed, status=11/SEGV
Feb 17 03:21:11 superior systemd[1]: phosphor-ipmi-host.service: Failed with result 'signal'.
Feb 17 03:21:12 superior systemd[1]: phosphor-ipmi-host.service: Service RestartSec=1s expired, scheduling restart.
Feb 17 03:21:12 superior systemd[1]: phosphor-ipmi-host.service: Scheduled restart job, restart counter is at 4.
Feb 17 03:21:12 superior systemd[1]: phosphor-ipmi-host.service: Start request repeated too quickly.
Feb 17 03:21:12 superior systemd[1]: phosphor-ipmi-host.service: Failed with result 'signal'.
```

Anybody can help me?

mine260309, 2019-02-26:
The service crashed due to segment fault.
I suspect something is wrong in your oem command handler.
Please [use gdb to analyze the coredump][1], check the stack trace and you should know what's wrong.

[1]: https://github.com/openbmc/docs/blob/master/cheatsheet.md#gdb

zhuysh1988, 2019-04-26:
Reason: 
IPMI has defined : MAX_IPMI_BUFFER 64

closed 2019-04-26 by zhuysh1988