openbmc/openbmc #3486: openbmc/openbmc refactoring for reference platforms

krtaylor, 2019-02-25:
Split out platform meta from openbmc common and move to different repos

amboar, 2019-03-03:
What does this mean? We already have separate repos for various manufacturers. Is this an issue to implement a different set of mechanics for integrating those repos to perform a build?

Also, what reference platforms? Do we have one (or more)?

krtaylor, 2019-05-06:
Universal reference platforms would be a good thing for the project, this is also the potential process of modularizing the common and platform specific parts of the project.

For example:
IPMI refactoring was a very good thing, but it broke other platforms. This would help ensure that a set of platforms would work before code will merge.

closed 2019-12-05 by stale[bot]