openbmc/openbmc #3487: BMC password for IPMI interface did not be modified by rest

xuweibj, 2019-02-27:
After modified BMC password by rest, could not connect IPMI interface with the new password.
It seems the password for IPMI interface was not be modified. Is this a bug? Or is this as design?

Steps:
1. set BMC password to ``abcdefgh``
```
curl -k -c cjar -b cjar -X POST -H "Content-Type: application/json" https://10.5.14.100/xyz/openbmc_project/user/root/action/SetPassword -d '{"data": ["abcdefgh"]}'
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
```
2. check password setting
```
# curl -k -c cjar -b cjar -X POST -H "Content-Type: application/json" https://10.5.14.100/login -d '{"data": ["root", "abcdefgh"]}'
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}
```
3. run ``ipmitool`` command with new password to get info
```
# ipmitool -I lanplus -H 10.5.14.100 -P abcdefghc mc info
Error: Unable to establish IPMI v2 / RMCP+ session
```
4. run ``ipmitool`` command with original password
```
# ipmitool -I lanplus -H 10.5.14.100 -P 0penBmc mc info
Device ID                 : 0
Device Revision           : 0
Firmware Revision         : 2.00
IPMI Version              : 2.0
Manufacturer ID           : 42817
Manufacturer Name         : Unknown (0xA741)
Product ID                : 16975 (0x424f)
Product Name              : Unknown (0x424F)
Device Available          : yes
Provides Device SDRs      : yes
Additional Device Support :
    Sensor Device
    SEL Device
    FRU Inventory Device
    Chassis Device
Aux Firmware Rev Info     :
    0x00
    0x00
    0x00
    0x46
```

BMC Version: ibm-v2.0-0-r46-0-gbed584c

rthomaiy, 2019-02-27:
If you are not specifying -U <username> then it means, that you are using null username for the ipmi interface. As of the latest code, the same has been removed. But i see that you are getting succeeded without -U option.

can you try the same using -U root option in ipmitool

Note: I am also not sure, how setPassword rest API works for you. Earlier i saw the code change for SetPassword rest API to use PAM API directly, but i am not sure, whether it ever got merged @ratagupt to confirm. i.e. Phoshor-user-manager doesn't support SetPassword method any more, and password has to be updated using pam_chauthtok() directly.

xuweibj, 2019-02-28:
@rthomaiy 
I tried the command with ``-U``, but failed even password is correct.
```
# ipmitool -I lanplus -H 10.5.14.100 -U root -P 0penBmc mc info
Error: Unable to establish IPMI v2 / RMCP+ session

# ipmitool -I lanplus -H 10.5.14.100 -P 0penBmc mc info
Device ID                 : 0
Device Revision           : 0
Firmware Revision         : 2.00
IPMI Version              : 2.0
Manufacturer ID           : 42817
Manufacturer Name         : Unknown (0xA741)
Product ID                : 16975 (0x424f)
Product Name              : Unknown (0x424F)
Device Available          : yes
Provides Device SDRs      : yes
Additional Device Support :
    Sensor Device
    SEL Device
    FRU Inventory Device
    Chassis Device
Aux Firmware Rev Info     :
    0x00
    0x00
    0x00
    0x46
```
My current problem is after modified openbmc password by rest, I could use the new password connect openbmc by rest and could ssh to openbmc using it. But could not connect openbmc by ipmitool, the old password is OK.

So my question could the password for ipmi interface be modified? Could the password of openbmc for ipmi and rest/ssh be different?
Or is our BMC version too old for ipmi interface?

rthomaiy, 2019-02-28:
It's definitely old, because the latest one, doesn't support RMCP+ without -U option.

Can you issue 
ipmitool user list 1   post the output. Note: If there are no users in the list, it means it is old version that we didn't add root user as default in IPMI interface. You can add new user name and can try the login.

Note: adding @tomjoseph83  to comment on this IBM version.

mine260309, 2019-03-01:
> I tried the command with `-U`, but failed even password is correct.

I met such issue before, and likely you need to remove the `/etc/group` file, reboot, and try again.
When the user management feature is introduced, it has a new `/etc/group` in rofs, however, `/etc/group` is preserved during code update, so this file in `rwfs` overlays the one in rofs.

thilo-maurer, 2019-03-04:
Which version has this fixed?

Upgraded to latest `ibm-v2.3-476-g2d622cb-r24-0-g8c6bbf8`, then changed password via GUI.

After that ipmi password still unchanged
```
$ ipmitool -H witherspoon -I lanplus -P 0penBmc power status
Chassis Power is on
```

rthomaiy, 2019-03-04:
@rahulmah @tomjoseph83 @ratagupt 
Can you guys, check this version. I am not sure, how RMCP+ connection is established without -U option in first place. We removed those, but i am not sure, whether you have enabled it in internal build. Can you please look in to this issue, as i don't have access to this system. Let me know if you have any questions

thilo-maurer, 2019-03-05:
Digging around lead me to https://github.com/openbmc/phosphor-host-ipmid/issues/128

rthomaiy, 2019-03-06:
@thilo-maurer https://github.com/openbmc/phosphor-host-ipmid/issues/128 is different issue, and not related to this one. I am waiting for @rahulmah @tomjoseph83 comments on the same (Hope they will be able to help on this issue).

rahulmah, 2019-03-06:
@rthomaiy @xuweibj : This issue is with older openbmc which does not have support for -U. With latest openbmc build(with -U support), we don't see this issue.

xuweibj, 2019-03-06:
Opened CQ defect https://w3.rchland.ibm.com/projects/bestquest/?defect=SW459125
@rahulmah , could you help to tell us how to modify the password of IPMI interface on the older BMC version?

sivassrr, 2019-03-06:
@xuweibj Please avoid sending internal IBM information onto github open community. Slack / contact Rahul.

closed 2019-03-06 by xuweibj