openbmc/openbmc #3489: KCS bridge daemon should not write to the device file when call times out

kunyi731, 2019-02-28:
In KCS bridge daemon, when the internal timer expires and some operation is still pending, it attempts to set (context->expired) and then return 0xce to the host system. See https://github.com/openbmc/kcsbridge/blob/master/kcsbridged.c#L324

However it violates the IPMI spec by doing so. KCS state machine has strict requirements that only when in phase WAIT_READ can a BMC write to the shared buffer. As a result, the timeout response would never reach the host. See https://elixir.bootlin.com/linux/latest/source/drivers/char/ipmi/kcs_bmc.c#L353 for the kernel driver handling for reference.

vijaykhemka, 2019-03-21:
I am seeing these timeout for less than a second where as timeout value is set as 5 sec. I see continuously screen rolling with timeout log.

kunyi731, 2019-03-27:
> I am seeing these timeout for less than a second where as timeout value is set as 5 sec. I see continuously screen rolling with timeout log.

Could you be more specific on how to reproduce the issue in your situation? Having screen rolling might be an artifact of chunked outputs getting flushed out, and should generally not be correlated with precise timing..

closed 2019-10-26 by stale[bot]