openbmc/openbmc #3492: Removal of Python

krtaylor, 2019-03-01:
Finish up migrating off and removing Python for footprint reduction

geissonator, 2019-05-06:
https://gerrit.openbmc-project.xyz/q/topic:%22rmpython%22+(status:open%20OR%20status:merged) was what did this in the end. Default is no python in systems like witherspoon and romulus.

geissonator, 2019-05-06:
@mine260309 how we looking for Romulus?

mine260309, 2019-05-07:
Just submitted https://gerrit.openbmc-project.xyz/c/openbmc/meta-phosphor/+/21208, with that, we do not have any python services.
But it's noticed that the python files in `/usr/lib/python2.7` are still there.

geissonator, 2019-05-20:
This is complete, by default python is not required for the meta-phosphor layer and it's defaults. This is not a guarantee that all systems are python free, but the capability is now there and  has been done on all meta-ibm systems.

closed 2019-05-20 by geissonator