openbmc/openbmc #3497: skiboot does not stating with enabled mbox on P8

nest1ing, 2019-03-11:
This issue was appeared during the testing of https://gerrit.openbmc-project.xyz/c/openbmc/openpower-pnor-code-mgmt/+/18759.
As it is not relative to that patch set, I've posted it on https://gerrit.openbmc-project.xyz/c/openbmc/meta-ibm/+/18462 and then here.

All tests were made on our VESNIN hardware (P8-based host and AST2400 BMC).

I've restored the mbox like in https://gerrit.openbmc-project.xyz/c/openbmc/meta-ibm/+/18462, built the latest PNOR firmware with enabled PNORDD_IS_IPMI and fix `flash_memory` offset in our DTS as it described in the commit message of that pull request.

SBE and Hostboot work fine, but skiboot freezes and watchdog on BMC side restarts the host.
The last messages from the host console:
```
...
[  391.501917468,7] PHB#0001:02:03.0 Bus 06..ff  scanning...
[  391.501994092,7] PHB#0001:06:00.0 Found VID:14e4 DEV:165f TYP:0 MF+ BR- EX+
[  391.502089399,7] PHB#0001:06:00.1 Found VID:14e4 DEV:165f TYP:0 MF+ BR- EX+
[  391.502198074,7] PHB#0001:02:04.0 Bus 07..ff  scanning...
[  391.505335199,7] PCI: Checking slot table against detected devices
```

At the same time the BMC journalctl has shown:
```
...
Mar 11 12:15:37 malevich.dev.yadro.com ipmid[1007]: IPMI Incoming: Seq 0x1b, NetFn 0x3a, CMD: 0x5a
Mar 11 12:15:37 malevich.dev.yadro.com ipmid[1007]: 0x000000: 04 10 e0 00 d3 00                               ......
Mar 11 12:15:37 malevich.dev.yadro.com ipmid[1007]: IPMI Response:
Mar 11 12:15:37 malevich.dev.yadro.com ipmid[1007]: 0x000000: 00 04 10 20 0f 10 00 e0                         ... ....
Mar 11 12:15:37 malevich.dev.yadro.com ipmid[1007]: IPMI Incoming: Seq 0x1c, NetFn 0x3a, CMD: 0x5a
Mar 11 12:16:32 malevich.dev.yadro.com phosphor-watchdog[1089]: watchdog: Timed out
Mar 11 12:16:33 malevich.dev.yadro.com phosphor-watchdog[1089]: watchdog: disabled
Mar 11 12:16:33 malevich.dev.yadro.com systemd[1]: Stopping Phosphor poweron watchdog...
Mar 11 12:16:33 malevich.dev.yadro.com systemd[1]: Created slice system-op\x2docc\x2ddisable.slice.
Mar 11 12:16:33 malevich.dev.yadro.com systemd[1]: Starting OpenPOWER OCC Active Disable...
Mar 11 12:16:33 malevich.dev.yadro.com systemd[1]: Created slice system-openpower\x2ddebug\x2dcollector\x2dwatchdog.sli
ce.
Mar 11 12:16:33 malevich.dev.yadro.com systemd[1]: Starting OpenPOWER debug data collector for host watchdog timeout...
Mar 11 12:16:33 malevich.dev.yadro.com systemd[1]: Stopped Phosphor poweron watchdog.
Mar 11 12:16:33 malevich.dev.yadro.com watchdog_timeout[1126]: Host watchdog timed out
Mar 11 12:16:34 malevich.dev.yadro.com systemd[1]: Started OpenPOWER OCC Active Disable.
Mar 11 12:16:35 malevich.dev.yadro.com phosphor-log-manager[896]: Failed to find metadata
Mar 11 12:16:35 malevich.dev.yadro.com phosphor-fru-fault-monitor[1014]: /xyz/openbmc_project/logging/entry/3 created
Mar 11 12:16:35 malevich.dev.yadro.com systemd[1]: Started OpenPOWER debug data collector for host watchdog timeout.
Mar 11 12:16:35 malevich.dev.yadro.com systemd[1]: Reached target Quiesce Target.
Mar 11 12:16:35 malevich.dev.yadro.com systemd[1]: Reached target Host instance 0 timed out refreshing Watchdog.
Mar 11 12:16:35 malevich.dev.yadro.com phosphor-host-state-manager[1015]: Auto reboot enabled, rebooting
Mar 11 12:16:35 malevich.dev.yadro.com phosphor-host-state-manager[1015]: Beginning reboot...
Mar 11 12:16:35 malevich.dev.yadro.com phosphor-host-state-manager[1015]: Host State transaction request
...
```
Full versions of both logs:
[malevich-bmc-trace-ipmid.log.gz](https://github.com/openbmc/openbmc/files/2952343/malevich-bmc-trace-ipmid.log.gz)
[malevich-host-trace-mem-and-drivers.txt.gz](https://github.com/openbmc/openbmc/files/2952344/malevich-host-trace-mem-and-drivers.txt.gz)

geissonator, 2019-03-11:
@amboar led the charge on this for us. Any thoughts Andrew?

nest1ing, 2019-03-12:
> What's the period of the watchdog timer?

I've tried to introspect corresponds dbus-object before the moment it was alarming:
```
root@malevich:~# busctl --no-pager --verbose introspect xyz.openbmc_project.Watchdog /xyz/openbmc_project/watchdog/host0
NAME                                TYPE      SIGNATURE RESULT/VALUE                             FLAGS
org.freedesktop.DBus.Introspectable interface -         -                                        -
.Introspect                         method    -         s                                        -
org.freedesktop.DBus.Peer           interface -         -                                        -
.GetMachineId                       method    -         s                                        -
.Ping                               method    -         -                                        -
org.freedesktop.DBus.Properties     interface -         -                                        -
.Get                                method    ss        v                                        -
.GetAll                             method    s         a{sv}                                    -
.Set                                method    ssv       -                                        -
.PropertiesChanged                  signal    sa{sv}as  -                                        -
xyz.openbmc_project.State.Watchdog  interface -         -                                        -
.ResetTimeRemaining                 method    b         -                                        -
.CurrentTimerUse                    property  s         "xyz.openbmc_project.State.Watchdog.Tim… emits-change writable
.Enabled                            property  b         true                                     emits-change writable
.ExpireAction                       property  s         "xyz.openbmc_project.State.Watchdog.Act… emits-change writable
.ExpiredTimerUse                    property  s         "xyz.openbmc_project.State.Watchdog.Tim… emits-change writable
.Initialized                        property  b         true                                     emits-change writable
.Interval                           property  t         60000                                    emits-change writable
.TimeRemaining                      property  t         16486                                    emits-change writable
```

amboar, 2019-03-12:
I assume that's 60000ms? we should definitely be through skiboot before that's expired. Need to dig some more. @mine260309 have you seen anything like this in your testing on Palmetto?

mine260309, 2019-03-13:
When I was testing on Palmetto, I was using a skiboot that has @amboar 's old patch. It worked fine.
I did not test the latest skiboot though.
Will try the latest skiboot.

mine260309, 2019-03-13:
Tested on Palmetto with latest skiboot 261ca8e, and it works fine.

amboar, 2019-03-14:
@nest1ing can you instrument ipmid/mboxd to help us understand what the hold-up is there?

nest1ing, 2019-03-15:
We have found out that the blame is on our local patch for the skiboot.
It tries to send list of the PCI devices to the BMC and waits an answer.
At the time the BMC (it was cleaned from most our internal patches for this test) does not know what to do with these ipmi messages and keep silence.

So, this issue may be closed.

But I still have questions:
1. We have 64M PNOR flash, but in our DTS we specify:
```
...
	flash_memory: region@5c000000 {
		no-map;
		reg = <0x5c000000 0x02000000>; /* 32M */
	};
...
```
Is it right?

2. We could not to reproduce the switching to the golden side of the PNOR flash. 
    Is it working anywhere? It still need with this new way?

amboar, 2019-03-18:
> We have 64M PNOR flash, but in our DTS we specify ... `/* 32M */`

Yes; a property of the protocol is that you do not need the amount of reserved memory to at least be the size of the PNOR. We do not try to load the entire PNOR into memory at once.

> We could not to reproduce the switching to the golden side of the PNOR flash. Is it working anywhere? Is it still needed with this new way?

Hmm... P8 is almost before my time, so I'm not actually sure how the golden-side functionality worked. Can you outline the behaviour you expect? Does the board design use two distinct PNOR chips or does it divide the one chip in half?

artemsen, 2019-03-18:
We use PNOR flash with golden side format:
https://github.com/open-power/pnor/blob/master/p8Layouts/defaultPnorLayoutWithGoldenSide.xml
It's one flash logically separated into two sides.

By design, OpenBMC must switch the side if it was corrupted: https://github.com/openbmc/skeleton/blob/666be2aee1f4c2e123a6823e59c7d883709db89c/op-hostctl/control_host_obj.c#L154
It depends on DBus property "flash_side", which handled by libopenbmc_intf from skeleton.
But in fact, there is no modules that set the property if boot process failed, instead of this we just enter the quiesce state.

Anyway, I think that if we will use mbox, that "golden" feature is useless as we can load the PNOR image from any place. Isn't it?

mine260309, 2019-03-18:
Oh, if it uses the way in https://github.com/openbmc/skeleton/blob/666be2aee1f4c2e123a6823e59c7d883709db89c/op-hostctl/control_host_obj.c#L154, it's OK for P9, but we need some work to do in P8.

* On P9, [openpower-proc-control][1] checks the boot count sensor, and select which SBE side to start. The sensor is decreased by 1 when a boot fails.
* On P8, it checks the `flash_side` dbus property, it is legacy. The code should be changed to check the boot count as P9, and then it's OK.

From mbox perspective, I do not think there is anything needs to be changed.

[1]: https://github.com/openbmc/openpower-proc-control/blob/master/procedures/p9/start_host.cpp#L70-L81

nest1ing, 2019-03-18:
Well, I close this issue as the skiboot works fine.

closed 2019-03-18 by nest1ing