openbmc/openbmc #3498: BIOS flash update on X86 platform

MichaelWang2017, 2019-03-12:
Hi all,
I am trying to upgrade BIOS for X86 platform.
We use a GPIO to control an external bus switch to select the SPI BIOS flash, which is directed to Host system or BMC SPI controller interface. 
So my approach is:
1. Pull GPIO to switch to BMC SPI controller interface.
2. and bind flash driver.
3. finally use flashcp to get it.
But I found /sys/bus/spi/devices is empty, and "echo -n "spi1.0" > /sys/bus/spi/drivers/m25p80/bind" failed.
My .config like this: 
CONFIG_MTD_M25P80=y
CONFIG_REGMAP_SPI=y
CONFIG_MTD_SPI_NOR=y
CONFIG_SPI_ASPEED_SMC=y
CONFIG_SPI=y
CONFIG_SPI_MASTER=y
CONFIG_SPI_MEM=y
CONFIG_SPI_SPIDEV=y
CONFIG_RTC_I2C_AND_SPI=y
CONFIG_BMP280_SPI=y
My dts like this:
&spi1 {
        status = "okay";
        pinctrl-names = "default";
        pinctrl-0 = <&pinctrl_spi1_default>;

        flash@0 {
                status = "okay";
                m25p,fast-read;
                label = "pnor";
        };
};
I have no idea why /sys/bus/spi/devices is empty (at least BMC flash should be visible). Is there something wrong? 
Please give me some ideas.
Thank you.

geissonator, 2019-03-12:
@vmauery @feistjj or @edtanous can maybe help?

MichaelWang2017, 2019-03-12:
I don't know if the X86 platform already has an implementation for upgrading the BIOS image. Does anyone know?

edtanous, 2019-03-13:
The x86 platform (I'm assuming you're talking about S2600WF) does not have an implementation of BMC based BIOS update.  The hardware itself takes a strong opinion on security, so while on a supermicro style server the BMC might have direct access to the BIOS SPI flash, the S2600WF has no access, and has to communicate directly to BIOS to do any for of out of band update, making it a very different process.
Does you have any prints about not being able to install the device at BMC boot time?
In general in these cases, usually I've found I had issues with conflicting pinctl definitions.  With that said, your guess is probably as good as mine.

MichaelWang2017, 2019-03-14:
Thank you for your reply @edtanous. I checked the hardware design to make sure there was no pin multiplexing conflict. Our hardware design only needs to simply pull up a GPIO to switch BIOS flash to BMC. I refer to [https://github.com/facebook/openbmc/blob/helium/meta-facebook/meta-fbtp/recipes-fbtp/plat-libs/files/bios/bios.c](url), which will execute "echo -n "SPI 1.0" > /sys/bus/spi/drivers/m25p80/bind" && flashcp. 
> root@superior:~# dmesg | grep "spi"  
root@superior:~# dmesg | grep "flash"
[    2.796185] aspeed-smc 1e620000.flash-controller: Using 50 MHz SPI frequency
[    2.803504] aspeed-smc 1e620000.flash-controller: w25m512jv (32768 Kbytes)
[    2.810505] aspeed-smc 1e620000.flash-controller: CE0 window [ 0x20000000 - 0x22000000 ] 32MB
[    2.819046] aspeed-smc 1e620000.flash-controller: CE1 window [ 0x22000000 - 0x2a000000 ] 128MB
[    2.827742] aspeed-smc 1e620000.flash-controller: read control register: 203b0641
[    3.373641] aspeed-smc 1e630000.flash-controller: Using 100 MHz SPI frequency
[    3.380975] aspeed-smc 1e630000.flash-controller: unrecognized JEDEC id bytes: ff, ff, ff
[    3.389174] aspeed-smc 1e630000.flash-controller: Aspeed SMC probe failed -2
[    3.402785] aspeed-smc: probe of 1e630000.flash-controller failed with error -2
[    3.411938] aspeed-smc 1e631000.flash-controller: Using 50 MHz SPI frequency
[    3.419069] aspeed-smc 1e631000.flash-controller: unrecognized JEDEC id bytes: 00, 00, 00
[    3.427385] aspeed-smc 1e631000.flash-controller: Aspeed SMC probe failed -2
[    3.439251] aspeed-smc: probe of 1e631000.flash-controller failed with error -2
root@superior:~# 
**Now I wonder why there is no "BIOS flash" in /sys/bus/spi/devices/, and if it really doesn't work, is there any other way to try?**

closed 2019-10-13 by stale[bot]

chnguyen-ampere, 2020-11-25:
Hi @MichaelWang2017 , anything else update in this issue. I'm facing the issue like that. Thank you !

mdmillerii, 2020-11-25:
The Aspeed SPI flash driver in the kernel requires all SPI sockets for a given instance that are described i the device tree  be populated and respond to the JEDEC spi device.    

Note that the device to bind  or unbind is the `aspeed-smc` driver not the `m25p80` and the device name to bind is a platform device.

chnguyen-ampere, 2020-11-27:
Hi ,

In my system, it uses MX25L25635E for BIOS Flash device.

After, I set GPIO to control an SPI bus switch to select the SPI BIOS flash to BMC. Then, I try to probe the ASPEED-SMC driver and get the issue:

```
[ 1821.154996] aspeed-smc 1e630000.spi: Using 50 MHz SPI frequency
[ 1821.161018] aspeed-smc 1e630000.spi: unrecognized JEDEC id bytes: 0a 00 01 00 0a 0a
[ 1821.168843] aspeed-smc 1e630000.spi: Aspeed SMC probe failed -2
[ 1821.198015] aspeed-smc: probe of 1e630000.spi failed with error -2 
```

But, after I try to reset the SPI BIOS flash by the RESET# pin and bind again driver . The issue not happen :

```
[  743.717440] aspeed-smc 1e630000.spi: Using 50 MHz SPI frequency
[  743.738876] aspeed-smc 1e630000.spi: mx25l25635e (32768 Kbytes)
[  743.744933] aspeed-smc 1e630000.spi: CE0 window [ 0x30000000 - 0x32000000 ] 32MB
[  743.752351] aspeed-smc 1e630000.spi: CE1 window [ 0x32000000 - 0x38000000 ] 96MB
[  743.759851] aspeed-smc 1e630000.spi: read control register: 203c0641

```

Do you have any idea about my case? Thank you !

mdmillerii, 2020-12-04:
I don't know what the expected bytes are to compare them.  Perhaps the device was left in the middle of a transfer when the mux switched, or the clock glitched, or some other issue.  If you also have GPIO control of the reset in addition to the mux I would consider moving the mux with the device in reset.

zpdsg123, 2022-08-11:
> Contributor

Do you have fixed this issue? I also face that recently.